Seagate/openSeaChest #108: Seagate SMART parsing for smartmontools

jmhands, 2023-03-15:
openSeaChest (obviously) correctly parses the SMART for Seagate drives with `-smartAttributes analyzed`

Many open-source tools still use smartctl (e.g. smartd, Prometheus / Node Exporter, TrueNAS, UnRAID)

Seagate SMART requires extra parsing on Raw_Read_Error_Rate and Seek_Error_Rate. They report the total bytes read and errors that needs to be parsed, which causes people to think that their drives have failed. I recommend the openSeaChest folks port the parsing over to smartmontools
https://www.smartmontools.org/browser

vonericsen, 2023-03-16:
Hi @jmhands,

Thanks for reporting this to us. We will look into how we can submit a patch for this.

vonericsen, 2023-05-12:
@sachitanands,

After some internal review, we should make this update for attributes 1, 7, and 195. Raw 6:4 should be the displayed counter like we have in the analyzed smart attributes output in openSeaChest.
Attribute 188 should also be updated to show raw 1:0 for the counter value.

I just tagged a couple updates to this issue that I thought were important and related even though they belong in openSeaChest rather than smartmontools. 
These updates help clarify the output for these attributes in the `openSeaChest_SMART -d <handle> --smartAttributes hybrid` output since it is possible to have non-warrantied attributes at/below threshold to indicate warnings that are not necessarily failures. This also fixes the counters for the above attributes within openSeaChest to match the error counts that users seem most interested in.

walterav1984, 2024-01-27:
Exos X20 20TB also affected, users running `smartctl` could run the command with the following addition which will show 0 for the current error Rates:
```
sudo smartctl /dev/sdX -a -v 1,raw48:54 -v 7,raw48:54
```
Or add a similar block like the Exos X16 for the Exos X20 in `/var/lib/smartmontools/drivedb/drivedb.h` and run `smartctl` normally:
```
  { "Seagate Exos X20", // tested with ST20000NM007D-3DJ103/SN03
        // ST20000NM007D-3DJ103/SC03
    "ST2[0]000NM00[7]D-.*",
    "", "",
    "-v 1,raw48:54,Read_Error_Rate "
    "-v 7,raw48:54,Seek_Error_Rate "
    "-v 18,raw48,Head_Health "
    "-v 200,raw48,Pressure_Limit "
    "-v 240,msec24hour32"
  },
```

@vonericsen could you show -v 188 and -v 195 or more important ones needs to be interpreted in above format?

vonericsen, 2024-01-29:
@walterav1984,
That might work. I was looking into finding a way to do this for any Seagate drive, that way each drive does not need to be added as another entry in the database.

This issue has unfortunately been on hold due to resource constraints lately, but I have left it open as this is still important and should get updated when we get the time again.

walterav1984, 2024-01-29:
Could you hint what the interpretation of the 188 and 195 value's are for the EXOS X20, since 4 out of 8 new disks show errors between 1-12 counting with openSeaChest_SMART. 

Its maybe a non worrying value as a left over result from 512 > 4K sector changing which disconnected and reconnected the drive itself when it finished.

walterav1984, 2024-02-01:
Thanks for the comprehensive answer, although I'm a little bit shocked by the so called "background" processing time of disks that might have introduced the "timeout counter" to go up.

The disks were directly set to 4k sectors on their first boot, it took a couple of minutes before it mentions succeed with a bit of a "shaddy" warning message which hints that atleast 1 hour to do some background stuff and that it may take longer. But reading your comment it looks like it can be like multiple 30 hours(whole read/write disk cycle)...

The machine was rebooted after an hour of the success message that it was set to 4k, but if its very important not to interrupt should it be a better idea to make this 4k sector switching a "online" process on which the terminal command won't quit or won't go to background? Or make an option to view the actual background process?

vonericsen, 2024-02-05:
@walterav1984,

I will figure out a better way to rewrite that message. The purpose of it was to inform the user not to worry if the utility appears to "hang" for an hour while the sector size change starts and not to interrupt it during this time. Interrupting during this time that it is busy can make the drive fail to work properly and require rerunning the format or some other vendor unique recovery process.
I will try to make it clear that background processing will occur once the drive has finished this first part of the sector size change.

In the standards for this command, it is described as issuing the command, the drive holds the bus in a "busy" state until the sector size change is done. There is no requirement on background processing as it is vendor unique what happens once the drive has returned status.
The busy state is the only part of this that is "online" and once it is done the drive is ready to read and write.

The background processing the drive is doing cannot be forced to the foreground unless you are writing data to the drive. There is no way to view the status of background processes the drive is performing.

The intention of this was to allow the user to write their own data rather than waiting for a complete write of the entire drive (possibly in a number of days with today's and the future's HDD capacities) and allow the user to quickly set it up and begin using the drive with their own data.

If you were to do a complete overwrite of the entire drive, this would essentially force the background processing the disk is doing into the foreground (this only applies to changing sector size...other background tasks may not be able to be forced to foreground). You can do this with whatever data you want, but it is not necessary to do before using the disk.

feinedsquirrel, 2024-12-13:
@vonericsen 

Is there documentation about the background operations somewhere?
Skimming through "About_Fast_Format.txt" and "About_Format_Unit.txt", and the man page for openSeaChest_Format, I haven't been able to find anything.

My questions are:

Once the terminal command returns (after printing "Successfully set sector size to ### \n Note: ..." is it safe to:
a) reboot during the background processes? (The man page for e.g.sanitize erase says the sanitize erase command is persistent across power cycles and cannot be stopped, is it the same for these background processes?)

aside: if (a) is safe, then these other questions/options are not applicable, I suspect

b) disconnect the sata cable leaving the drive powered, will it finish the background processes?
b-1) if I shutdown the machine, disconnect the drive entirely, and connect it to power on a bench power supply, will it finish the background tasks by itself?
c) reboot with a different drive attached (both powered, only new drive connected to SATA port on mobo), to format it from 512e to 4kn? Then if successfully converted, reboot again with both drives attached and force completion of background ops by forcing write of zeros to both drives with a separate command to each?
d) disconnect the entire machine from power for a couple of days to re-locate it?

Thanks for the guidance on all of this!
I also wanted to say thank you to you, the seachest team, and the firmware team. I had a conversion to 4kn fail, but after panicking and taking several logs that were accessible, running format to 512 recognized it had failed and did a successful recovery. To me, that seems to be an indication of thorough programming.  :)  so great work. I have since been able to convert that same drive to 4kn successfully.

vonericsen, 2025-01-06:
@feinedsquirrel,

I have written something in various issue comments and I have been slowly working on adding it to the openSeaChest Wiki pages.
At the time of this comment I have not pushed that new wiki page yet (I have had other tasks I need to complete first).

>Once the terminal command returns (after printing "Successfully set sector size to ### \n Note: ..." is it safe to:
a) reboot during the background processes? (The man page for e.g.sanitize erase says the sanitize erase command is persistent across power cycles and cannot be stopped, is it the same for these background processes?)

Yes. This is safe. Backround processes may vary between operations and what they are doing during the operation. In the case of sanitize it is continuing to run the sanitization of user data.
For other operations like fast-format the tasks are different and will resume when necessary and other conditions are met in the firmware (not actively reading or writing data and not forced to a standby power state by a command).

> b) disconnect the sata cable leaving the drive powered, will it finish the background processes?

This should be fine. The drive will continue running the background tasks uninterrupted. Note that this may be slower than just starting to write data to the drive. As you write after a fast format has completed, this background operation's task is performed for each write to the drive in the foreground (each write issued to it). You could do a full overwrite or you could start writing the data you want to store there.
The purpose of fast format was to change the sector size quickly and allow the user to start writing their data immediately instead of waiting hours or days before being able to use the drive after it overwrote with zeros like the SAS format unit command requires. If the user wants to overwrite with zeroes first, they can...or they can start writing their own data to the drive instead.

> b-1) if I shutdown the machine, disconnect the drive entirely, and connect it to power on a bench power supply, will it finish the background tasks by itself?

I believe this would work without a problem. This is one way I ran sanitize early in development with multiple drives....I had to start an overwrite, find a bug, then put the drive to the side to let it finish and grab another one and try again after my code changes.
This should work for the post fast format background tasks too, but even uninterrupted this is likely a lot slower than just writing to the drive after the format. Sanitize is different since its task is to remove all user data it will do so as quickly as it can without trying to minimize power consumption in an idle state like the post fast-format task is doing.
As long as the drive is powered and idle, it will run this task after a fast format until each sector has been written to or the background task has reviewed each sector on the drive.

> c) reboot with a different drive attached (both powered, only new drive connected to SATA port on mobo), to format it from 512e to 4kn? Then if successfully converted, reboot again with both drives [...]

This would work, but you could just fast format both, leaving the interface attached. Then start the full overwrite on both. It won't matter to the background tasks as this will just override some of the work it is doing in the background as you write new data (even zeroes).

> d) disconnect the entire machine from power for a couple of days to re-locate it?

This is fine. The background task will start again when it receives power again....or if you have written data it will only service areas you have not yet written to.