Seagate/openSeaChest #109: How to update Firecuda firmware if I use a SSD enclosure?

shanewazabbas, 2023-03-31:
```
.\command line tools\SeaChest\Win64> .\SeaChest_Firmware_x64_windows.exe -d PD1 -i --usbChildInfo
```

I have simplified the output below. Using `--usbChildInfo` seems to display the Firecuda details. But for some reason I don't have the ability to download the firmware file to it. Any ideas how to solve it.

```
SCSI Translator Reported Information:
        Vendor ID: Realtek
        Model Number: RTL9210B-CG
        Serial Number: 0000000000000000
        Firmware Revision: 1.00
        World Wide Name: 3001237923792379
        Drive Capacity (TB/TiB): 2.00/1.82
        Read Look-Ahead: Enabled
        Write Cache: Enabled
        SMART Status: Unknown or Not Supported
        ATA Security Information: Not Supported
        Firmware Download Support: Full, Segmented
        Specifications Supported:
                SPC-4
                UAS
                SPC-4
                SBC-3
        Features Supported:
                UNMAP
                Power Conditions [Enabled]
                Informational Exceptions [Mode 0]
        Adapter Information:
                Vendor ID: 0BDAh
                Product ID: 9210h
                Revision: 2001h
ATA Reported Information:
        Model Number: Seagate FireCuda 530 ZP2000GM30023
        Serial Number: 7VR01XM1
        Firmware Revision: SU6SM003
        World Wide Name: Not Supported
        Drive Capacity (B/B): 0.00/0.00
        Last DST information:
                Not supported
        Long Drive Self Test Time: Not Supported
        Interface speed:
                Max Speed (MB/s): 3.30 (PIO-0)
                Negotiated Speed (MB/s): Not Reported
        Read Look-Ahead: Not Supported
        Write Cache: Not Supported
        SMART Status: Unknown or Not Supported
        ATA Security Information: Not Supported
        Firmware Download Support: Not Supported
        Specifications Supported:
                None reported by device.
        Features Supported:
                None reported or an error occurred while trying to determine
                the features.
        Adapter Information:
                Vendor ID: 0BDAh
                Product ID: 9210h
                Revision: 2001h
```

vonericsen, 2023-03-31:
Hi @shanewazabbas,

USB attachments are always tricky as they add an extra layer of translation to work through.
Not all of them allow all the commands you want through to the target drive. Most adapters only focus on the ability to read/write the device...not all the additional diagnostics or support you may otherwise need such as a situation like this.
This adapter also did something non-standard to make the firecuda drive look like an ATA drive rather than an NVMe drive that it should be. During this odd translation it did not set the support fields for doing a firmware download using the ATA command sequence, which is likely why openSeaChest returned a `NOT SUPPORTED` message.

We have not been able to get a specification from Realtec on NVMe passthrough commands for their USB to NVMe adapters when I reached out through their support portal in the past.
I have implemented USB to NVMe passthrough for JMicron's chip and ASMedia's chips (only one of theirs allows full passthrough. I believe it is the 2364. Earlier 2362 is limited to a few commands).

The only thing you might be able to do to update the firmware with this adapter is to try forcing it to use a SCSI firmware update sequence instead and see if it translates that. According to the chip listing I [found here](https://www.realtek.com/en/products/communications-network-ics/item/rtl9210b-cg), this might be able to work since firmware update is part of the SCSI to NVMe translation whitepaper and Realtek is claiming to support all translations.
`openSeaChest_Firmware -d <handle> --downloadFW file.bin --forceSCSI`
You may need to force the download mode too, so also try
`openSeaChest_Firmware -d <handle> --downloadFW file.bin --downloadMode deferred --activateFW --forceSCSI`
If this does complete without indicating errors, you may need to power cycle the adapter depending on how it translated the activation event for you.
Please let me know if this ends up working for you or not. I'm curious to know what this adapter will do. I may be able to add at least some "rules" for how to handle this specific USB adapter for other commands as well.

shanewazabbas, 2023-03-31:
Hi, @vonericsen. Thank you for the quick and detailed reply. 
I will try the second command soon. Just wondering when you say "need to power cycle the adapter" is there a specific command I should run, and if so what would that be. Or do I need to just eject the usb drive.

Also the above command that I ran (`Seachest -d PD1 -i --usbChildInfo`), I noticed that it takes a bit long to run. Is that expected. I could post the verbose level 4 output if needed.

shanewazabbas, 2023-03-31:
I ran the command to update and unfortunately it didn't work:
<details><summary>Update firmware result</summary>
<p>

```
==========================================================================================
 SeaChest_Firmware - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2021 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Firmware Version: 3.0.0-2_2_1 X86_64
 Build Date: Apr 27 2021
 Today: Fri Mar 31 16:20:22 2023        User: Test2 (admin)
==========================================================================================
Sending SCSI Inquiry
Inquiry returning: SUCCESS

Sending SCSI Inquiry, VPD = 00h
Inquiry returning: SUCCESS

Sending SCSI Inquiry, VPD = 80h
Inquiry returning: SUCCESS

Sending SCSI Inquiry, VPD = 83h
Inquiry returning: SUCCESS

Sending SCSI Inquiry, VPD = B1h
Inquiry returning: SUCCESS

Sending SCSI Read Capacity 10 command
Read Capacity 10 returning: SUCCESS

Sending SCSI Read Capacity 16 command
Read Capacity 16 returning: SUCCESS

Sending SCSI Inquiry, VPD = 89h
Inquiry returning: NOT SUPPORTED

Sending ATA Identify command
Identify returning: SUCCESS

        Forcing SCSI Drive

\\.\PhysicalDrive1 - RTL9210B-CG - 0000000000000000 - SCSI
Sending SCSI Inquiry, VPD = 86h
Inquiry returning: NOT SUPPORTED

Requesting SCSI Supported Op Codes
Supported Op Codes returning: NOT SUPPORTED

Sending SCSI Inquiry, CmdDt = 3Bh
Inquiry returning: NOT SUPPORTED

Sending SCSI Read Buffer
Read Buffer returning: NOT SUPPORTED

Sending SCSI Write Buffer
Write Buffer returning: NOT SUPPORTED

.
Firmware Download not supported
Sending SCSI Inquiry, VPD = 86h
Inquiry returning: NOT SUPPORTED

Requesting SCSI Supported Op Codes
Supported Op Codes returning: NOT SUPPORTED

Sending SCSI Inquiry, CmdDt = 3Bh
Inquiry returning: NOT SUPPORTED

Sending SCSI Read Buffer
Read Buffer returning: NOT SUPPORTED
[... 321 lines cut ...]
                Max Speed (MB/s): 3.30 (PIO-0)
                Negotiated Speed (MB/s): Not Reported
        Annualized Workload Rate (TB/yr): Not Reported
        Total Bytes Read (B): Not Reported
        Total Bytes Written (B): Not Reported
        Encryption Support: Not Supported
        Cache Size (MiB): Not Reported
        Read Look-Ahead: Not Supported
        Write Cache: Not Supported
        SMART Status: Unknown or Not Supported
        ATA Security Information: Not Supported
        Firmware Download Support: Not Supported
        Specifications Supported:
                None reported by device.
        Features Supported:
                None reported or an error occurred while trying to determine
                the features.
        Adapter Information:
                Vendor ID: 0BDAh
                Product ID: 9210h
                Revision: 2001h
```

</p>
</details>

vonericsen, 2023-03-31:
@shanewazabbas,

> Just wondering when you say "need to power cycle the adapter" is there a specific command I should run, and if so what would that be. Or do I need to just eject the usb drive.

Just eject it and plug it back in. If you do this through the operating system's eject it will tell the drive it is being powered off. I think it will be similar to `openSeaChest_PowerControl -d <handle> --transitionPower sleep` or possibly only standby mode. This just sends the device into a low-power condition that flushes any write cache safely before you eject it to prevent any data loss.

> I noticed that it takes a bit long to run. Is that expected. I could post the verbose level 4 output if needed.

Yes and no. The device discovery that is done by openSeaChest is very thorough and looks for more things than most USB adapters know how to reply to...and many USB adapters I've tested have issues recovering from commands that they do not know how to service.
While the `-i -v4` could be useful to learn more about this device, we have `openSeaChest_PassthroughTest` which will give a more detailed report and list of workarounds that we can implement that will likely speed it up again by telling the software which commands not to send to it.
If you are willing to run the passthrough test tool, can you run it each of these ways and share the output it gives you?

1. `openSeaChest_PassthroughTest -d <handle> --runPTTest --disablePassthroughTesting`
2. `openSeaChest_PassthroughTest -d <handle> --runPTTest`

I'm a little unsure if the output results will tell everything I need to know, but I think it will help start some changes.
I'm not sure I will need the verbose output from this yet, I will have to see what it gives as a response to this first.

Do you happen to have a link to where you purchased this adapter? Seagate can look into buying it for me to test it myself since the testing could get complicated depending on what you get as a result....and that the product page says it supports SATA and NVMe, so one set of rules for NVMe might not work well for SATA or visa-versa.

vonericsen, 2023-03-31:
One more thing you can try is a full buffer download in a single command. It is not used often since the transfers can be so large a bus like USB may not allow them and some OS's (*cough* *cough* Windows) does not like large transfers (anything above 64KiB is usually not allowed).
`openSeaChest_Firmware -d <handle> --downloadFW full --forceSCSI`

shanewazabbas, 2023-03-31:
First passthrough command output:
<details><summary>Details</summary>
<p>

```
==========================================================================================
 openSeaChest_PassthroughTest - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_PassthroughTest Version: 1.3.1-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Fri Mar 31 16:57:53 2023        User: XXX
==========================================================================================

\\.\PhysicalDrive1 -  -  -  - SCSI
        Vendor ID: Realtek
        Model Number: RTL9210B-CG
        Serial Number: 0000000000000000
        Firmware Revision: 1.00
        World Wide Name: 3001237923792379
        Drive Capacity (TB/TiB): 2.00/1.82
        Temperature Data:
                Current Temperature (C): Not Reported
                Highest Temperature (C): Not Reported
                Lowest Temperature (C): Not Reported
        Power On Time: Not Reported
        Power On Hours: Not Reported
        MaxLBA: 3907029167
        Native MaxLBA: Not Reported
        Logical Sector Size (B): 512
        Physical Sector Size (B): 512
        Sector Alignment: 0
        Rotation Rate (RPM): SSD
        Form Factor: Not Reported
        Last DST information:
                Not supported
        Long Drive Self Test Time: Not Supported
        Interface speed:
                Not Reported
        Annualized Workload Rate (TB/yr): Not Reported
        Total Bytes Read (B): Not Reported
        Total Bytes Written (B): Not Reported
        Encryption Support: Not Supported
        Cache Size (MiB): Not Reported
        Percentage Used Endurance Indicator (%): Not Reported
        Read Look-Ahead: Enabled
        Write Cache: Enabled
        SMART Status: Unknown or Not Supported
        ATA Security Information: Not Supported
        Firmware Download Support: Full, Segmented
        Specifications Supported:
                SPC-4
                UAS
                SPC-4
                SBC-3
        Features Supported:
                UNMAP
                Power Conditions [Enabled]
                Informational Exceptions [Mode 0]
        Adapter Information:
                Adapter Type: USB
                Vendor ID: 0BDAh
                Product ID: 9210h
                Revision: 2001h
Performing Pass-through test.
If at any point during the test, a crash, a hang, or the device
seems to stop responding to any normal request, save all previously
reported hacks, stop the tool, reset the device, then restart
the test, skipping to another part of the test to avoid the same issue.
All discovered hacks will be necessary in order to fully communicate
with the end target device to the highest capabilities of the adapter or
[... 402 lines cut ...]
                TURF:34
        SCSI Hacks: RW6, RW10, RW12, RW16, NLP, NRSUPOP, MXFER:344064
        ATA Hacks:  SAT, A1, NDMA,
Recommendations For Device Makers:
1       Test this device with this tool in multiple OS's. Linux and Windows testing should be done to find highest compatibility.
2       Improve error handling for unknown commands. This device takes longer
            and longer to respond with each unknown command.
3       Rerun this test to test for commands that are known to hang some devices to ensure
        all hacks are known for this device to perform properly...if the firmware can be
        fixed to prevent these hangs, that is the best solution!
4       The maximum transfer length is less than was reported by the block limits VPD page!
        This page should report the maximum transfer length supported by the SCSI device or SCSI translator
        This should be a true maximum. If this seems low, retest on another OS to ensure it is not a
        limitation on a specific OS. For example, it is fairly common to be limited to 64k in Windows.
```

</p>
</details> 

I bought the ssd enclosure from a pc repair store. Can't find the exact link. The box legit calls it: `M.2 SSD Enclosure`. But design of it matches up with these:
- [Amazon]( https://a.co/d/fPb5hbc)
- [Alibaba](https://www.alibaba.com/product-detail/Usb3-1-Type-C-10Gbps-M_1600527803750.html )
- [Ebay](https://www.ebay.com/itm/394472664600)

[image]

shanewazabbas, 2023-03-31:
Tried the full download command and this failed as well (verbose level 3 this time):
<details><summary>Details</summary>
<p>

```
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Fri Mar 31 17:33:34 2023        User: XXX
==========================================================================================

====Sending SCSI Miniport Firmware Request====
Windows Error: 1117 - The request could not be performed because of an I/O device error.

Firmware Miniport Status: Success
Sending SCSI Test Unit Ready

  CDB:

        0  1  2  3  4  5
  0x00 00 00 00 00 00 00

  Sense Data Buffer:

        0  1  2  3  4  5  6  7
  0x00 00 00 00 00 00 00 00 00

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
Information: 0000000000000000h
Command Specific Information: 0000000000000000h
Command Time (us): 783.10

Test Unit Ready returning: SUCCESS

Sending SCSI Inquiry

  CDB:

        0  1  2  3  4  5
  0x00 12 00 00 00 60 00

  Sense Data Buffer:

        0  1  2  3  4  5  6  7
  0x00 00 00 00 00 00 00 00 00

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
Information: 0000000000000000h
Command Specific Information: 0000000000000000h
Command Time (ms): 1.31

Inquiry returning: SUCCESS

Sending SCSI Inquiry, VPD = 00h

  CDB:

        0  1  2  3  4  5
  0x00 12 01 00 00 60 00

  Sense Data Buffer:

        0  1  2  3  4  5  6  7
  0x00 00 00 00 00 00 00 00 00
[... 291 lines cut ...]

        0  1  2  3  4  5  6  7  8  9
  0x00 3B 05 00 00 00 00 20 20 00 00

Windows Error: 87 - The parameter is incorrect.

  Sense Data Buffer:

        0  1  2  3  4  5  6  7
  0x00 00 00 00 00 00 00 00 00

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
Information: 0000000000000000h
Command Specific Information: 0000000000000000h
Command Time (us): 15.10

Write Buffer returning: OS PASSTHROUGH FAILURE

Firmware Download failed
```

</p>
</details>

vonericsen, 2023-03-31:
`Windows Error: 87 - The parameter is incorrect.` -> This is Microsoft's way of reporting "Transfer size too large" so the update did not even make it past the OS. There is no way around this in Windows.

I submitted a purchase request for one of those adapters and I also reached out to Seagate's USB engineering team to see if they have a Realtek support contact so I can try and get some documentation or at least ask them some questions to figure out what we can do.

shanewazabbas, 2023-03-31:
Thank you so much. Really appreciate it. Seagate support suggested I update the firmware for my firecuda to resolve crashing issues with my PS5. Hence why I bought the enclosure as I don't have a computer that supports m2 natively.

vonericsen, 2023-03-31:
@shanewazabbas,
No problem. I'll do the best I can.
I'm not sure how long this will take me to figure out or get an answer on though.

I know you may not want to buy another adapter, but this one claims to use one of the USB to NVMe chips I know openSeaChest and SeaChest can support for a firmware update: https://www.amazon.ca/s?k=jms583&crid=2H35Q1NUEVYFK&sprefix=jms583%2Caps%2C158&ref=nb_sb_noss_1
This one also lists it, but might cost a little more: https://www.amazon.ca/JEYI-Enclosure-External-Support-JMS583A3/dp/B0BCZ9JZ7F/ref=sr_1_1?crid=2H35Q1NUEVYFK&keywords=jms583&qid=1680304964&refinements=p_72%3A11192170011&rnid=11192166011&sprefix=jms583%2Caps%2C158&sr=8-1
I *think* I bought this one a few years ago when I was first testing my code to see which USB-NVMe adapters I could find on Amazon would work.

I was just searching `JMS583` to find it and making sure it mentioned this same chip somewhere on the product page.

I just don't want to keep you waiting too long while I work to get an answer and figure this out.

shanewazabbas, 2023-04-01:
Hey, no worries. Its not a rush. I'm also in communication with support to try to see if I can get mine replaced since I am under warranty. I personally don't want to spend on anymore on adapters. If anything I'll try to see if I can find a pc repair place thats willing to put the firecuda ssd directly in the computer.

vonericsen, 2023-04-13:
Well I finally got the USB adapter in and have been playing with it to see if I can find a way to get it working.
So far I'm having the exact same results that are reported in this issue and I have not found a workaround yet.
I did send an email to a contact I was given today to see if they can help me out with finding a way to support this better. 

I tried a few things like forcing the download command to use SCSI commands, ATA commands, and I tried forcing 4k transfer sizes, and so far nothing has made a difference with getting a firmware to the drive.

While I wait for a response from Realtek, I will play around with this and see if I can at least setup some rules to make this perform a little better with the `-i` output for now.

shanewazabbas, 2023-04-14:
Hey, thanks for the update. From my end, I'm supposed to received the replaced ssd today. So will wait to see how that one goes. If its the same firmware version and the same issues, this will be worth it for sure to have a way to update it with that adapter. I also went to a few places and they said they wouldn't update it for me through motherboard. So here's hoping the new one works.

vonericsen, 2023-05-19:
Well I have tried everything I can think of to get this working and have reviewed the SCSI to NVMe translation white paper from original 1.0 to 1.5 versions for any differences that would affect the ability to update firmware and nothing has worked.
The adapter is responding with "Invalid operation code" indicating it does not support this command translation for an NVMe device.

I have also failed to hear anything from Realtek about how to do this as well.

So the changes I have pushed will make some things work a little better, but cannot fix the ability to update firmware on this adapter. I need a spec from Realtek to do this, the SCSI translation is not complete.

Here is a list of what I have tried with this adapter (from NVMe to SCSI translation whitepaper 1.5):

| Command | subcommand | Result | Notes|
|---------|------------|--------|------|
| Inquiry | | Success | Vendor ID, Product ID, FWRev do not match translation whitepaper. Did not validate all fields, but otherwise looked ok |
| | supported pages | Success | |
| | unit serial number | Success | Does not match SCSI translation whitepaper. Set to all zeroes rather than useful value. |
| | device identification | Success | Does not match SCSI translation whitepaper. Locally administered value rather than translated. |
| | extended inquiry data | missing, but optional | whitepaper lists this as optional. Not a major issue. |
| | block limits | Success | Not all fields validated |
| | block device characteristics | Success | not all fields validated, but seems correct |
| | logical block provisioning | Success | not all fields validated, but appears to match translation requirements |
| Log Sense | all pages | Fail | Command is rejected with Invalid operation code". Without this cannot read endurance used or temperature |
| Mode Sense | | partial success | Only seems to respond properly with "current" data. Saved, default, changable all seem to cause trouble and slow response |
| | read-write error recovery | Fail | Page not supported. Slow response |
| | caching | Success | |
| | control | Fail | |
| | power condition control | Success | |
| | return all pages/subpages | Success | Return all pages and subpages worked, as did the all pages (3f-ff & 3f-00)
| | informational exceptions control | Success | Matches whitepaper, but is missing the matching log page (see above) |
| Mode select | | Not tested | |
| Test unit ready | | Success | |
| report luns | | Success | |
| Security Protocol In/Out | | Not tested | Not really tested, but appears to respond as expected for attached device. |
| Write Buffer | All modes | Fail | Responds with "Invalid operation code". This is the command required to perform firmware updates on NVMe attached devices, but since this is not being implemented for translation, it is impossible to update firmware at this time. |
| Persistent reserve in/out | | Not tested | Not tested since likely only to be found on enterprise drives and unlikely to be used over USB interface anyways |
| Send Diagnostic | | Not tested | Not tested, but is a no-op in translation, so probably works fine |
| Read Capacity | 10B | Success | |
| | 16B | Success | |
| Compare and Write | | Not tested | Not expected to work since requires fusing NVMe commands. As verify for compare is not supported, I suspect this isn't implemented either. |
| Format Unit | | Fail | Command rejected with "Invalid operation code", so cannot use this to format/reformat nvm and cannot be used to change sector size if target drive supports multiple sector sizes. |
| Read | 6 | Success | |
| | 10 | Success | |
| | 12 | Success | |
| | 16 | Success | |
| Write | 6 | Success | |
| | 10 | Success | |
| | 12 | Success | |
| | 16 | Success | |
| Unmap (TRIM) | | Success | Not fully tested, but appears to work properly and match limits mentioned in VPD pages |
| Synchronize cache | 10 | Success | |
| Synchronize cache | 16 | Fail | Not surprising. Microsoft only uses 10B, but both translate exactly the same to NVMe's flush command.  No reason this couldn't work |
| Verify | 10 | ??? | Translation spec uses this for translation to NVMe compare command. NVMe verify added after translation. The compare translation is missing since you get "Invalid operation code". |
| | 12 | ??? | Translation spec uses this for translation to NVMe compare command. NVMe verify added after translation. The compare translation is missing since you get "Invalid operation code". |
| | 16 | ??? | Translation spec uses this for translation to NVMe compare command. NVMe verify added after translation. The compare translation is missing since you get "Invalid operation code". |
| Write long | | Not tested | Translation to NVMe write uncorrectable using the "flagged" method. Not tested, but this is more for testing drive error recovery. Not really something that would be useful on USB anyways |

I also tested this adapter with a SATA M.2 drive and the SAT CDBs seem to work fine in that case to get internal drive information and do other things. There are some odd things there, but it seems to mostly work. I did not do as in-depth of a command verification as I did with NVMe since the SAT passthrough CDBs worked well enough it was not necessary.

Unfortunately, there is not currently a way that I can figure out to update firmware using this adapter. I will need a specification from Realtek on how to do this, but since I cannot seem to get a response from them, there is nothing else I can currently do about this.

shanewazabbas, 2023-05-19:
Hey, thanks for putting so much of your time into this. Really appreciate it. Thus far, the replaced ssd has been working smoothly. Luckily it already had an updated firmware version, so I don't have to rely on this usb adaptor fix yet. I would say its up to you if you want to carry on or not for your researching findings. But I'm totally cool if you want to close the ticket as well as I have no issues yet.

closed 2026-09-30 by vonericsen