Seagate/openSeaChest #113: openSeaChest_PowerControl FreeBSD 12 always reports power condition as standby_z

chrcoluk, 2023-05-07:
Hi

The tool can set power state timers, it can also change the power state on demand.

However it always reports active power state as "Device is in the PM2: Standby state and device is in the Standby_z power condition"

I confirm actual state using camcontrol as well as monitoring power use and noise.

Problem is using camcontrol will report the correct power state but also bring the drive into idle_a state.

I was hoping openSeaChest could report the actual power state without waking up the drive, as the drive has no reason to spin up at full speed to report power state, this should just be accessing the chip.

syntax used below, for a SATA device.  (ironwolf 8tb)

openSeaChest_PowerControl --checkPowerMode -d /dev/ada2

current configuration for the drive to help testing, 2 mins idle idle_b (rest head). 5 mins idle idle_c (spin at lower rpm), **disabled** standby_z (spin down completely).

Watunder, 2023-11-06:
`Exos X18` `TrueNAS CORE 13.0-U5.3`
[checkPowerMode.txt](https://github.com/Seagate/openSeaChest/files/13261751/checkPowerMode.txt)

vonericsen, 2023-11-07:
@Watunder,
Thanks for sharing that!
The reason it is always reporting standby_z is the return task file registers are showing up as all zeroes. Even the status code is zero instead of something more common like 50h.

Reviewing the low-level CAM IO code, I do not see why these would be coming back as all zeroes unless that is what the drive reported.
I'm making a few changes to how some flags are set for issuing the command to see if it will help and possibly give more information about what happened but I'm not sure what else will correct this right now.

Do you have the same issue with camcontrol changing the power state when running `camcontrol /dev/ada0 powermode` like @chrcoluk reported in the original issue?

Watunder, 2023-11-07:
@vonericsen
> Do you have the same issue with camcontrol changing the power state when running like @chrcoluk reported in the original issue?`camcontrol /dev/ada0 powermode`

No, this works fine for me.
|Name |Current Timer |Default Timer |Saved Timer |Recovery Time |C |S
|:-|:-|:-|:-|:-|:-|:-
|Idle A | *1 | *1 | *0 | 1 | Y | Y
|Idle B | 0 | *1200 | 0 | 4 | Y | Y
|Idle C | 0 | 6000 | 6000 | 20 | Y | Y
|Standby Z | *6000 | 9000 | *9000 |110 | Y | Y

vonericsen, 2023-11-07:
Thanks for the feedback.
I'll work on my low-level changes and see if it helps. It may take a little bit as I'll have to setup a new FreeBSD test machine due to internal IT requirements/changes. If I get a build ahead of my testing from the CI I will also share it here.

mplzik, 2023-11-11:
Hello @vonericsen, apologies for hijacking this a bit, but I seem to have a similar issue running on Linux, making it possibly a drive firmware issue. Let me know if I should file a separate issue for this.
Despite of the drive being active, it always reports itself being in PM2. Attaching this:

```
(command that accesses the drive and spins it up)
# openSeaChest_PowerControl --checkPowerMode -d /dev/sg2 -v 4 >checkPowerState-debug.txt
# openSeaChest_PowerControl --checkPowerMode -d /dev/sg2 --SATInfo >SATInfo.txt
```
[SATInfo.txt](https://github.com/Seagate/openSeaChest/files/13326183/SATInfo.txt)
[checkPowerState-debug.txt](https://github.com/Seagate/openSeaChest/files/13326184/checkPowerState-debug.txt)

The drive had originally Idle C and Standby Z states disabled; I had to manually re-configure them. Even before, the reported state was Standby Z.

vonericsen, 2023-11-13:
@mplzik,

Looking at the information in your logs, I suspect your issue is related to the USB adapter you are using.
It might be possible to create a workaround for the adapter though.
Can you open a new issue for that since it's most likely an adapter issue whereas this is a low-level passthrough issue for FreeBSD specifically.

What you will need to do to help me add a workaround is run 
    
    openSeaChest_PassthroughTest -d <handle> --runPTTest --ptDriveHint ata --ptTypeHint sat > passthroughTest.txt

Please attach that output text file to that issue.
This is a data-safe automated test that I wrote to try and figure out which workaround will work best for a given adapter when we run into these kinds of issues that appear to be adapter specific.

mplzik, 2023-11-13:
@vonericsen I filed a separate issue at https://github.com/Seagate/openSeaChest/issues/128

vonericsen, 2023-11-14:
I have gotten a FreeBSD system running and have repeated this problem and I've verified that camcontrol is returning the correct value for the drives I have.
From my first pass over this issue it is not something easy like a missing flag, at least the ones I've looked at and tried, but since camcontrol works it's likely an issue with how the command structure is being built in openSeaChest.

vonericsen, 2023-11-15:
@Watunder & @chrcoluk,

I have found and fixed the bug causing this issue.
The change is on the develop branch and the release/Release-23.09 branch at this time.

These are the builds I have from our CI if you want to try these out instead of building:

[freebsd_12.3.zip](https://github.com/Seagate/openSeaChest/files/13369686/freebsd_12.3.zip)
[freebsd_13.1.zip](https://github.com/Seagate/openSeaChest/files/13369688/freebsd_13.1.zip)

vonericsen, 2023-12-07:
Marking this as closed since this code is in the latest release ([v23.12](https://github.com/Seagate/openSeaChest/releases/tag/v23.12)) and it has been merged to both master and develop branches.

If you have any other trouble, please reopen this issue or create a new one and we will do our best to resolve it!

closed 2023-12-07 by vonericsen

chrcoluk, 2024-02-16:
Thank you for working on this and I apologise I didnt respond.

The drive waking up might be caused by the SATA card I am using (not on board) as even basic smart commands will wake up the drives so its useful others confirmed that was unique to me.

I will grab the newer tools.  Thank you.