Seagate/openSeaChest #114: Failed to Restore the Max LBA

nick-hayhurst, 2023-05-24:
Hi All, 

I am attempting to work with a HPA on a Samsung (Dell EMC) ATA (SATA) SSD.

I am using openSeaTools on Ubuntu 22.04.

` openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 3.5.2-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Wed May 24 20:52:58 2023	User: root`

When trying to read the Native Max Address of this drive, it reports "Not Reported", see below Inquiry Log.

`/dev/sg6 - MZ7KH480HAHQ0D3 - S47GNA0M816130 - HF58 - ATA
	Model Number: MZ7KH480HAHQ0D3
	Serial Number: REDACTED SERIAL NUMBER
	Firmware Revision: HF58
	World Wide Name: REDACTED WWN
	Drive Capacity (GB/GiB): 480.10/447.13
	Temperature Data:
		Current Temperature (C): 31
		Highest Temperature (C): 55
		Lowest Temperature (C): 15
	Power On Time:  3 years 31 days 9 hours 
	Power On Hours: 27033.00
	MaxLBA: 937703087
	Native MaxLBA: Not Reported
	Logical Sector Size (B): 512
	Physical Sector Size (B): 4096
	Sector Alignment: 0
	Rotation Rate (RPM): SSD
	Form Factor: 2.5"
	Last DST information:
		Time since last DST (hours): 29.00
		DST Status/Result: 0x0
		DST Test run: 0x1
	Long Drive Self Test Time:  29 minutes 
	Interface speed:
		Max Speed (Gb/s): 6.0
		Negotiated Speed (Gb/s): 6.0
	Annualized Workload Rate (TB/yr): 186.18
	Total Bytes Read (TB): 358.50
	Total Bytes Written (TB): 216.03
	Encryption Support: Full Disk Encryption
	Cache Size (MiB): Not Reported
	Percentage Used Endurance Indicator (%): 4.00000
	Read Look-Ahead: Enabled
	Write Cache: Enabled
	SMART Status: Good
	ATA Security Information: Not Supported
	Firmware Download Support: Full, Segmented, Deferred
	Specifications Supported:
		ACS-4
		ACS-3
		ACS-2
		ATA8-ACS
		ATA/ATAPI-7
		ATA/ATAPI-6
		ATA/ATAPI-5
		ATA/ATAPI-4
		ATA-3
		ATA-2
		SATA 3.2
		SATA 3.1
		SATA 3.0
		SATA 2.6
		SATA 2.5
		SATA II: Extensions
		SATA 1.0a
[... 1022 lines cut ...]
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
	Return Task File Registers:
	[Error] = 04h
	[Count Ext] = 00h
	[Count] = 00h
	[LBA Lo Ext] = 00h
	[LBA Lo] = 00h
	[LBA Mid Ext] = 00h
	[LBA Mid] = 00h
	[LBA Hi Ext] = 00h
	[LBA Hi] = 00h
	[Device] = 40h
	[Status] = 51h

Command Time (us): 74.10

Read Native Max Address Ext returning: ABORTED

Failed to restore the max LBA!
`

Looking at the closing command result, which appears to be showing as "ABORTED".

Any ideas? :)

Swiss3003, 2023-05-25:
nick-hayhurst,
Thanks for the infomation. I'm going to have another engineer look at the information and the FDE spec to see about the native MaxLBA being not reported. I have a feeling the aborted command is do the FDE setting. We will try to get back to you late next week.
:)

DebabrataSTX, 2023-05-26:
Loos like the command is failing as the drive fails to report native Max LBA. 
If you see at the very last command issued, then you can clearly see that this is where the tool try to get max native lba by issuing Read Max Address Ext.

***************************************************
Sending ATA Read Native Max Address Ext
Sending SAT ATA Pass-Through Command:
	Protocol: NON-Data
	Data Direction: No Data
	Task File Registers:
	[FeatureExt] = 00h
	[Feature] = 00h
	[CountExt] = 00h
	[Count] = 00h
	[LBA Lo Ext] = 00h
	[LBA Lo] = 00h
	[LBA Mid Ext] = 00h
	[LBA Mid] = 00h
	[LBA Hi Ext] = 00h
	[LBA Hi] = 00h
	[DeviceHead] = 40h
	[Command] = 27h

  CDB:

        0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F  
  0x00 85 07 20 00 00 00 00 00 00 00 00 00 00 40 27 00 
*************************************************************

This command failed and returned command aborted 

******************************************************
Sending command with send_IO
SG IO Issued as Indirect IO
SG Masked Status = 01h - Check Condition
SG Driver Status = 08h - Driver Sense Data Available

  Sense Data Buffer:

        0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F  
  0x00 72 0B 00 00 00 00 00 0E 09 0C 01 04 00 00 00 00 
  0x10 00 00 00 00 40 51 

Sense Key: Bh = Aborted Command
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
	Return Task File Registers:
	[Error] = 04h
	[Count Ext] = 00h
	[Count] = 00h
	[LBA Lo Ext] = 00h
	[LBA Lo] = 00h
	[LBA Mid Ext] = 00h
	[LBA Mid] = 00h
	[LBA Hi Ext] = 00h
	[LBA Hi] = 00h
	[Device] = 40h
	[Status] = 51h

Command Time (us): 74.10

Read Native Max Address Ext returning: ABORTED

Failed to restore the max LBA!
*******************************************************

At this point we don't know what was the original max lba of the drive and thus we can not reset the drives max lba to the original one.

The failure of Read Max Address Ext (which is unexpected) is the main reason of the observed issue.
Can you let us now how the drive is connected to the system. Previously we have witnessed some command failure if a drive is connected through USB.

Swiss3003, 2023-05-31:
nick-hayhurst,
Looking through the FDE/TCG specs, DCO, Max Adrress will command abort until you disable the encryption and security features of the device.

nick-hayhurst, 2023-06-02:
> Loos like the command is failing as the drive fails to report native Max LBA. If you see at the very last command issued, then you can clearly see that this is where the tool try to get max native [...]

Hi! Thanks for the response!

I am connected by a SAS HBA (LSI) - other devices of similar types don't show this issue when tested or when issuing Read Max Address.

Let me know if you need more information!

nick-hayhurst, 2023-06-02:
> nick-hayhurst, Looking through the FDE/TCG specs, DCO, Max Adrress will command abort until you disable the encryption and security features of the device.

Interesting!

Is there a way to do so within openSeaChest? 

Thanks so much!

Swiss3003, 2023-06-05:
Well, SeaChest_Security has --revert method and --revertSP method. But those will revert the drive back to factory settings, at the same time all data on the device will be lost. Also you will need the PSID/admin password and your device must support those methods.  Once all security is disabled, then you should be able to set the max address of the drive. There are other tools out on GitHub that would get you more functionality for the security and would have more options to access user lBA's.

vonericsen, 2023-07-21:
I'm catching up on this issue.
I'm not sure exactly why this command would be aborted either, but I suspect it's due to some other feature being active.
I suspect it's some other security thing as @Swiss3003 and @DebabrataSTX  mentioned earlier.
The commit I pushed does add detection of the HPA security feature to the `-i` option which may help track this a little further, but I didn't see the bits for this feature reported in the identify data, so I doubt it is the cause but it might help for other devices that run into a similar issue.

ankitrslr, 2023-11-25:
It seems that the command is failing because the drive is not reporting the native Max LBA (Logical Block Address). The tool attempts to get the max native LBA by issuing the "Read Max Address Ext" command, but this command fails and returns "command aborted."

The last command issued is:

Sending ATA Read Native Max Address Ext
Sending SAT ATA Pass-Through Command:
Protocol: NON-Data
Data Direction: No Data
Task File Registers:
[FeatureExt] = 00h
[Feature] = 00h
[CountExt] = 00h
[Count] = 00h
[LBA Lo Ext] = 00h
[LBA Lo] = 00h
[LBA Mid Ext] = 00h
[LBA Mid] = 00h
[LBA Hi Ext] = 00h
[LBA Hi] = 00h
[DeviceHead] = 40h
[Command] = 27h

This command fails and returns "command aborted." The sense data indicates an aborted command with no additional sense information.

The failure to get the Max LBA is causing the observed issue, and it's mentioned that the drive is connected via a SAS HBA (LSI). Other devices of similar types don't exhibit this issue when tested.

In summary, the problem seems to be specific to this drive and its connection via the SAS HBA, as other similar devices connected in the same way do not encounter this problem

closed 2025-06-02 by vonericsen

fong15399, 2025-07-11:
I encountered a problem after trying to install UBUNTU on my SEAGATE flash drive with a capacity of 2TB, during the installation it took 2 hours and never finished so I restarted the device and unplugged the SEAGATE flash drive and after that my flash drive was no media and there was no available storage of 0 bytes. and I searched for information on Google and finally reflash it with MpTools from the site www.usbdev.ru. after reflashing with MpTools suddenly my storage was only 120GB even though the flash drive was 2TB SATA. 

can anyone help me recover the lost storage