Seagate/openSeaChest #118: Exos X16 fails to change sector size

danderson, 2023-07-11:
I'm doing initial setup on some new ST16000NM003G drives (Exos X16 16TB SATA). `openSeaChest_Format -d /dev/sdi --showSupportedFormats` says the drives support 4096b sectors, and are currently configured with 512b sectors. However, attempting to change the sector size fails with `Set Sector Configuration Ext returning: ABORTED`.

Hardware-wise, the drive is connected to a Supermicro SSG-5028R-E1CR12LA-CE010 server. The device chain from CPU to drive is:
 - X10SRH-CLN4F motherboard
 - LSI/BCM 3008 SAS3 HBA
 - BPN-SAS3-826EL1 SAS expander backplane (based on LSISASx28 expander IC)

Searching the issue tracker, I believe I'm seeing exactly the same symptoms as https://github.com/Seagate/openSeaChest/issues/79 , although possibly with slightly different hardware (X10 motherboard instead of X11, but also an LSI/BCM 3008 HBA, and also a supermicro server so likely similar backplane SAS expander).

I've attached the output of `openSeaChest_Info -d /dev/sdi -i`, `openSeaChest_Format -d /dev/sdi --showSupportedFormats`, and `openSeaChest_Format -d /dev/sdi --setSectorSize=4096 --confirm this-will-erase-data-and-may-render-the-drive-inoperable`.

[sdi-info.txt](https://github.com/Seagate/openSeaChest/files/12018144/sdi-info.txt)
[sdi-supportedformats.txt](https://github.com/Seagate/openSeaChest/files/12018143/sdi-supportedformats.txt)
[sdi-format.txt](https://github.com/Seagate/openSeaChest/files/12018145/sdi-format.txt)

The linked issue has a workaround (execute the sector reconfig from a different system without all the LSI, Supermicro and SAS<>SATA stuff in the chain), **so really I'm filing this issue to ask: is there any more data I could provide you to get to get more insight into this issue?** Given that I can apparently reproduce it, and I'm going to be doing destructive burn-in on these drives for a few days, I can run debug commands and invasive drive changes without harming data.

danderson, 2023-07-11:
Reproducing relevant info from #79, so people don't have to go digging: in that bug the reporter had a Supermicro X11DPH-T motherboard, and the same Supermicro AOC-S3008L-L8e HBA as me. No info on the backplane in that bug, but given Supermicro's product lineup, it seems likely that it's the same expander backplane as my system, since those boards don't change much even between different server models.

danderson, 2023-07-11:
One more datapoint: I moved one of the drives to an older Supermicro server with a SAS2 storage chain, and I was able to change the sector size there successfully. Listing the hardware in that server too, just in case the A/B datapoints help:
 - Motherboard: Supermicro X10SLM+-LN4F
 - HBA: Broadcom / LSI 9211-8i
 - Backplane: Supermicro BPN-SAS2-826EL1 (based on LSI SAS2X28 expander IC)

This server is a franken-machine assembled from a used chassis+backplane, motherboard and HBA. This is not a configuration sold by Supermicro directly (whereas the one in my original report, afaik, is).

vonericsen, 2023-07-20:
Hi @danderson,
Thanks for the logs, I will take a look and see if I find something else that might help track this down.
While debugging #79, I asked Seagate's engineer who works with Supermicro to test the Supermicro hardware we have and he could not repeat it. Seagate's engineer asked Supermicro's lab to also see if they could repeat this issue, but we never got it to repeat with the same hardware that was reported in that issue...so we really do not know what the issue is.

danderson, 2023-07-21:
Thanks for taking a look! I don't envy having to track this through all the layers to find where things are going wrong.

I filed this purely in case it provides additional clues, or if I can provide further data about the configuration that wasn't working. If that's not the case, then I'm happy to close this bug as there's only so much digging that's possible across multiple vendors like this.

vonericsen, 2023-07-21:
I reviewed the logs and I cannot figure out what would be wrong right now.
Everything is being populated in the command correctly according to the specifications.

I've asked to see if someone in Seagate's firmware group can help me understand the spec's abort reason "the device is unable to complete processing of the command" abort case to see if that can help me track it back to a feature interaction or something else in the firmware that I may be able to control.
The other cases for the command abort from the spec are not the issue since the fields are all being filled in properly (unless for some reason the HBA firmware is filtering them out on the bus, but you would need a bus trace to see this).

The only other thing I can think of while I dig backwards is have you tried updating the HBA firmware at all?
I'm not sure if it will fix it, but sometimes updating HBA firmware resolves odd things like this. 
In #111, updating the HBA firmware resolved a strange bug where the drive was not going into the idle or standby modes like it should. Maybe there is something similar going on here and causing the drive to think it cannot do the fast format right now because of some other bus activity from the HBA.

vonericsen, 2025-05-30:
I have pushed a major change that should help prevent issues when changing sector size.
I'm marking this fixed for now with this change that is part of [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

Please reopen this issue or open a new issue if this is still a problem.

antifuchs, 2025-08-20:
I have run into this same issue with v25.05 on Exos X24s in a very similar SuperServer and an [Inspur LSI SAS 9300-8i SAS3008](https://parts.theserverstore.com/products/inspur-lsi-sas-9300-8i-sas3008-8-port-12gb-pcie-3-0-it-mode-freenas-unraid-hbayzca-00424-101) controller, initially on firmware rev 16.00.10.00, then subsequently 16.00.12.00. Both versions resulted in these exact same errors ("Aborted" on the `ATA Set Sector Configuration Ext` command).

Not worth opening another issue I think, but I thought I'd provide more data points. This sounds to me like the 9300-8i isn't able to do this.

insunaa, 2026-01-06:
Hi @vonericsen ,

thank you for the reply. That makes sense! Is there some way to flash the proper firmware on this drive or do I just have to accept that the custom firmware has turned it into trash?

Feel free not to reply as this is now off topic, I apologize.

antifuchs, 2026-01-08:
So I just tried this same operation on a different LSI compatible controller on a different machine, with openSeaChest_Format utility version 3.5.1, and it worked! Here's the `lspci -vvv` output for the HBA, in case it helps:

```
1b:00.0 Serial Attached SCSI controller: Broadcom / LSI SAS3008 PCI-Express Fusion-MPT SAS-3 (rev 02)
    Subsystem: Super Micro Computer Inc AOC-S3008L-L8e
    Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx+
    Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
    Latency: 0, Cache Line Size: 32 bytes
    Interrupt: pin A routed to IRQ 305
    NUMA node: 0
    Region 0: I/O ports at 5000 [size=256]
    Region 1: Memory at aae40000 (64-bit, non-prefetchable) [size=64K]
    Region 3: Memory at aae00000 (64-bit, non-prefetchable) [size=256K]
    Expansion ROM at aad00000 [disabled] [size=1M]
    Capabilities: [50] Power Management version 3
        Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
        Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
    Capabilities: [68] Express (v2) Endpoint, IntMsgNum 0
        DevCap: MaxPayload 4096 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
            ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W TEE-IO-
        DevCtl: CorrErr- NonFatalErr- FatalErr+ UnsupReq-
            RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
            MaxPayload 256 bytes, MaxReadReq 512 bytes
        DevSta: CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
        LnkCap: Port #0, Speed 8GT/s, Width x8, ASPM not supported
            ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
        LnkCtl: ASPM Disabled; RCB 64 bytes, LnkDisable- CommClk+
            ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt- FltModeDis-
        LnkSta: Speed 8GT/s, Width x8
            TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
        DevCap2: Completion Timeout: Range BC, TimeoutDis+ NROPrPrP- LTR-
             10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
             EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
             FRS- TPHComp- ExtTPHComp-
             AtomicOpsCap: 32bit- 64bit- 128bitCAS-
        DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-
             AtomicOpsCtl: ReqEn-
             IDOReq- IDOCompl- LTR- EmergencyPowerReductionReq-
             10BitTagReq- OBFF Disabled, EETLPPrefixBlk-
        LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
        LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
             Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
             Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
        LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete+ EqualizationPhase1+
             EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
             Retimer- 2Retimers- CrosslinkRes: unsupported, FltMode-
    Capabilities: [d0] Vital Product Data
        Not readable
    Capabilities: [a8] MSI: Enable- Count=1/1 Maskable+ 64bit+
        Address: 0000000000000000  Data: 0000
        Masking: 00000000  Pending: 00000000
    Capabilities: [c0] MSI-X: Enable+ Count=96 Masked-
        Vector table: BAR=1 offset=0000e000
        PBA: BAR=1 offset=0000f000
    Capabilities: [100 v2] Advanced Error Reporting
        UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP-
            ECRC- UnsupReq- ACSViol- UncorrIntErr- BlockedTLP- AtomicOpBlocked- TLPBlockedErr-
            PoisonTLPBlocked- DMWrReqBlocked- IDECheck- MisIDETLP- PCRC_CHECK- TLPXlatBlocked-
        UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP-
            ECRC- UnsupReq+ ACSViol- UncorrIntErr- BlockedTLP- AtomicOpBlocked- TLPBlockedErr-
            PoisonTLPBlocked- DMWrReqBlocked- IDECheck- MisIDETLP- PCRC_CHECK- TLPXlatBlocked-
        UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+
            ECRC- UnsupReq- ACSViol- UncorrIntErr+ BlockedTLP- AtomicOpBlocked- TLPBlockedErr-
            PoisonTLPBlocked- DMWrReqBlocked- IDECheck- MisIDETLP- PCRC_CHECK- TLPXlatBlocked-
        CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+ CorrIntErr- HeaderOF-
        CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+ CorrIntErr- HeaderOF-
        AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
            MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
        HeaderLog: 04000001 1710000f 1b080000 99a17b38
    Capabilities: [1e0 v1] Secondary PCI Express
        LnkCtl3: LnkEquIntrruptEn- PerformEqu-
        LaneErrStat: 0
    Capabilities: [1c0 v1] Power Budgeting <?>
    Capabilities: [190 v1] Dynamic Power Allocation <?>
    Capabilities: [150 v1] Single Root I/O Virtualization (SR-IOV)
        IOVCap: Migration- 10BitTagReq- IntMsgNum 0
        IOVCtl: Enable- Migration- Interrupt- MSE- ARIHierarchy+ 10BitTagReq-
        IOVSta: Migration-
        Initial VFs: 16, Total VFs: 16, Number of VFs: 0, Function Dependency Link: 00
        VF offset: 1, stride: 1, Device ID: 0097
        Supported Page Size: 00000553, System Page Size: 00000001
        Region 0: Memory at 0000000000000000 (64-bit, non-prefetchable)
        Region 2: Memory at 0000000000000000 (64-bit, non-prefetchable)
        VF Migration: offset: 00000000, BIR: 0
    Capabilities: [148 v1] Alternative Routing-ID Interpretation (ARI)
        ARICap: MFVC- ACS-, Next Function: 0
        ARICtl: MFVC- ACS-, Function Group: 0
    Kernel driver in use: mpt3sas
    Kernel modules: mpt3sas
```

ascendbeing, 2026-10-09:
To the reply above: There is a 16.00.16.00 firmware. H3C released it, iirc. I am successfully running on all our 9300 8i and 9340 8i units.