Seagate/openSeaChest #119: tracking "unkown command"

theedge456, 2023-07-17:
Hello,
I don't know if this is the correct place to post this issue.
If not, mzy someone direct me to the correct one.

I have a ST1000DM010 handled by a gigabyte motherboard a320ma-m.2.
The OS is devuan chimera (based on debian bullseye), using kernel 6.1.37 from kernel.org.
I'm tracking a problem displayed in the logs because the disk is rarely used.
It is only used to perform compilation in multicore mode.

 ```
kernel: [  750.902215] sd 5:0:0:0: [sda] tag#2 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=5s
 kernel: [  750.905013] sd 5:0:0:0: [sda] tag#2 Sense Key : Illegal Request [current] 
 kernel: [  750.907809] sd 5:0:0:0: [sda] tag#2 Add. Sense: Unaligned write command
 kernel: [  750.910572] sd 5:0:0:0: [sda] tag#2 CDB: Read(10) 28 00 12 c2 4c 80 00 00 38 00
 kernel: [  750.916106] sd 5:0:0:0: [sda] tag#22 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=5s
 kernel: [  750.918889] sd 5:0:0:0: [sda] tag#22 Sense Key : Illegal Request [current] 
 kernel: [  750.921675] sd 5:0:0:0: [sda] tag#22 Add. Sense: Unaligned write command
 kernel: [  750.924462] sd 5:0:0:0: [sda] tag#22 CDB: Read(10) 28 00 17 14 8d 00 00 00 88 00
 kernel: [  750.930065] ata6: EH complete
```

smartctl shows:
```
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000f   100   063   006    Pre-fail  Always       -       811937
  3 Spin_Up_Time            0x0003   099   097   000    Pre-fail  Always       -       0
  4 Start_Stop_Count        0x0032   099   099   020    Old_age   Always       -       1397
  5 Reallocated_Sector_Ct   0x0033   100   100   010    Pre-fail  Always       -       0
  7 Seek_Error_Rate         0x000f   077   060   045    Pre-fail  Always       -       56586838
  9 Power_On_Hours          0x0032   100   100   000    Old_age   Always       -       599
 10 Spin_Retry_Count        0x0013   100   100   097    Pre-fail  Always       -       0
 12 Power_Cycle_Count       0x0032   099   099   020    Old_age   Always       -       1399
183 Runtime_Bad_Block       0x0032   086   086   000    Old_age   Always       -       14
184 End-to-End_Error        0x0032   100   100   099    Old_age   Always       -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
188 Command_Timeout         0x0032   100   100   000    Old_age   Always       -       0 0 0
189 High_Fly_Writes         0x003a   100   100   000    Old_age   Always       -       0
190 Airflow_Temperature_Cel 0x0022   066   057   040    Old_age   Always       -       34 (Min/Max 34/34)
193 Load_Cycle_Count        0x0032   100   100   000    Old_age   Always       -       1398
194 Temperature_Celsius     0x0022   034   007   000    Old_age   Always       -       34 (0 7 0 0 0)
195 Hardware_ECC_Recovered  0x001a   100   001   000    Old_age   Always       -       811937
197 Current_Pending_Sector  0x0012   100   100   000    Old_age   Always       -       0
198 Offline_Uncorrectable   0x0010   100   100   000    Old_age   Offline      -       0
199 UDMA_CRC_Error_Count    0x003e   200   200   000    Old_age   Always       -       0
240 Head_Flying_Hours       0x0000   100   253   000    Old_age   Offline      -       589h+00m+00.656s
241 Total_LBAs_Written      0x0000   100   253   000    Old_age   Offline      -       21343524455
242 Total_LBAs_Read         0x0000   100   253   000    Old_age   Offline      -       10832830779

```
I compiled the last version of openSeaChest (2.2.1-6_0_1 X86_64).

I started a long test with this command:
sudo ./openSeaChest_GenericTests  -d /dev/sg1 --longGeneric

The result showed no errors. I only saw messages about "unknown command".
Is there a way to display these messages ?

I saw on the seagate website that there was an updated firmware for ST1000DM series (ST1000DM004 or ST1000DM007) but not the ST1000DM010, still driven by the CC43 firmware.
May I try to update the firmware ?
Any other hint ?

-
Fabien

theedge456, 2023-07-17:
Forget this message. Everything went back to normal when I connected the disk to other data and power supply ports  on the motherboard.

vonericsen, 2023-07-18:
Hi @theedge456,

Just to confirm, the drive is working properly with new cabling and this message no longer shows in the system logs?

theedge456, 2023-07-19:
Yes. Lucky me
-
Fabien

vonericsen, 2023-07-19:
Thanks for confirming that!
I have seen cabling issues before, but not something that shows up like this. A lot of times the attribute 199 will start to increase when there is a problem and there are other symptoms but generally not "unaligned write command", so that is really weird. Cabling issues always have odd symptoms, but this is not one I've seen before.

If it happens again, please update this issue or create a new one and we can see if we can dig a little deeper to find out more information behind the cause, but sometimes it is as simple as replacing a flaky cable.
I'm going to mark this closed for now, but please reopen it if you need to.

theedge456, 2023-07-20:
I generated this file at that time.
Tell me if it helps.

[HD_statistics.zip](https://github.com/Seagate/openSeaChest/files/12103903/HD_statistics.zip)

vonericsen, 2023-07-21:
@theedge456, 
I took a look at the file and didn't see anything that would be an indication of a cabling issue like I would expect.

If you run into this again, capturing the device statistics and SMART attributes will likely be most helpful.
`openSeaChest_SMART -d <handle> --smartAttributes analyzed --deviceStatistics > debug.txt`

Another thing that might help is dumping the SATA phy event counters. This is not part of openSeaChest yet, but it is present in smartctl. It is possible whatever is happening was logged there was well.

theedge456, 2023-07-23:
@vonericsen,
This is the result of
 `sg_sat_phy_event --ck_cond --verbose 1>stdout.log 2>stderr.log`

Tell me if it helps
[ata_phy_event.zip](https://github.com/Seagate/openSeaChest/files/12138381/ata_phy_event.zip)

theedge456, 2024-12-21:
I noticed that my CMOS battery voltage was under 3V.
Not sure if it has any connection to the problem.

vonericsen, 2024-12-30:
Hi @theedge456,

I'm sorry I completely forgot to respond to this after you attached the phy event counters earlier!

Reviewing that again there is nothing in there to indicate an error related to this issue.

> I noticed that my CMOS battery voltage was below 3V.
Not sure if it has any connection to the problem.

I doubt this is related to this issue. 

Reviewing the original information you shared, I see it is reading an LBA when this error occurs, which is weird to report "unaligned write command" since it's a read, not a write.
I have seen similar issues from the low-level linux libata on occasion but I have not pinpointed the exact cause of this.
What libata does is it provides the SCSI to ATA translation so that the linux kernel handles all drives as "SCSI" devices.
This is specified in a spec by the T10 committee and I am very familiar with it.
In part of the spec is a table of how to translated ATA errors into an appropriate SCSI error.
The LibATA code follows this but reuses code from an HBA driver (I cannot remember which one exactly) and I think that part of the code may be triggering this error accidentally.

In ATA there is a bit in the error output to notify the host (controller/OS) that a command was not aligned correctly. But that only applies on writes. 
The alignment error bit replaces an obsolete bit called "Corrected data" from an older specification.

My current guess is that your drive completed the read, set the "corrected data" bit, but since it is an old bit and the error translation code likely does not handle this it translated it to an unaligned write command which is the appropriate error on more modern drives.

Can you share the output of `openSeaChest_SMART -d <handle> -i`?
This will list all the specifications this drive says it conforms to both new and old which may help determine if this is the cause or not.

theedge456, 2024-12-31:
Hi @vonericsen,
Here is the result of the command:
```
./openSeaChest_SMART -d /dev/sda -i
==========================================================================================
 openSeaChest_SMART - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_SMART Version: 2.3.2-6_0_1 X86_64
 Build Date: Jul 16 2023
 Today: Tue Dec 31 09:37:19 2024	User: root
==========================================================================================

/dev/sg1 - ST1000DM010-2EP102 - Z9AX12SE - CC43 - ATA
	Model Number: ST1000DM010-2EP102
	Serial Number: Z9AX12SE
	Firmware Revision: CC43
	World Wide Name: 5000C500B284FDBA
	Drive Capacity (TB/GiB): 1.00/931.51
	Native Drive Capacity (TB/GiB): 1.00/931.51
	Temperature Data:
		Current Temperature (C): 24
		Highest Temperature (C): 43
		Lowest Temperature (C): 16
	Power On Time:  30 days 11 hours 
	Power On Hours: 731.00
	MaxLBA: 1953525167
	Native MaxLBA: 1953525167
	Logical Sector Size (B): 512
	Physical Sector Size (B): 4096
	Sector Alignment: 0
	Rotation Rate (RPM): 7200
	Form Factor: 3.5"
	Last DST information:
		Time since last DST (hours): 44.00
		DST Status/Result: 0x0
		DST Test run: 0x1
	Long Drive Self Test Time:  1 hour 40 minutes 
	Interface speed:
		Max Speed (Gb/s): 6.0
		Negotiated Speed (Gb/s): 1.5
	Annualized Workload Rate (TB/yr): 245.14
	Total Bytes Read (TB): 6.72
	Total Bytes Written (TB): 13.74
	Encryption Support: Not Supported
	Cache Size (MiB): 64.00
	Read Look-Ahead: Enabled
	Write Cache: Enabled
	Low Current Spinup: Disabled
	SMART Status: Good
	ATA Security Information: Supported
	Firmware Download Support: Full, Segmented, Deferred
	Specifications Supported:
		ATA8-ACS
		ATA/ATAPI-7
		ATA/ATAPI-6
		ATA/ATAPI-5
		ATA/ATAPI-4
		SATA 3.0
	Features Supported:
		SATA NCQ
		SATA Software Settings Preservation [Enabled]
		SATA Device Initiated Power Management
		HPA
		Power Management
		Security
		SMART [Enabled]
		DCO
		48bit Address
		PUIS
		APM [Enabled]
		GPL
		SMART Self-Test
		SMART Error Logging
		Write-Read-Verify
		SCT Write Same
		Host Logging
	Adapter Information:
		Adapter Type: PCI
		Vendor ID: 1022h
		Product ID: 43B8h
		Revision: 0002h
```

vonericsen, 2025-06-04:
@theedge456,

That is good to know!
It is possible that the BIOS update or OS update could have fixed it (Maybe CMOS battery, but very unlikely).
I have definitely run into a BIOS update or two that corrected a problem I was having and never being able to repeat it again...and OS updates fix lots of things so that is also likely what helped.

One issue I ran into early in my time at Seagate was resolved by a BIOS update, but it was strange what I observed.
It was very slow performance in Windows XP with SeaTools.
After a lot of debugging going nowhere I checked for system updates and the BIOS had some SATA fix in it.
Flash it and....no change.

Then I reviewed the BIOS settings and tried changing something just to test and then the issue was gone....so I changed it back....and it was still gone. I could never get it to repeat again no matter what settings I changed.
The only thing I could come up with was that there was some BIOS saved memory somewhere that needed a rewrite to correct this issue. I have since tried to always change something and change it back to force a resave of settings to prevent this kind of thing again, even in my personal computers. haha.

Anyways, I think I will mark this closed for now, but if you see this issue again or any other similar issue do not hesitate to reopen this issue or create a new one so we can look into it.

closed 2025-06-04 by vonericsen