Seagate/openSeaChest #120: Prebuilt EFI binaries?

achelon5, 2023-08-29:
Hi,

I know you don't support UEFI binaries, but is there any chance you could put some pre-built ones on the website? I tried to follow the guide to build them but I couldn't make it work.

Regards,
Richard

vonericsen, 2023-08-30:
Hi @achelon5,

It has been a few years since we've built them, but we can try looking into getting it going again.
I don't know when we will get to it though, so it may take a while to get completed due to other projects we are currently working on.

Can you share which issues you ran into in the mean time?
It would not surprise me if the build files need updating or there is a C code change necessary after all this time since they were last built.

achelon5, 2023-08-31:
Hi,

I've reflected on this request and I think what I want to be able to achieve is better done using the linux compatible provided code and some sort of custom minimal build using buildroot. I think it is less work all round and more likely to work.

Regards,
Richard

vonericsen, 2023-08-31:
Sounds good.
I'll leave this open. I think it would be good to revisit when we have some time.

From my experience with working in UEFI, it was a bit complicated to get working and unpredicatable what systems would contain all the necessary drivers or just not crash.
Things may have changed since I did that work a few years ago and may have better support for both ATA passthrough and NVMe passthrough, but one thing I had to do was build and load that driver manually on some systems. 
Some systems hung trying to execute and I couldn't find the reason for that issue.
Some of the UEFI secure boot systems from that time also failed to verify the chain of trust. They would verify the Microsoft signed shim, but failed to boot anything else despite containing our trusted signature on those other EFI binaries (rEFInd, rEFInd drivers, Linux kernel). This was not a majority of systems, but it was surprising how many had this issue.

We were investigating UEFI shell tools as a solution for a customer when we ran into these issues, and it was easier for us to continue using Linux since it has great support for most hardware and worked for them across all the same systems we were testing and had different issues.

If anyone else comes across this issue and is interesting in prebuilt UEFI binaries, please add a comment so we know how many people want these available.