Seagate/openSeaChest #121: Firmware update failing - Firmware Download Failed - Download Microcode returning: ABORTED

nachopants, 2023-09-20:
Firmware upgrade failure, with segmented download variable it says 'successfully upgraded to SN03' (which is current firmware)

Block size failure as well.

Firmware upgrade from SN03 to SN04

Linux nachoServ 4.4.302+ #64570 SMP Thu Jul 20 00:07:27 CST 2023 x86_64 GNU/Linux synology_broadwell_3617xs

I have tried with various versions of SeaChest and SeaChest lite.

It also fails to change block size from 512 to 4096.

Consistent error across 8 x 16tb exos 18's

ash-4.4# ./openSeaChest_Firmware -d /dev/sg6 --downloadFW EvansBPExosX18SATA-STD-512E-SN04.LOD --downloadMode segmented
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Wed Sep 20 15:43:32 2023        User: root
==========================================================================================

/dev/sg6 - ST16000NM000J-2TW103 - ZR70SWR4 - SN03 - ATA
......
Firmware Download successful
Firmware Download time (ms): 162.77
Average time/segment  (ms): 1.64
Activate Time          (ms): 90.97
New firmware version is SN03

[https://pastebin.com/wSGV7krx](url)

Swiss3003, 2023-09-20:
> This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Nachopants,
The first thing is to power cycle the drive. Sounds like the it did a differed download to the drive and the firmware will not update until the drive is power cycled. IF, I remember correctly "New firmware version is SN03" reports what the device reports after the download. If it is a differed download, well it will not update until it updates after the next power cycle.

after the power cycle check the firmware version by doing a -i  on that device.

If it has not updated, then we need to check to see if the firmware is correct for your drive. Quick look it should be correct. But if the firmware does not match the drive will reject the firmware on the next power cycle.

For the format:
run a --showSupportedFormats and Please send that output.

next run the --setSectorSize 4096 --confirmThis-may-erase-data -v3 and save all the output and send it to us. it will help with the debug.

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:40a82c92-95f2-4789-901f-3bf0712affa5]

Seagate Internal

nachopants, 2023-09-20:
Hi Tim,

First of all thanks for your time in assisting with this. I must say all indications to a layman look as though the firmware is incorrect or being blocked despite the fact I've downloaded it from the Seagate website having tried a range of my SN03 -> SN04 serial numbers. I've tried with the bundled SeaChest binary, the latest SeaChest binary from Seagate and now the openSeaChest suite so far.

I have given up on the firmware for now, am hopeful you can point me in the right direction given the above.

Firstly, I'd like to make clear this is the current suite of drives

https://pastebin.com/ECertmi2

My focus for this exercise is /dev/sg2 and /dev/sg3

In attempting to upgrade the firmware once again, it was not successful, and yes have power cycled the drive etc.

https://pastebin.com/VXCDU06b

I decided to revisit the firmware later and for now push /dev/sg2 and /dev/sg3 to 4096 from 512.

/dev/sg2 appeared to be all good https://pastebin.com/KR9FY78G

/dev/sg3 does not... look good at all https://pastebin.com/iMFbCWEG

It now seemingly looks as though /dev/sg3 is unreadable entirely https://pastebin.com/4mMKbpm5

So now our mission is to recover /dev/sg3 so both sg2 & sg3 are stable at 4096.

Then re-visit why this firmware update isn't working...

nachopants, 2023-09-20:
-rwx------+ 1 synology users  384 Oct  4  2022 ENBP-SN04.CFS
-rwx------+ 1 synology users 2.6M Oct  3  2022 EvansBPExosX18SATA-STD-512E-SN04.LOD
-rwx------+ 1 synology users 601K Mar 28 08:16 openSeaChest_Erase
-rwx------+ 1 synology users 505K Mar 28 08:16 openSeaChest_Firmware
-rwx------+ 1 synology users 578K Mar 28 08:16 openSeaChest_Format

nachopants, 2023-09-21:
https://pastebin.com/bwGKgvie

Also this outlines the successfully modified drive and the failed drive.

Unsure on next steps!

DebabrataSTX, 2023-09-21:
@nachopants I could not access to the link you have provided. It will be much better if you post it somewhere more accessible or attach it here. 
Also if you add "-v 4" to the command. This will provide us more info that will be helpful to debug the issue.

Swiss3003, 2023-09-21:
Nochopants,
Let's start with a list of the serial numbers of the drives. I'll get someone to look / hunt for firmware to upgrade to.

Second, Looks like our firewall will not let me (IT issue) see any of the shares you did. Can you copy the text into the email. painful I know, but that way I know the information will come through and not get blocked by IT.

Also can you give me a breakdown on the system. OS, cards, RAID anything else that might be in the system that we should know about. Seems like the OS passthrough is working for you, because one drive did update. But I just would like to know what we are dealing with. Also are all the drive sata or are some of them sas?

I'll do some new builds on SeaChest to update your tools. Once I know the OS. I'll get you some new builds to make sure you are up to date.
Erase, format, firmware.  Do you need any other tools?

Format changes:
here are a few things to look at when changing the sector sizes.
Background tasks - If a drive has or needs to perform background tasks the change and the format can take a long time for the big drives.
Let the tool finish the task. Any hard reset or power cycle can brick the drive during this process.
MBR - sometimes I have found that a MBR or a partition on the drive will cause the drive to not change the sector size.
           - I found that erasing the drive, or writing 00's to the MBR ( LBA 0 -1000) and to MaxLBA (maxlba-1000) helps

          SeaChest_Basics -d /dev/sg<#> --displayLBA 1000
  SeaChest_Basics -d /dev/sg<#> --overwrite 0
  SeaChest_Basics -d /dev/sg<#> --overwrite 1000 --overwriteRange 2000

[image]

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:fc606380-b1c3-46f4-9add-c72297284824]

Seagate Internal

https://pastebin.com/bwGKgvie

nachopants, 2023-09-21:
Hi team,

As a first step here are all of the debug links above in order...

https://seagate-debug.s3.ap-southeast-2.amazonaws.com/debug1.txt
https://seagate-debug.s3.ap-southeast-2.amazonaws.com/debug2.txt
https://seagate-debug.s3.ap-southeast-2.amazonaws.com/debug3.txt
https://seagate-debug.s3.ap-southeast-2.amazonaws.com/debug4.txt
https://seagate-debug.s3.ap-southeast-2.amazonaws.com/debug5.txt
https://seagate-debug.s3.ap-southeast-2.amazonaws.com/debug6.txt
https://seagate-debug.s3.ap-southeast-2.amazonaws.com/debug7.txt

I hope you can open these?

nachopants, 2023-09-21:
ash-4.4# ./openSeaChest_Basics -s
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 3.5.2-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Fri Sep 22 09:27:16 2023        User: root
==========================================================================================
Vendor   Handle       Model Number            Serial Number          FwRev     
ATA      /dev/sg0     ST3000VN007-2E4166      Z6A01TKZ               SC60      
ATA      /dev/sg1     ST3000VN007-2AH16M      ZDH6DN24               SC60      
ATA      /dev/sg2     ST3000VN007-2E4166      Z6A01CHP               SC60      
ATA      /dev/sg3     ST3000VN007-2E4166      Z6A01RZJ               SC60      
ATA      /dev/sg4     ST16000NM000J-2TW103    ZR60EBCQ               SN03      
ATA      /dev/sg5     ST16000NM000J-2TW103    ZR60EBCH               SN03      
ATA      /dev/sg6     ST16000NM000J-2TW103    ZR70SWR4               SN03      
         /dev/sg7                                                              
ATA      /dev/sg8     ST16000NM000J-2TW103    ZR60595N               SN03      
ATA      /dev/sg9     ST16000NM000J-2TW103    ZRS0J0P8               SN03      
ATA      /dev/sg10    ST16000NM000J-2TW103    ZR505EW1               SN03      
ATA      /dev/sg11    ST16000NM000J-2TW103    ZR60595Q               SN03      
MARVELL  /dev/sg12    Virtual Device          V Device               1.00      
Synology /dev/sg13    DiskStation             615609077010           PMAP      

ash-4.4# ./openSeaChest_Basics -d /dev/sg7 --displayLBA 1000
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 3.5.2-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Fri Sep 22 09:27:19 2023        User: root
==========================================================================================

/dev/sg7 -  -  -  - SCSI
Could not allocate memory to read LBA.: No such device or address

ash-4.4# ./openSeaChest_Basics -d /dev/sg7 --overwrite 0 --confirm this-will-erase-data
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 3.5.2-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Fri Sep 22 09:27:45 2023        User: root
==========================================================================================

/dev/sg7 -  -  -  - SCSI
Floating point exception (core dumped)

ash-4.4# ./openSeaChest_Basics -d /dev/sg7 --overwrite 1000 --overwriteRange 2000 --confirm this-will-erase-data
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 3.5.2-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Fri Sep 22 09:28:07 2023        User: root
==========================================================================================

/dev/sg7 -  -  -  - SCSI
Floating point exception (core dumped)

Swiss3003, 2023-09-22:
So, I have verified that the *.LOD file that you have for updating the firmware is the correct one and that it will update on a similar drive.  All 7 ST16000NM000J-2TW103 drives should be able to update to the SN04 code. I'll look into some more options to try in order to get the drive to upgrade to the new firmware.

Also, We are able to see the debug output now. We see that the devices are command aborting the set sector configuration ext command. Right now, it's hard to say what is going on. But it might be the drive stopping us from changing the sector size. Now we have to figure out why it might stop us.  We will have to start looking at the identify information you sent us and see if something got disabled on the drive. This will take some time to figure out.  I have a few people working on it now. I'll talk to two other guys on Monday when they come into the office.

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:be4800a1-0b2c-4287-9476-821c544c610c]

Seagate Internal

Swiss3003, 2023-09-22:
Core dump failures - wow that's not good. First time I've seen that on Seachest basic. I'll get someone to look into that.

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:dd1164a5-ee2a-4b61-ab0c-12b0a732daee]

Seagate Internal

Swiss3003, 2023-09-22:
I didn't see which OS you have. So I will give you a few new builds.

Cent OS 7
Unbuntu 20.04
ALPINE

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:68732b53-543b-43d5-9c7b-0b4e2b17b9ba]

Seagate Internal

nachopants, 2023-09-23:
ash-4.4# uname -a

Linux nachoServ 4.4.302+ #64570 SMP Thu Jul 20 00:07:27 CST 2023 x86_64 GNU/Linux synology_broadwell_3617xs

To confirm, I get 'firmware update failed' on ALL of those drives unless I set downloadMode to segmented, in which case it says 'firmware successfully update to SN03'

It's really strange, it's like there is a setting on the drive refusing it from accepting firmware updates.

This is the Synology UI. It's showing:

- Drive 7 has successfully updated to 4kn but cannot update firmware to SNO4
- Drive 8 appears to have died entirely as per the discussions above
- I'm too scared to change any of the other drives for now so they're all still 512b SNO3

I have tried updating firmware to SN04 on maybe 5 of the drives? None are successful

[https://i.imgur.com/lafbV1L.png](url)

nachopants, 2023-09-25:
`ash-4.4# ./openSeaChest_Basics -s
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 3.5.2-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 25 12:46:05 2023        User: root
==========================================================================================
Vendor   Handle       Model Number            Serial Number          FwRev     
ATA      /dev/sg0     ST3000VN007-2E4166      Z6A01TKZ               SC60      
ATA      /dev/sg1     ST3000VN007-2AH16M      ZDH6DN24               SC60      
ATA      /dev/sg2     ST3000VN007-2E4166      Z6A01CHP               SC60      
ATA      /dev/sg3     ST3000VN007-2E4166      Z6A01RZJ               SC60      
ATA      /dev/sg4     ST16000NM000J-2TW103    ZR60EBCQ               SN03      
ATA      /dev/sg5     ST16000NM000J-2TW103    ZR60EBCH               SN03      
ATA      /dev/sg6     ST16000NM000J-2TW103    ZR70SWR4               SN03      
         /dev/sg7                                                              
ATA      /dev/sg8     ST16000NM000J-2TW103    ZR60595N               SN03      
ATA      /dev/sg9     ST16000NM000J-2TW103    ZRS0J0P8               SN03      
ATA      /dev/sg10    ST16000NM000J-2TW103    ZR505EW1               SN03      
ATA      /dev/sg11    ST16000NM000J-2TW103    ZR60595Q               SN03      
MARVELL  /dev/sg12    Virtual Device          V Device               1.00      
Synology /dev/sg13    DiskStation             615609077010           PMAP      

ash-4.4# ./openSeaChest_Firmware --downloadFW EvansBPExosX18SATA-STD-512E-SN04.LOD -d /dev/sg4
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 25 12:47:15 2023        User: root
==========================================================================================

/dev/sg4 - ST16000NM000J-2TW103 - ZR60EBCQ - SN03 - ATA
......
Firmware Download failed

ash-4.4# ./openSeaChest_Firmware --downloadFW EvansBPExosX18SATA-STD-512E-SN04.LOD -d /dev/sg8
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 25 12:47:20 2023        User: root
==========================================================================================

/dev/sg8 - ST16000NM000J-2TW103 - ZR60595N - SN03 - ATA
......
Firmware Download failed

ash-4.4# ./openSeaChest_Firmware --downloadFW EvansBPExosX18SATA-STD-512E-SN04.LOD -d /dev/sg9
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 25 12:47:21 2023        User: root
==========================================================================================

/dev/sg9 - ST16000NM000J-2TW103 - ZRS0J0P8 - SN03 - ATA
......
Firmware Download failed

ash-4.4# ./openSeaChest_Firmware --downloadFW EvansBPExosX18SATA-STD-512E-SN04.LOD -d /dev/sg10
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 25 12:47:24 2023        User: root
==========================================================================================

/dev/sg10 - ST16000NM000J-2TW103 - ZR505EW1 - SN03 - ATA
......
Firmware Download failed

ash-4.4# ./openSeaChest_Firmware --downloadFW EvansBPExosX18SATA-STD-512E-SN04.LOD -d /dev/sg11
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 25 12:47:26 2023        User: root
==========================================================================================

/dev/sg11 - ST16000NM000J-2TW103 - ZR60595Q - SN03 - ATA
......
Firmware Download failed

ash-4.4# `

Swiss3003, 2023-09-27:
Tim Gilmer ***@***.***) has sent you a protected message.
                                          Read the message            Learn about messages protected by Microsoft Purview Message Encryption.        
    
    
    
        
           Privacy Statement        
    
    
        
              Learn More on email encryption.             Microsoft Corporation, One Microsoft Way, Redmond, WA 98052

nachopants, 2023-09-28:
Hi Tim,

Sorry mate that reply is corrupt somehow or unreadable

Swiss3003, 2023-09-28:
second try with out the firmware file.

Sorry for not getting back so quickly. Got stuck with a build that was failing and causing the CLI to fail across the company. Fun filled 18 hr day. 🙂

So I looked up a few of those drives. They were all big drives over 2TB making them 4096 sector native drives. BUT they are emulating 512 sector size. Therefore, you should be able to get the drive to run in 4096. Fully believe that to be true. I looked up the controller, but it's not on my list for cards that have issues with the passthrough. Meaning we haven't had issue "YET". hoping we can figure it out.

So, I downloaded the firmware file and testing it on a "much older" drive, and it worked. Went from SN02 to SN04 (intel chipset and Ubuntu 20.04)  I'm attaching the file to this email. It is the firmware for the 512 Emulation. IF you can try it in a different system with no raid, intel chipset works best.  If it works there... then we know it's the controller and the firmware file is fine.

Now, for this chipset.    - Segment download works BUT fails to change the code.
so add --downloadMode  segmented --fwdlSegSize 512  -v3
Then try forcing the ATA   to DMA , PIO, UDMA.
      --forceATA
            Using this option will force the current drive to
            be treated as a ATA drive. Only ATA commands will
            be used to talk to the drive.

      --forceATADMA     (SATA Only)
            Using this option will force the tool to issue SAT
            commands to ATA device using the protocol set to DMA
            whenever possible (on DMA commands).
            This option can be combined with --forceATA

      --forceATAPIO     (SATA Only)
            Using this option will force the tool to issue PIO
            commands to ATA device when possible. This option can
            be combined with --forceATA

      --forceATAUDMA    (SATA Only)
            Using this option will force the tool to issue SAT
            commands to ATA device using the protocol set to UDMA
            whenever possible (on DMA commands).
            This option can be combined with --forceATA

So I don't think the deferred and deffered+activate will work. They always try to force 4096 for the block size. But maybe if you force DMA.PIO,UDMA with that option it might work also.  Don't forget to send the output. It helps a lot. Plus I can pass those on to other people to look at.

Good luck.

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:b69a86b3-6d81-43e5-a44c-00a9a64fa2c2]

Seagate Internal

nachopants, 2023-09-28:
Hi Tim,

Thanks so much for this!!

I can confirm I have upgraded all of the drives successfully to 4096 and they’re all reporting as 4K native.

I’m scared to ‘guess’ at the firmware as I’ve already ‘bricked’ and returned that 1 failed drive.

Now that they’re all 4K native, what command should I execute to attempt to do the firmware upgrade in isolation?

The Synology is using an Intel Xeon E3-1230 v2 cpu

Thanks mate,
Andrew

Swiss3003, 2023-09-28:
Good, I'm happy they all made it to 4k. That  was the hard part.

The firmware should not brick the drive. The drive should check the signature and the internal flags before performing the update on the firmware. So where you able to get the file I sent you with the firmware on it?

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:e4951e8b-3a33-4594-b3c0-aaf659dd64ad]

Seagate Internal

nachopants, 2023-10-04:
Hello mate,

no sorry there is no firmware file here. I think because you’re replying via email. I think if you open the GitHub issue itself you can see the post was corrupted

if you could please attach the firmware or provide a URL, and the exact command to run, I will attempt an upgrade

nachopants, 2023-10-10:
Hi Tim,

Just wondering if you figured out a way to provide the firmware binary as yet?

nachopants, 2023-10-10:
This continues to be the problem.

https://pastebin.com/LpyVSfA5

Swiss3003, 2023-10-18:
So far, No. I think it's our IT stopping the attachment. I'm out of town on a mission to help my mother for a week. Monday I should be back in the office where I can try again.

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:9eaeced2-98ec-47b1-bc69-2b13cf531bd6]

Seagate Internal

Swiss3003, 2023-10-18:
Link seems to be a broken. Can you resend it?

Tim Gilmer
Staff Engineer
Field Diags
Office: (720)-684-2624
Seagate Technology
[cid:894b00d7-79a3-4142-800f-2a9f272bcbf8]

Seagate Internal

https://pastebin.com/LpyVSfA5

nachopants, 2023-10-18:
Hi Tim,

I think it will be easier if you visit the github issue directly, all links seem fine here.

https://github.com/Seagate/openSeaChest/issues/121

Swiss3003, 2023-10-18:
Good morning, even in Github the links go to a broken link.
[image]

Put just to ask did you ever do the 
--downloadMode  segmented --fwdlSegSize 512  -v3 
and did you do any of the forced commands?   forceATA forceATAPIO forceATAUDMA  any of those ???

Swiss3003, 2023-10-19:
[READMEFIRST-ExosX18-EvansBP-SN04-SATA STD 512E firmware update.pdf](https://github.com/Seagate/openSeaChest/files/13037382/READMEFIRST-ExosX18-EvansBP-SN04-SATA.STD.512E.firmware.update.pdf)
[EvansBPExosX18SATA-STD-512E-SN04.zip](https://github.com/Seagate/openSeaChest/files/13037404/EvansBPExosX18SATA-STD-512E-SN04.zip)

So github doesn't support the *.LOD files. That's why it stripped it off. I 7zip it and added it. So this is the same file you have and this is the firmware that you should be using to flash the drive to SN04. If the firmware still doesn't load / change I don't think it's the tool. You might want to check with the customer service and see if they have any helpful tips.

nachopants, 2023-10-19:
Hi Tim,

I've downloaded the provided firmware file. I just tried the force commands for the first time with no luck.

./openSeaChest_Firmware --downloadFW EvansBPExosX18SATA-STD-512E-SN04.LOD --downloadMode segmented --fwdlSegSize 512 --forceATA --forceATAUDMA -d /dev/sg10 -v3

I still basically just get a success message but the firmware stays on SN03

/dev/sg10 - ST16000NM000J-2TW103 - ZR505EW1 - SN03 - ATA
..
Firmware Download successful
Firmware Download time (s): 4.13
Average time/segment  (ms): 375.49
Activate Time          (ms): 77.92
New firmware version is SN03

Swiss3003, 2023-10-19:
Well you really have me stumped. I think it's time to call the call center and see if they know why the drive is not updating. There has to be a reason. Do the drives ever stay in idle for a long time?

nachopants, 2023-10-20:
Thanks mate I'll give them a call. Nope, up 24/7, all less than 3 months old, all running in a Synology DS3617xs

vonericsen, 2023-12-07:
Recently I had a member of our Seagate Support reach out with questions about this same issue.

I'm not sure if it was @nachopants or a different customer running into this issue, but the same product and update failed in their case as well.

This was not a software bug causing the issue, but there is some kind of drive level incompatibility going on.
I'm not sure what the drive level issue is, but the firmware is doing the correct thing by rejecting incompatible code.

It is possible that the update only applies to a drive with a certain manufacturing time, but that level of information is not available to the Seagate.com firmware update lookup tool with the serial number provided, so it shows this update as available even though it does not apply in this case.
The online update I reviewed as "available" came with a `.cfs` file. This is a proprietary file that closed source SeaChest_Firmware can use to assess if a drive is compatible with a given update or not and it appeared to be setup correctly when I reviewed the information I had available with the response I saw on a similar model drive.

I'm marking this as closed since there isn't anything I can do from the software side of this issue. Please feel free to repoen this if I can do anything else to help or create new issues for anything else you run into with openSeaChest.

closed 2023-12-07 by vonericsen