Seagate/openSeaChest #124: How do we scan, find new firmware and update it? I am on windows

FurkanGozukara, 2023-09-26:
I tested this command and below the info

But what is next? Can't find in documentation

```
C:\Users\King\Downloads\openSeaChest-v23.03.1-win-x64>openSeaChest_Basics.exe openSeaChest_Firmware --scan
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 3.5.2-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Tue Sep 26 12:23:33 2023        User: King (admin)
==========================================================================================
Vendor   Handle       Model Number            Serial Number          FwRev
NVMe     PD0          Netac NVMe SSD 2TB      PS202303062T86457025   3.S.F.C
ATA      PD1          ST8000NM000A-2KE101     WSD78KVD               SN03
ATA      PD2          ST8000NM000A-2KE101     WSD78MD3               SN03
ATA      PD3          Samsung SSD 850 EVO 250GB S21MNSAG500430V        EMT02B6Q
ATA      PD4          ST4000NM0024-1HT178     Z4F1C4A4               SN06
NVMe     PD5          KINGSTON SNV2S2000G     50026B768622D93C       SBM02103
NVMe     PD6          SanDisk Ultra 3D NVMe   2038FG802487           211070WD
```

[image]

DebabrataSTX, 2023-09-26:
Please run the help command by issuing 
openSeaChest_Firmware -h
or 
openSeaChest_Firmware --help

This should provide you the details of the supported commands.

vonericsen, 2023-11-07:
Currently the only way to check for a firmware update is to take the serial number and put it into this form on the Seagate website to check if a new firmware is available. The form is at the bottom of https://www.seagate.com/support/downloads/
[image]

It is unlikely SeaChest or openSeaChest will ever implement a method to download a firmware from our website as our enterprise customers have been very clear that they do not want tools that talk outside of their system for any reason. 

If a firmware update is available, the next step would be to download it to the drive:

    openSeaChest_Firmware -d <handle> --downloadFW firmwareFile.lod

FurkanGozukara, 2023-11-07:
thanks i updated my firmware

here made a mini guide as well : https://medium.com/@furkangozukara/how-to-upgrade-update-firmware-of-your-seagate-driver-tutorial-e9cdff297f86

closed 2023-11-07 by FurkanGozukara