Seagate/openSeaChest #127: openSeaChest_Format: unknown option --showSupportedSectorSizes

zohourih, 2023-10-31:
Documentation for [openSeaChest_Format](https://github.com/Seagate/openSeaChest/blob/develop/docs/openSeaChest_Format.152.txt) mentions an option called `--showSupportedSectorSizes`. However, such option does not seem to exist any longer in the latest version of openSeaChest (v23.03.1). Is this supposed to be `--showSupportedFormats` now?

```
test:/opt/openSeaChest # ./openSeaChest_Format --device /dev/sg1 --showSupportedSectorSizes
openSeaChest_Format: unknown option -- showSupportedSectorSizes
openSeaChest_Format: Unable to parse --showSupportedSectorSizes command line option
Please use --help for more information.

test:/opt/openSeaChest # ./openSeaChest_Format | grep -i support
        openSeaChest_Format -d /dev/sg<#> --showSupportedFormats
        4 = Operation not supported
        --showSupportedFormats
```

vonericsen, 2023-11-06:
Hi @zohourih,

It should say `--showSupportedFormats` now as that option replaced the older `--showSupportedSectorSizes`.
I will fix the documentation.

vonericsen, 2025-05-30:
Fixed in [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

closed 2025-05-30 by vonericsen