Seagate/openSeaChest #129: issue

ankitrslr, 2023-11-25:
It seems that the command is failing because the drive is not reporting the native Max LBA (Logical Block Address). The tool attempts to get the max native LBA by issuing the "Read Max Address Ext" command, but this command fails and returns "command aborted."

The last command issued is:

Sending ATA Read Native Max Address Ext
Sending SAT ATA Pass-Through Command:
Protocol: NON-Data
Data Direction: No Data
Task File Registers:
[FeatureExt] = 00h
[Feature] = 00h
[CountExt] = 00h
[Count] = 00h
[LBA Lo Ext] = 00h
[LBA Lo] = 00h
[LBA Mid Ext] = 00h
[LBA Mid] = 00h
[LBA Hi Ext] = 00h
[LBA Hi] = 00h
[DeviceHead] = 40h
[Command] = 27h

This command fails and returns "command aborted." The sense data indicates an aborted command with no additional sense information.

The failure to get the Max LBA is causing the observed issue, and it's mentioned that the drive is connected via a SAS HBA (LSI). Other devices of similar types don't exhibit this issue when tested.

In summary, the problem seems to be specific to this drive and its connection via the SAS HBA, as other similar devices connected in the same way do not encounter this problem.

vonericsen, 2023-11-27:
Hi @ankitrslr,

Thanks for reporting this. Sometimes there are some HBAs/adapters that do not properly report the sense data and without a result for a command like this it will be considered a failure.

I do have some questions about this though since we've usually seen good results on LSI HBAs to report correctly.

Which model is the HBA?

Which firmware revision is the HBA running?

Can you share the full output from openSeaChest so I can see all the commands that were issued? Saving it to a text file, then posting it here will be helpful.
Since this is a SAS HBA, I need to also see the CDB that was issued since that is what the HBA will use to translate the command and I want to verify it was created correctly and this will be in the full output.

You can output to a text file like this: 

      openSeaChest_Basics -d <handle> -i -v 4 > verboseInfo.txt

vonericsen, 2023-12-07:
Closing as this appears to be a duplicate of #114 
If you have additional information to share, can you please share it on that issue?

closed 2023-12-07 by vonericsen