Seagate/openSeaChest #131: Some detailed changelog

soder10, 2024-01-19:
Hello all,

I would like to apply EVANSBP-SN05 firmware to the Exos X18 drive. But the downloaded package does not contain any form of changelog at all (also nothing for previous SN04, SN03 versions in the same package).

Is it possible to include at least some minimal details? Are the fixes very specific to some usecase, or affect a broad range of customers?  I would like to have a rough understanding what is the benefit versus risk ratio, to perform such a risky procedure.

Please, if you can provide any meaningful detail, that would be more than welcome. You are my last hope, traditional support was no use after days of waiting no meaningful reply was received.

vonericsen, 2024-02-06:
Hi @soder10,

Sorry for the delay, this has taken a bit of time to look into and figure out.

Seagate does not share the changelog for firmware updates, but any firmware update that is released for a product generally affects all use cases.
All Seagate firmware updates go through rigorous testing and validation before public release to ensure the update process works without issues and that the new firmware continues to meet the standards and validate all fixes to the new firmware.
 
While firmware updates often sound risky, this is not the same level of risk as drives from the past.
Seagate firmware of today uses a lot of technology to verify a firmware is compatible before accepting it.
Part of this process is the Secure Downloads and Diagnostics feature (SD&D) which verifies a firmware
has not been corrupted or tampered. This also verifies that all firmware is signed by Seagate and it is authentic.
Before a modern drive activates the new firmware, it verifies that it is compatible with all components inside
of the drive as well and in the event that something does go wrong, it can restore its previous state.

soder10, 2024-02-17:
@vonericsen Well, what can I say... :( 
I am quite disappointed that even the enterprise grade seagate devices have no known changelog history, and risking firmware updates (its always a risk, even if a low one, still not zero) cannot be really reasoned with firm advantages gained after the update.
I appreciate your attempt though.

closed 2024-02-17 by soder10