Seagate/openSeaChest #132: openSeaChest .deb package misses export PATH variable

walterav1984, 2024-01-22:
The openSeaChest [V23.12 64bit .deb package](https://github.com/Seagate/openSeaChest/releases/download/v23.12/openseachest_23.12_amd64.deb) installs fine on both Ubuntu 23.04 and Debian 12. 
However after install OpenSeaChest commands nor manual cannot be found nor used directly in shell from neither distro's until a specific path variable is exported or whole global path is passed:

```
export PATH=/usr/bin/openSeaChest:$PATH
```
https://github.com/Seagate/openSeaChest/pull/94#issuecomment-1478103832

Compared to earlier pull request the location of the binaries & manual have changed from `/opt/` to `/urs/bin/`. 

Currently openSeaChest is [officially test packaged](https://tracker.debian.org/pkg/openseachest) by Debian itself and uses `/usr/bin/` directly without a additional subfolder `openSeaChest`  for all its binaries. 

For lowering confusion it may be better to use a `/opt/` or some other `local` bin for this out of tree github created deb package with a correct export path?

vonericsen, 2024-01-22:
I will look into this.
I think this was setup prior to openSeaChest being available directly through debian's package manager.

alexmyczko, 2024-03-18:
@walterav1984 so far nobody asked for it in Debian 12 (official backport), but it should be no problem, if someone does, and officially this should be done by BTS (debian bug tracker system), but i also accept requests on IRC (you'll figure how to find me)

alexmyczko, 2024-06-09:
meanwhile i uploaded openseachest for debian 12, it should appear shortly at https://ftp-master.debian.org/backports-new.html

alexmyczko, 2024-08-30:
@walterav1984 do you think it would make sense to close this issue?

molohov, 2024-09-05:
I recently upgraded from 23.03 to 24.08 and now I cannot find openSeaChest anywhere, including /usr/bin/openSeaChest. Could someone advise?

molohov, 2024-09-05:
```
apt policy openseachest 
openseachest:
  Installed: 24.08
  Candidate: 24.08
  Version table:
 *** 24.08 100
        100 /var/lib/dpkg/status
     23.12-1~bpo12+1 100
        100 http://deb.debian.org/debian bookworm-backports/main amd64 Packages

dpkg -L openseachest
/usr
/usr/local
/usr/local/bin
/usr/local/bin/openSeaChest
/usr/local/bin/openSeaChest/openSeaChest_Basics
/usr/local/bin/openSeaChest/openSeaChest_Configure
/usr/local/bin/openSeaChest/openSeaChest_Defect
/usr/local/bin/openSeaChest/openSeaChest_Erase
/usr/local/bin/openSeaChest/openSeaChest_Firmware
/usr/local/bin/openSeaChest/openSeaChest_Format
/usr/local/bin/openSeaChest/openSeaChest_GenericTests
/usr/local/bin/openSeaChest/openSeaChest_Info
/usr/local/bin/openSeaChest/openSeaChest_Logs
/usr/local/bin/openSeaChest/openSeaChest_NVMe
/usr/local/bin/openSeaChest/openSeaChest_PassthroughTest
/usr/local/bin/openSeaChest/openSeaChest_PowerControl
/usr/local/bin/openSeaChest/openSeaChest_Raw
/usr/local/bin/openSeaChest/openSeaChest_Reservations
/usr/local/bin/openSeaChest/openSeaChest_SMART
/usr/local/bin/openSeaChest/openSeaChest_Security
/usr/local/bin/openSeaChest/openSeaChest_ZBD
/usr/local/share
/usr/local/share/man
/usr/local/share/man/man8
/usr/local/share/man/man8/openSeaChest_Basics.8
/usr/local/share/man/man8/openSeaChest_Configure.8
/usr/local/share/man/man8/openSeaChest_Defect.8
/usr/local/share/man/man8/openSeaChest_Erase.8
/usr/local/share/man/man8/openSeaChest_Firmware.8
/usr/local/share/man/man8/openSeaChest_Format.8
/usr/local/share/man/man8/openSeaChest_GenericTests.8
/usr/local/share/man/man8/openSeaChest_Info.8
/usr/local/share/man/man8/openSeaChest_Logs.8
/usr/local/share/man/man8/openSeaChest_NVMe.8
/usr/local/share/man/man8/openSeaChest_PassthroughTest.8
/usr/local/share/man/man8/openSeaChest_PowerControl.8
/usr/local/share/man/man8/openSeaChest_Raw.8
/usr/local/share/man/man8/openSeaChest_Reservations.8
/usr/local/share/man/man8/openSeaChest_SMART.8
/usr/local/share/man/man8/openSeaChest_Security.8
/usr/local/share/man/man8/openSeaChest_ZBD.8
```

vonericsen, 2024-09-05:
I see the issue. I left it one directory too deep.
My apologies.

I have another bug reported and a feature I'm almost done with that I am also looking into so I can push another release tag with a `.1` to resolve this when I have those completed. If this is needed more urgently, let me know.

vonericsen, 2024-09-26:
I have created release v24.08.1 that fixes this in the RPM and DEB packages.
Please let me know if you have any other issues and thank you for reporting this!

walterav1984, 2024-10-13:
Dear @alexmyczko thanks for the packaging effort. If you don't mind fixing the path variables from the upstream Seagate package downstream when repackaging for Debian, this bug report may be closed.

alexmyczko, 2024-10-17:
i guess this can be closed now