Seagate/openSeaChest #133: multiple ST20000NM007D EXOS X20 20TB firmware SC03 fails warranty claim

walterav1984, 2024-01-22:
Close this issue and possible feature request if not appropriate, but please pass it on internally to the right department within Seagate since I was not able to contact seagate.com personally (consumer chat is not working in Netherlands, no email no phone number since no RMA yet).

Feature request:
Its not possible to check warranty directly from openSeaChest cli tools yet, could this be added as feature request? Maybe put it behind a privacy parameter/flag so its clear that openSeaChest sents information over the network to Seagate.

Issue:
Currently this (seagate website)[https://www.seagate.com/nl/nl/support/warranty-and-replacements] should be used to verify warranty claim but fails on halve of the ST20000NM007D EXOS X20 disks I tried. There might be a production date/firmware pattern to it, hopefully not a gray market or OEM issue!
`https://www.seagate.com/nl/nl/support/warranty-and-replacements/`

All the ST20000NM007D EXOS X20 20TB disk with SN03 firmware produced on or before 28August2023 give valid 5 year warranty claims, all other disks with firmware SC03 and produced on or after 15September2023 shows that I need to contact the reseller?

Obtained multiple mixed firmware disks from 2 [official listed Seagate NL authorized resellers](https://www.seagate.com/nl/nl/where-to-buy/) in the Netherlands `Alternate & ProShop`.

Anyone else having these inconsistent warranty claim checks?

vonericsen, 2024-01-22:
Hi @walterav1984,

For the warranty check, I am not sure how this is checked by the website, but I will look into it to see if it's a feature we may want to implement in the future.

For your issue verifying your warranty, this is very odd. I asked someone in customer support and they recommended contacting Seagate customer support so that they can help look into why the warranty check is coming out like that. 
They did not expect a different firmware or production date to affect that lookup, but would need to get a serial number from you on one of these drives to figure out what is going on.
If you have an email address that I can share with them, I will send that to my contact in customer support, but otherwise I recommend trying to open a case with Seagate's customer support directly.

walterav1984, 2024-02-08:
Future buyers and customers of Seagate EXOS enterprise products in the Netherlands, please take notice where to buy seagate products since the "official" list of seagate sellers on seagate its own website will list companies that will sell OEM disks with no distinction to retail for instance `Alternate.nl and Proshop.nl`!

These companies sells these Seagate products with 5 year warranty advertised on their website (all OEM), but will claim that you will only get 2 years via the reseller themself `Proshop.nl` and the trailing warranty with Seagate who officially rejects the OEM serials. The other one  `Alternate.nl` ships mixed OEM, retail and whitelabel (refurbished) in the same order... but informs you that you may contact the reseller themself in this 5 year period for warranty issues...

Seagate customer support was not able to be more clear about where to buy or able to hint special model numbers which may indicate OEM products beforehand.

Maybe someday this warranty check feature can be extended to OpenSeaChest and Seagate and its resellers can be more transparant what they are selling to who under what conditions.

closed 2024-02-08 by walterav1984