Seagate/openSeaChest #137: openSeaChest_PowerControl standby time ignored

jo-ap, 2024-03-11:
OS: Debian 12 with kernel 6.1.0-18-amd64
Device: Seagate Barracuda 2.5, ST5000LM000-2AN170, Firmware revision 0001

I would like the drive to spin down automatically, but only when it has been inactive for some time so I don't get too many start/stop cycles. As far as I can tell, EPC settings are not supported on this device, so I was trying to achieve this by changing the APM level and set the (legacy) spin down timer as follows
```
sudo openSeaChest_PowerControl -d /dev/sda --setAPMLevel 127 # permit standby
sudo openSeaChest_PowerControl -d /dev/sda --standby 200000 # standby after 20 min
```
This has indeed allowed the drive to spin down, but the timer is ignored, i.e. the drive already spins down when it is idle for a few seconds. 
Is this not the correct way to achieve the wanted behaviour, not possible for my device or a bug?

vonericsen, 2024-03-11:
Hi @jo-ap,

This is what the standard mentions about APM and the standby timer:
[image]

Most likely you are seeing the drive spin down faster because the APM mode you have set is triggering its own timer for when to spin the drive down before the timer set by the `--standby` option timer.

You can try disabling the APM feature to see if that changes the behavior to the longer timeout you requested.
`openSeaChest -d <handle> --disableAPM`

jo-ap, 2024-03-11:
Thanks for the info, I didn't see this in the manpage.
For reference: Disabling APM completely did not work for me, but setting the level back to 128 and enabling a standby timer seem to do the job.

closed 2024-03-11 by jo-ap

vonericsen, 2024-03-11:
@jo-ap ,

Thanks for the feedback!
I will look into the best place to document interaction between the timers and add this information.