Seagate/openSeaChest #148: Issue with Setting Sector Size on Seagate ST24000NM002H after Interruption, Drive Bricked

huahuasousou, 2024-06-10:
Hello everyone,

I am experiencing difficulties with formatting and setting the sector size on my Seagate ST24000NM002H-3KS133 drive using SeaChest tools. I perfom the commad  **--setSectorSize 4096**. But the command showed that the setSectorSize operation failed and a quick format operation was automatically initiated to recover the drive. However, I mistakenly believed that the quick format operation had completed, so I unplugged the SATA cable, leading to a disaster. Now, the LBA is showing as 0, and the SMART information only displays basic details. Any operations I try to perform are indicated as not supported.
Below is the result of my attempt to perform the setSectorSize operation again:
[setSectorSize_log.log](https://github.com/user-attachments/files/15755276/setSectorSize_log.log)

Here are the details of the device and system:

Environment
SeaChest Version: 4.3.6-6_2_0 X86_64
Operating System: Windows 10

Device Information
Model Number: ST24000NM002H-3KS133
Serial Number: ZYD079RP
Firmware Revision: SE05
Physical Sector Size: 4096 bytes
Logical Sector Size: 512 bytes
Steps and Commands Used:
\SeaChest_Format_x64_windows.exe -d PD1 --setSectorSize 4096 --confirm this-will-erase-data-and-may-render-the-drive-inoperable

Request
The issue to have started after an interruption during a sector size change operation (the command shown failed and try to use quick format to recover, but ). Now, the drive is unresponsive, and I suspect it might be due to an internal security lock in the firmware that has bricked the device. Could you please assist in resolving this issue? Is there any way to unlock the drive and restore its functionality? Any insights or additional steps would be greatly appreciated. If more information is needed, I can provide additional details.

Thank you for your support!

vonericsen, 2024-06-12:
Hi @huahuasousou,

On #145 I shared a test build with a `--force` option. Can you give that a try and see if it solves this issue for you?

huahuasousou, 2024-06-12:
> Hi @huahuasousou,
> 
> On #145 I shared a test build with a `--force` option. Can you give that a try and see if it solves this issue for you?

Hi, thank you for the suggestion. I had already checked all the issues related to setSectorSize and even tried using the pre-built version with the --force option before sending it in for RMA. Unfortunately, that did not work for me. 
Since the SMART information is currently unreadable, and both long and short tests cannot be performed, maybe I has thoroughly corrupted the firmware, since I was directly unplugging the SATA cable. There is likely little chance of recovery now. Nevertheless, I appreciate your response.

vonericsen, 2024-06-17:
@huahuasousou,

Thank you for letting me know that did not work.
I am still investigating what else we can do in our tools to mitigate this further.
RMA is the best choice when at the point where the tool is unable to get it to recover.

If you have any suggestions on additional warnings that can be printed on the screen that would have helped, please let me know.

Jacco011, 2024-08-11:
I'm having the same issue with two of the same ST24000NM002H. Interesting, with the ST18000NM000J it works flawlessly.

- Debian 12.6 fresh installation.
- Latest openSeaChest v23.12-linux-x86_64-portable
- Latest Firmware installed where available.
- Connected with SATA directly to the motherboard.

Getting the error "The device was reset during sector size change. Device my not be usable! Attempting Seagate quick format to recover the device."
I'm now waiting over 30 min, but I think it's stuck.

I should have used a live USB, but maybe the X24 is cursed?

manotroll, 2024-08-11:
openSeaChest_Format.exe -d PD1 --setSectorSize 4096 --confirm this-may-erase-data --force --seagateQuickFormat
==========================================================================================
 openSeaChest_Format - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2022 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Format Version: 2.4.0-3_2_1 X86_64
 Build Date: Jul 26 2022
 Today: Sun Aug 11 18:35:14 2024        User: pc (admin)
==========================================================================================

\\.\PhysicalDrive1 - ST16000NM000J-2TW103 - ZR5D5GR5 - SN02 - ATA
WARNING: Forcing Seagate quick format command!
WARNING: Seagate Quick format completed successfully!
         Reading LBAs after a quick format without a write may result in errors!
         A full overwrite is strongly recommended!

Set Sector Size to 4096

WARNING: It is not recommended to do this on USB enclosures as not
         all USB adapters can handle a 4k sector size.
WARNING (SATA): Do not interrupt this operation once it has started or
         it may cause the drive to become unusable. Stop all possible background
         activity that would attempt to communicate with the device while this
         operation is in progress
Press CTRL-C to cancel this operation before the timer runs out.
  0
Setting the drive sector size quickly.
Please wait a few minutes for this command to complete.
It should complete in under 5 minutes, but interrupting it may make
the drive unusable or require performing this command again!!
WARNING: Unable to erase MBR. If unable to write a partition after this operation, erase the first sector of the device
         and the last sector (max LBA) then try creating new partitions again.
Successfully set sector size to 4096

apparently it worked

Jacco011, 2024-08-12:
@manotroll The ST16 aren't the problem. The problem are the new Exos X24 (ST24) ones.
I have filled a RMA for the three disks, wish me luck.

manotroll, 2024-08-12:
Have you tried with the force command?
or see firmware update and use quick format option
they probably won't deny the rma

staaled, 2024-09-19:
I've bricked 3x `ST24000NM002H-3KS133` trying this as well, from Ubuntu booted in single mode:
- 1 via direct SATA to MB (Supermicro H13SSL-NT) (drive FW `SE03`)
- 1 via JBOD config on Broadcom eHBA 9600-24i... (drive FW `SE03`)
- 1 via direct SATA to MB (Supermicro H13SSL-NT) (drive FW `SE04`)

I should note my first attempt (4th drive, FW `SE03`, connected to the 9600 HBA), worked just fine... Same procedure, same system.

The SAS version of the X24 drives works fine (`ST24000NM007H`)

Shaderbug, 2025-05-20:
I've got `ST24000NM002H-3KS133` drives on firmware `SE05`, of which I intended to change the sector size to 4K, before finding this issue. Is this generally unsafe to attempt at the moment?

vonericsen, 2025-05-21:
@Shaderbug,

Thanks for checking in about this!
I have pushed quite a few commits lately that should mitigate most of the impact for external software causing resets, but it isn't released just yet, although I plan to this month (Within the next week or two).

One issue I have run into that I am still debugging is with Broadcom 9300 cards not taking the command correctly. I do not know the cause of that issue yet.

From my testing with AHCI cards (motherboard chipsets, PCIe AHCI cards, etc) I am not seeing the resets happen with all the changes I have made for exclusive access.
This code is on the develop branch of this repo now, so if you are wanting to build and run it you can go for it.

Please let me know if you have any other questions and I will do my best to answer them.

vonericsen, 2025-05-30:
I have pushed a major change that should help prevent issues when changing sector size.
I'm marking this fixed for now with this change that is part of [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

Please reopen this issue or open a new issue if this is still a problem.

closed 2025-05-30 by vonericsen