Seagate/openSeaChest #149: new release? would this work on macOS?

alexmyczko, 2024-07-25:
is there an eta for a new release? debian 13 should ship a newer version in 2025 than 23.x.

do you think this could be made to work on macOS? i could create a homebrew reciepe?

vonericsen, 2024-07-25:
Hi @alexmyczko,

I do not have an exact ETA for a new release, but I know we need one soon.
I've been working on a lot of security updates, and I am making sure they are thoroughly tested before merging them to our develop branch.
The one that is taking the longest to validate it switching to a secure file process.
I have this working on my local machine, but I am still testing that it works correctly and working through some API updates around this. This is my current top priority to finish so we can roll a new release. I am hoping to finish testing my secure file changes within a week to begin a release branch where we will begin the release process.

> do you think this could be made to work on macOS? i could create a homebrew reciepe?

The issue with macOS is that it does not really have a passthrough mechanism built into the OS, which is what we need to issue the commands we use in openSeaChest.
There is an old, and very limited, SMART IO for ATA, but that will only work on internal drives on ATA controllers. While that gets some functionality, it does not really help since most modern Macs do not use ATA drives internally like they used to...not to mention almost all new features and capabilities will not be available on this old API.
There is an opensource USB driver for MacOS we also know about and have considered adding support for, however it requires a VID/PID lookup like our code already does and would require a lot of active maintenance to keep up with new products (I receive a LOT of USB products to test for compatibility with our software and I cannot test them all as fast as I would like to).
The license for this driver is an apple open source license which we need to evaluate if it is compatible with our existing license(s).
I'm not entirely sure if this driver will allow us all the capabilities we provide in openSeaChest or if it is also limited to SMART.

I have some contacts with our USB team inside Seagate that have experience working on Apple drivers, but I have not had the time to look into developing one for our use.

vonericsen, 2024-08-06:
I have completed my `feature/hardening` branch and will be starting the release branch soon...probably this week.
Then we will do some more testing to make sure nothing new shows up as an issue and work on closing it and publishing a new release.

closed 2024-08-30 by alexmyczko