Seagate/openSeaChest #151: Using CFS file results in segfault

reb00tz, 2024-08-16:
Attempting to update firmware on Seagate Exos X18 16TB (ST16000NM000J-2TW103) fails.

Using CFS file results in segfaults.

Tried different variations of commands used (including invalid `--downloadMode` attempts). Used both RAID and Non-RAID versions.

Attached are the `-v3` logs and output, including system and drive information `*_system_drive_info.txt` along with the two RAID/Non-RAID scripts used to generate them.

More platform information as follows:
* Intel Xeon E3-1225 v6
* Asrock C236 WSI w/Intel RAID disabled (ATA mode)
* 2x 2R 16GB DDR4 2400MHz ECC uDIMMs

[run_fwupgrade_RAID.tar.gz](https://github.com/user-attachments/files/16631685/run_fwupgrade_RAID.tar.gz)
[run_fwupgrade_Non-RAID.tar.gz](https://github.com/user-attachments/files/16631682/run_fwupgrade_Non-RAID.tar.gz)

[config_segfault.tar.gz](https://github.com/user-attachments/files/16631651/config_segfault.tar.gz)

vonericsen, 2024-08-16:
@reb00tz,

Thank you for reporting this. I will debug it to figure out what is causing the segmentation fault to fix that.

Starting in the Dec 1, 2023 release support for the CFS was dropped for a new format.
Was this the version of SeaChest_Firmware included with the firmware package from Seagate.com?
As I recall there should have been a version of SeaChest included with that package that supports the CFS file.

reb00tz, 2024-08-16:
The files in OP were from the SeaChest Utilities straight from [Seagate's website](https://www.seagate.com/sg/en/support/software/seachest/#downloads).

It appears that you are correct; using the CFS file, it reports "Model matched but the current Firmware version does not match available updates.".

If this drive has a firmware that somehow is unable to be flashed with the latest bug-fixed firmware, what is my next step? Since the release notes for the latest SN05 is not available, I do not know if the drive will fail prematurely.

[packaged.zip](https://github.com/user-attachments/files/16638782/packaged.zip)

vonericsen, 2024-08-16:
> If this drive has a firmware that somehow is unable to be flashed with the latest bug-fixed firmware, what is my next step? Since the release notes for the latest SN05 is not available, I do not know if the drive will fail prematurely.

I will look into this to see if I can find out some more information on why it would not match as expected.

I am getting close to ready for a new software release, so I definitely want to fix the segfault and provide a better error message for this case.

vonericsen, 2025-08-01:
This was closed back in v24.08, but I forgot to mark it closed...whoops!
Closing now...v25.05.2 SeaChest has been posted to the ToolBin repo and will be updated on the website within a couple weeks.

closed 2025-08-01 by vonericsen