Seagate/openSeaChest #155: Seagate ST2000NM00XX Drive Various Power Management Not Supported

andersonsmith, 2024-09-02:
Hello, thanks for the work on this toolset.  I have a system running 8 seagate SAS drives and cannot get spindown to work.  I first worked with sdparm and would see it bouncing in and out of idle.  I was hoping to use this to monitor the timer and see if it's getting passed idle_a.  The drive manuals for the three models I have all say they support PowerChoice and have the 3 idle states as well as 2 standby states.  When polling the drives, I get the following message returned.

```
==========================================================================================
 openSeaChest_PowerControl - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2024 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_PowerControl Version: 3.6.0-8_0_1 X86_64
 Build Date: Aug 28 2024
 Today: 20240902T091153 User: root
==========================================================================================

/dev/sg7 - ST2000NM0001 - Z1P20M8C - XRBA - UNKNOWN
Showing EPC Settings not supported on this device.
```

I have verified that EPC is enabled, so I'm not sure exactly what to do next.

``` 
       Features Supported:
                Protection Type 1
                Protection Type 2
                Application Client Logging
                Self Test
                Automatic Write Reassignment [Enabled]
                Automatic Read Reassignment [Enabled]
                EPC [Enabled]
                Informational Exceptions [Mode 6]
                Background Media Scan [Enabled]
                Translate Address
                Format Unit
        Adapter Information:
                Adapter Type: PCI
                Vendor ID: 1000h
                Product ID: 0097h
                Revision: 0002h
```

For reference, the drives are all connected through a SAS3008 HBA flashed to IT mode.  The HBA is integrated into a Gigabyte MD70 motherboard.  The host is Proxmox running ZFS on the drives.

Please let me know if I've missed anything.  Thanks again.

vonericsen, 2024-09-03:
Hi @andersonsmith,

I think the reason you are getting this message is because it did not detect the device as a SCSI drive type (It is saying `UNKNOWN` in your output).
I'm not sure why this is not detecting as a SCSI drive type, but this seems to be a software bug so I will debug it to figure out where it went wrong since the rest of the output makes sense.
This is what is also preventing displaying the EPC timer values since it is checking the drive type to figure out how to read it.

An alternative to `--spindown` is `--transitionPower standby` or more specifically for EPC `--transitionPower standby_z` (these should be the same though).
Can you let me know if this works? 
This would be good to know as the code paths between spindown and transition power are a little different and it may be better to make them the same if transition power works correctly.

The other thing I think can help me figure out where this bug is coming from is the output from `openSeachest_PowerControl -d <handle> --llInfo`
That is two lowercase `L`s and a capital `I` for info (short for low-level info).
This will dump a lot of information that openSeaChest reads from `/sys` to figure out how the drive is attached, the adapter IDs (which read correctly for your output), and some other information that may help me figure out what went wrong and triggered this bug.

vonericsen, 2024-09-03:
Oh, one more thing you can do to workaround the `UNKNOWN` drive issue is to use the `--forceSCSI` option which will override the drive type detection (or in this case miss-detection) which should allow it to read and dump the EPC settings.

andersonsmith, 2024-09-04:
Thanks for the replies.  Using `--forceSCSI` did indeed allow the timers to show correctly.  I tried to force the standby mode with `--transitionPower standby_z` but only received `Device is in the Active State or in an Unknown state.` afterwards.  I continued to every 5 seconds or so but it didn't change from this.  

Regarding the info dump, here it is:

```
==========================================================================================
 openSeaChest_PowerControl - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2024 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_PowerControl Version: 3.6.0-8_0_1 X86_64
 Build Date: Aug 28 2024
 Today: 20240903T205432 User: root
==========================================================================================

/dev/sg7 - ST2000NM0001 - Z1P20M8C - XRBA - UNKNOWN

---Low Level tDevice information---
        ---Drive Info---
                media type: HDD
                drive type: unknown
                interface type: SCSI/SAS/FC/etc
                zoned type: not zoned
                ---adapter info---
                        PCI/PCIe:
                        VendorID: 1000h
                        ProductID: 0097h
                        Revision: 0002h
                ---driver info---
                        driver name: mpt3sas
                        driver version string:
                SCSI Version: 6
                ---Passthrough Hacks---
                        Passthrough type: SAT/system/none
                                ---SCSI Hacks---
                                ---NVMe Hacks---
                                ---ATA Hacks---
        ---OS Info---
                handle name: /dev/sg7
                friendly name: sg7
                minimum memory alignment: 8
                ---Linux Unique info---
                        FD is valid
                        SG Driver Version:
                                Major: 3
                                Minor: 5
                                Revision: 36
                OS read-write recommended: false
                last recorded error: 0
                File system Info:
                        No active file systems detected
```

vonericsen, 2024-09-05:
@andersonsmith, 

This bug has been very difficult to locate so far walking through the code with the information you have shared so far.
Do you see this message show up at all? `SCSI interface, unknown drive type`

Can you run this command line and share the output with me?
`openSeaChest_PowerControl -d <handle> --transitionPower standby_z --checkPowerMode -v4 | tee verboseSCSIPower.txt`

This will run with a lot of verbose output showing me all the command issued and all the data transfers to and from the drive for me to validate manually to see if something else is going on unexpectedly.

vonericsen, 2024-09-06:
Thank you for the logs!
I believe I have identified the issue causing "Unknown Drive" so that `--forceSCSI` is not necessary.

Looking through the output with --forceSCSI I see the command that is issued (request sense) to check the power state and it is just reporting a normal "good" status.
While reviewing the standards I can try adding another command into this process to see if it can detect this state: test unit ready.
It may not give as much detail, but I think I can make that work.

You can try spinning the drive down, then trying `openSeaChest_PowerControl -d <handle> --testUnitReady` to see if it has spun down. If it has spun down, then the sense will most likely say `Logical Unit Not Ready - Initializing Command Required` which essentially means it is in standby.
I'll work on adding this in as well, but if you could try this and let me know what you see that would also be helpful to confirm what the drive is doing and reporting in your case.

vonericsen, 2024-09-06:
@andersonsmith,

Can you give the attached build a try and let me know if it works for you to show standby?
If it does not, can you share the output again like you did previously? 
`openSeaChest_PowerControl -d <handle> --transitionPower standby --checkPowerMode -v4 | tee verbosePowerCheck.txt`

[linux-x86_64-portable.zip](https://github.com/user-attachments/files/16912011/linux-x86_64-portable.zip)
[multiple.intoto.jsonl.zip](https://github.com/user-attachments/files/16912015/multiple.intoto.jsonl.zip)

andersonsmith, 2024-09-07:
Output attached.  When I check with `--showEPCSettings` immediately after it went back to `Active or unknown state` but I assume that due to disk activity.

[verbosePowerCheck.txt](https://github.com/user-attachments/files/16914205/verbosePowerCheck.txt)

vonericsen, 2024-09-09:
@andersonsmith,

Thanks for the log again!

>When I check with --showEPCSettings immediately after it went back to Active or unknown state but I assume that due to disk activity.

Yeah, even querying the settings can potentially spin up a disk. It depends if the disk stores the settings on the media or in flash or RAM.
This can change between products as well.

It could also be some other system activity (even something you are not triggering directly, but the OS decides it needs something from the device).

One other thing I have observed in the past is depending on how the drive's handle is opened in Linux, that can cause the drive to spin up.
When I had this reported years ago the issue was using the `sd?` handle caused a spinup, but using `sg` did not.
My solution to this to keep customer usage easy for the handle they are familiar with was to map it myself using sysfs.
If the mapping fails, it uses whatever handle the user provides since some kernels disable /dev/sg? handles (usually only embedded devices).
This issue is not appearing in your logs since I see it using the sg handle, but it is always possible that something in the kernel has changed this behavior, although I am not aware of anything that would affect this right now.

One other thing that may factor into how this is working is the drive firmware. In your case the firmware string I see indicates a customer unique firmware and sometime those have customizations that may cause some changes from the default behavior of OEM code. I don't know of any specific changes to power settings on this code, but I have not worked with it a lot. I think most likely it is spinning back up due to querying the mode page for the current settings or some other background activity from the OS.

SAS drives also have one field that allows specifying preference between background tasks and power modes which could be another part of it. openSeaChest_PowerControl does not currently have a way to toggle this, but I can look into this as part of troubleshooting this issue. We can also try flipping it using openSeaChest_Configure's options to change a mode page with some raw field info if needed.

> Should an asterisk be shown with each timer in --showEPCSettings? I see it appearing and disappearing when I enable and disable idle_a but it doesn't appear when I enable standby_y or standby_z

Yes, that should be changing.
I just reviewed the code and this is a bug. 
It looks like when reading one page there was a mix up on which bit was used for standby y vs standby z. I also found a bug reading the mode page settings where the length of the mode page header was not being factored into the offset to check.
This only seems to affect the standby states which is why you didn't see the issue with idle a.

I'm reviewing more of the code before I push the fix, but I will share that for you to check out as well.

vonericsen, 2024-09-09:
@andersonsmith,

Can you give these updated builds a test to see if it fixes the asterisk in the EPC settings?
I fixed the issues I mentioned in my last comment that should resolve it.

[multiple.intoto.jsonl.zip](https://github.com/user-attachments/files/16933939/multiple.intoto.jsonl.zip)
[linux-x86_64-portable.zip](https://github.com/user-attachments/files/16933940/linux-x86_64-portable.zip)

andersonsmith, 2024-09-10:
Thanks for the deep dive on this one!  Can confirm that the asterisk is showing in the EPC settings output now.

andersonsmith, 2024-09-10:
Thanks for the suggestion.  Looks to me like it's having trouble getting beyond `idle_A`  The timer for 'idle_B' is set to 4 min and `standby_Z` is set to 15 minutes for this particular drive.  Will dive into the stats to see what the system is doing.  This is mostly media so should spend a significant amount of the day idle.  Thanks again for your work on this - this is a very handy tool.

```
---Power Condition Transitions---
 Accumulated Transitions to Active                            N/A              589
 Accumulated Transitions to Idle A                            N/A              4431195
 Accumulated Transitions to Idle B                            N/A              583
 Accumulated Transitions to Idle C                            N/A              0
 Accumulated Transitions to Standby Z                         N/A              6
 Accumulated Transitions to Standby Y                         N/A              0
```

vonericsen, 2024-09-10:
@andersonsmith,

No problem! Thanks for reporting this issue.

I'm currently working on writing up Wiki pages to not only help with openSeaChest use but also to help educate on how drives and their features work.
I have not yet started a page on power options like EPC, but I will put some of the information from here into that wiki page when I get to writing it.

vonericsen, 2024-09-26:
I have made a patch release #v24.08.1 that fixes this in the deb package.

If you still have any issues, please let me know!

vonericsen, 2025-05-30:
This was fixed in the previous patch release, but I forgot to close it.
Also marking fixed for new [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

closed 2025-05-30 by vonericsen