Seagate/openSeaChest #157: [ST16000NM000J] errors after switching sector size to 4096

insunaa, 2024-09-27:
I was testing a new X18 16TB S-ATA drive I bought a few days ago via a USB dock, since all of my hot-swap bays are in use and cannot currently be spared. It showed up normally in `smartctl` under linux and after a 24h smart long test all came back clear.

I then attempted to swap to 4096 sector size, despite the warnings, which failed and left the drive in a temporarily inoperable state. I then plugged the drive into my desktop PC where I connected it with S-ATA directly and using the latest release of openSeaChest at the time of writing I ran the command again, and this time it ***apparently*** completed successfully. `smartctl -a` properly reported all of the info and openSeaChest seemed happy. I disconnected the drive again, but when I plugged it back into the USB enclosure to do a badblocks test on my server, `smartctl` no longer detected the device.

When I run `openSeaChest_Basic -i --SATInfo` I get the following output:

```
=========================================================================================
 openSeaChest_Format - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2024 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Format Version: 3.2.0-8_0_1 ARM
 Build Date: Sep 27 2024
 Today: 20240927T175333	User: root
==========================================================================================

/dev/sda - U3 - 000000004BA8 - 0501 - SCSI
	Vendor ID: IB-1122-
	Model Number: U3
	Serial Number: 000000004BA8
	Firmware Revision: 0501
	World Wide Name: 3000000000485500
	Drive Capacity (TB/TiB): 16.00/14.55
	Temperature Data:
		Current Temperature (C): Not Reported
		Highest Temperature (C): Not Reported
		Lowest Temperature (C): Not Reported
	Power On Time: Not Reported
	Power On Hours: Not Reported
	MaxLBA: 3906469887
	Native MaxLBA: Not Reported
	Logical Sector Size (B): 4096
	Physical Sector Size (B): 4096
	Sector Alignment: 0
	Rotation Rate (RPM): Not Reported
	Form Factor: Not Reported
	Last DST information:
		Not supported
	Long Drive Self Test Time: Not Supported
	Interface speed:
		Not Reported
	Annualized Workload Rate (TB/yr): Not Reported
	Total Bytes Read (B): Not Reported
	Total Bytes Written (B): Not Reported
	Encryption Support: Not Supported
	Cache Size (MiB): Not Reported
	Read Look-Ahead: Enabled
	Write Cache: Enabled
	SMART Status: Unknown or Not Supported
	ATA Security Information: Not Supported
	Firmware Download Support: Not Supported
	Specifications Supported:
		SPC-4
		UAS T10/2095-D revision 04
		SPC-4 (no version claimed)
		SBC-3 (no version claimed)
	Features Supported:
		EPC
	Adapter Information:
		Adapter Type: USB
		Vendor ID: 152Dh
		Product ID: 0578h
		Revision: 0501h
```
Is there any way to recover from this state, given that showing all of this information has worked through that USB enclosure in the past?

```
smartctl 7.4 2023-08-01 r5530 [armv7l-linux-5.4.167+] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

Read Device Identity failed: Connection timed out

If this is a USB connected device, look at the various --device=TYPE variants
A mandatory SMART command failed: exiting. To continue, add one or more '-T permissive' options.
```

```
smartctl 7.4 2023-08-01 r5530 [armv7l-linux-5.4.167+] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

Smartctl open device: /dev/sda [USB JMicron] failed: No device connected
```

I've tested the enclosure on an AMD64 system, too. This is not caused by the current machine being armv7

vonericsen, 2024-09-30:
Hi @insunaa,

It is possible that changing the sector size has caused a bug within the USB adapter's firmware.
Some of these adapters will only recognize drives properly with 512B sectors but there is no way to tell ahead of time.

While `openSeaChest` sees the drive, it had a similar failure trying to do a passthorugh identify command that failed the same as smartctl did which is why it is recognizing it as a SCSI device (The USB adapter is still responding to SCSI commands it appears).
It's possible the firmware on the USB adapter is expecting all data transfers to match the drive's sector size, but that would be wrong on 4k ATA drives. If this is the case it may not allow these passthrough commands to complete as it is stuck waiting on a data transfer that will never happen. For ATA drives with sector sizes other than 512B, data transfers depend on the command. If it's LBA access (reads and writes), it's the logical sector size (4K in this case), otherwise all other commands are multiples of 512B. So for this identify command the drive will only responds with 512B and the adapter may be waiting for a 4K transfer that will not happen due to this internal bug in its firmware.
I've only seen this once before on an old adapter with a modern drive and it had a lot of other issues besides this.

It sounds like you were able to complete the command to get the drive running again on a SATA port on another system.
Can you put this drive back on that system and try running the `-i` option on one of the `openSeaChest` tools to see if it outputs as expected? 
I mostly want to know if it's reporting all information correctly to see if it may be a USB adapter issue or something else on the drive.

One case I have seen (although extremely uncommon) is that all the `-i` data reports correctly but attempting to read/write the drive has issues. You could use `openSeaChest_GenericTests` and the `--shortGeneric` option to see if reads are working on the drive (This tests the inner diameter of the drive, outer diameter of the drive, and some randomly selected sectors).
By default `--shortGeneric` uses reads so it does not affect any data and this should be able to complete if the sector size change completely correctly.
If you can also run this from your SATA motherboard that should help determine if this drive is working properly or not.

insunaa, 2024-09-30:
I'll do the test on S-ATA as soon as I'm able.

In the meantime this is the shortGeneric test via the USB enclosure:

```
==========================================================================================
 openSeaChest_GenericTests - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2024 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_GenericTests Version: 2.3.0-8_0_1 ARM
 Build Date: Sep 27 2024
 Today: 20240930T191815 User: root
==========================================================================================

/dev/sda - U3 - 000000004BA8 - 0501 - SCSI
Starting short generic test.
Sequential Read Test at OD for 39064698 LBAs
Reading LBA: 39064698
Sequential Read Test at ID for 39064698 LBAs
Reading LBA: 3906469885
Random Read Test of 5000 LBAs
Reading LBA: 2461189674
Short generic test completed successfully!
```

insunaa, 2024-10-01:
`smartctl -a`
```
smartctl 7.4 2023-08-01 r5530 [x86_64-linux-6.10.10-273-tkg-bore] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Device Model:     ST16000NM000J-2TW103
Serial Number:    XXXXXXXX (redacted by me)
LU WWN Device Id: 5 000c50 0e81e8989
Firmware Version: SN04
User Capacity:    16,000,900,661,248 bytes [16.0 TB]
Sector Size:      4096 bytes logical/physical
Rotation Rate:    7200 rpm
Form Factor:      3.5 inches
Device is:        Not in smartctl database 7.3/5528
ATA Version is:   ACS-4 (minor revision not indicated)
SATA Version is:  SATA 3.3, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Tue Oct  1 07:26:55 2024 CEST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x82)	Offline data collection activity
					was completed without error.
					Auto Offline Data Collection: Enabled.
Self-test execution status:      (   0)	The previous self-test routine completed
					without error or no self-test has ever
					been run.
Total time to complete Offline
data collection: 		( 1100) seconds.
Offline data collection
capabilities: 			 (0x7b) SMART execute Offline immediate.
					Auto Offline data collection on/off support.
					Suspend Offline collection upon new
					command.
					Offline surface scan supported.
					Self-test supported.
					Conveyance Self-test supported.
					Selective Self-test supported.
SMART capabilities:            (0x0003)	Saves SMART data before entering
					power-saving mode.
					Supports SMART auto save timer.
Error logging capability:        (0x01)	Error logging supported.
					General Purpose Logging supported.
Short self-test routine
recommended polling time: 	 (   1) minutes.
Extended self-test routine
recommended polling time: 	 (1430) minutes.
Conveyance self-test routine
recommended polling time: 	 (   2) minutes.
SCT capabilities: 	       (0x70bd)	SCT Status supported.
					SCT Error Recovery Control supported.
					SCT Feature Control supported.
					SCT Data Table supported.

SMART Attributes Data Structure revision number: 10
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000f   067   067   044    Pre-fail  Always       -       559835
  3 Spin_Up_Time            0x0003   094   094   000    Pre-fail  Always       -       0
  4 Start_Stop_Count        0x0032   100   100   020    Old_age   Always       -       8
  5 Reallocated_Sector_Ct   0x0033   100   100   010    Pre-fail  Always       -       0
  7 Seek_Error_Rate         0x000f   073   061   045    Pre-fail  Always       -       17961792
  9 Power_On_Hours          0x0032   100   100   000    Old_age   Always       -       78
 10 Spin_Retry_Count        0x0013   100   100   097    Pre-fail  Always       -       0
 12 Power_Cycle_Count       0x0032   100   100   020    Old_age   Always       -       4
 18 Unknown_Attribute       0x000b   100   100   050    Pre-fail  Always       -       0
[... 111 lines cut ...]
		PUIS
		GPL
		Streaming
		SMART Self-Test
		SMART Error Logging
		DSN
		AMAC
		EPC [Enabled]
		Sense Data Reporting [Enabled]
		Write-Read-Verify
		SCT Write Same
		SCT Error Recovery Control
		SCT Feature Control
		SCT Data Tables
		Host Logging
		Set Sector Configuration
		Storage Element Depopulation + Restore
		Field Accessible Reliability Metrics (FARM)
		Seagate In Drive Diagnostics (IDD)
	Adapter Information:
		Adapter Type: PCI
		Vendor ID: 1022h
		Product ID: 7901h
		Revision: 0051h
```

vonericsen, 2024-10-01:
@insunaa,

Based on your output it looks like the drive is working properly and that the fast format completed successfully.

> The thing that throws me off is how or why would a USB HDD controller care what the sector size is? Shouldn't it just pass along ATA/SCSI commands to the controller? How does the sector size even interact with that in any way?

I will try to explain this as best I can. The first thing to understand is that USB adapters are all SCSI translators to change from SCSI commands to ATA commands that the drive needs.
This translation is specified by the SAT (SCSI to ATA Translation) spec, however there are lots of cases cut out for translators to do some proprietary things and some parts are just not implemented by some translators.

USB translators are notorious for missing translations and bugs in translation.
Generally USB adapters are focused on meeting the minimum required to function with the OS (most specifically work to meet Windows and Mac requirements and that is it): 
* Inquiry - usually unit SN and devID VPD pages will be supported and maybe block limits and block device characteristics pages.
* Test unit ready
* Read 10 & 16
* write 10 & 16
* verify 10 & 16
* read capacity 10 & 16. 
* A lot will implement some support for SAT ATA-Passthrough however that can vary in the format the USB adapter works best with (some want 12B, some want 16B, some only respond with TPSIU set, some will never respond with TPSIU, etc).

The first thing to keep in mind is many of these devices started being developed before 4K sectors were implemented on any device. Even once 4K became available, most consumer models of HDD didn't change or allow changing to a 4K logical sector size on the drive so the adapters still saw everything as a traditional 512B device.

> Shouldn't it just pass along ATA/SCSI commands to the controller?

For reads and writes, yes. 
And your USB adapter seems to be able to do this part fine since the short generic completed over the USB adapter.

Where it gets tricky is the SAT spec describes a command called `ATA Passthrough` which is what tools like openSeaChest and smartmontools need in order to function and get things like SMART data as the translation does not have another method to access that data other than ATA passthrough. Even the things the SAT spec does specify for translation (like the DST log and a few others), I have never seen a USB adapter implement, only SAS/SATA HBAs have implemented them that I have seen.

Inside of an ATA Passthrough CDB you specify all the command inputs to send to the ATA drive (task file registers) as well as some additional information to tell the translator how to do the data transfer (direction, protocol, how to calculate the total transfer length).
While openSeaChest and smartmontools put all this information into the CDB correctly, the adapter does not seem to be honoring the calculation for the data transfer correctly which is why these commands are failing.

When you run a drive with a 512B logical sector size, all commands to and from the device are all multiples of 512B.

When you run a drive with a 4096B logical sector size, all read and write commands to LBA space are 4096B multiples and all other commands are still the 512B multiples (Identify, read SMART data, read log data, etc).

What seems to be happening with this USB adapter is that has something in its firmware hard-coded to expect all data transfers to be multiples of the logical sector size or it is not properly handling the byte_block field in the CDB that tells the adapter that the transfer is multiples of 512B or multiples of the logical sector size (it may be treating these two options as one in the same). So, while your drive was in 512e mode it worked fine, but switching to 4096B now it cannot complete the ATA passthrough requests for sizes that are not multiples of the logical sector size.
In smartmontools it outputs that a timeout occurred on the identify command which I have seen often when a transfer length is not set correctly in a command, but in this case since the transfer length is correct it tells me that the USB adapter is waiting for the drive to send it more data that will never come, hence the timeout.
There is no way to force an ATA drive to change how it responds or to pad the responses, so if there is a bug in something like a USB adapter firmware, there is no simple workaround like this.

Now we get into the next part of this whole mess....there are multiple ways to construct equivalent SAT passthrough CDBs.

Due to how ATA has developed there are a lot of weird cases to handle which is why SAT passthrough can get complicated quite quickly and it is why there are so many different bit fields used to communicate to the translator how much data is being transferred. 
In our openSeaChest code we use a set of "default rules" when constructing the SAT CDBs that seems to be as compatible as possible for as many devices as possible. Smartmontools does this as well.
However, both openSeaChest and smartmontools have had to implement custom ways to make specific USB adapters work better.
In openSeaChest that happens in [this function](https://github.com/Seagate/opensea-transport/blob/develop/src/common_public.c#L3379)
In smartmontools I believe some of this is set in the drivedb file.

[... cut ...]
These are the two outputs from that tool that tend to help me the most:
`openSeaChest_PassthroughTest -d <handle> --runPTTest --ptDriveHint ata --ptTypeHint sat | tee passthroughtest.txt`
`openSeaChest_PassthroughTest -d <handle> --runPTTest --ptDriveHint ata --ptTypeHint sat --enableHangCmdsTest all | tee passthroughtestextras.txt`

Both of these are data-safe tests. The second is a bit more aggressive to look for a few additional cases that seem common to cause the USB adapter to hang and stop responding which is why it is not enabled by default.

insunaa, 2024-10-01:
@vonericsen You, sir, are a saint.
You have my deep gratitude for your extremely easy to understand and well written explanation of the issue, as well as your incredible willingness to help.
Thank you for the time you've given me, and also the time you've given to the openSeaChest project!

[passthroughtest.txt](https://github.com/user-attachments/files/17213642/passthroughtest.txt)
[passthroughtestextras.txt](https://github.com/user-attachments/files/17213643/passthroughtestextras.txt)

If there are any other tests you'd like me to run, regardless of data safety as this HDD is meant to become a cold spare drive anyway, please do tell me. Thank you very much, again!

vonericsen, 2024-10-02:
@insunaa,

Thanks! I do my best and try to find ways to explain these problems as best I can in a way others can understand.

Would you mind running this in verbose mode?
`openSeaChest_PassthroughTest -d <handle> --runPTTest --ptDriveHint ata --ptTypeHint sat -v 4 | tee passthroughtest.txt`

It did not find a way to get the ATA passthrough commands to work with the known workarounds.
If you have a link to buy this adapter, I can look into buying one to test myself since it does not appear to be an easy workaround if one even exists.

> Is it feasible to convert the drive back to 512e or will trying to use --setSectorSize=512 just make things worse?

You can convert it back to 512e if you want. I don't suspect it will make anything worse, however changing sector size on the drive does have its own risks as you see in all the warnings (and some issues/discussions from other users).
If changing sector size is interrupted, it can leave the drive in a state where it does not work properly. I have yet to find a conclusive way to mitigate this issue and I've been trying a lot of experiments to find something else we can do to mitigate the issues further with little success.

If you convert it back to 512e that will get it reporting SMART data again with this adapter since that was working for you originally.
If you decide to do this, can you run the same openSeaChest_PassthroughTest commands you ran earlier in 512e mode?
This will likely provide some results showing what worked or didn't work in ATA passthrough. It might help with 4KN mode, but it may not either. This is the _joy_ of working with USB /s.

insunaa, 2024-10-02:
@vonericsen Thank you for your reply! The resulting file was too large to attach to GitHub, so I sent an e-mail with a Google Drive link to the E-Mail you list on GitHub.

I will convert the device back to 512e later tonight and run another passthrough test. This is probably the best course of action since the other Seagate drives in array, which are currently actively in use, are also formatted to 512e, so no real reason to break the mold after all.

insunaa, 2024-10-02:
@vonericsen Thank you for your patience. Here are the passthrough tests with 512e:

[passthroughtest2.txt](https://github.com/user-attachments/files/17237129/passthroughtest2.txt)
[passthroughtestextras2.txt](https://github.com/user-attachments/files/17237130/passthroughtestextras2.txt)

vonericsen, 2024-10-03:
@insunaa,

Thank you for the email!

I found the same enclosure on amazon for the US. I'll see if we can get it or if I can get the same adapter another way.
One possible downside of me purchasing this box is sometimes the vendors of these boxes switch bridge chip models at different times in production.
We experienced this buying some in the past where one manufactured earlier in the year had one chip and the adapters we bought later had a completely different chip.
I was able to get both working in their own ways, but it can add to the challenge of getting them working.

I pushed the info about what worked properly from the 512e output so at least that is in there for now.
I don't think that will help in the 4KN case though based on the other output I see from the long, verbose output you shared.

I have a couple ideas on things that I can try to see if I can get it working in 4KN mode, but no guarantee they will work.
As I mentioned earlier there are multiple ways to build a SAT ATA Passthrough command.
Currently it is setup (and the 512e mode supports this) to use the "Sector Count" register to determine the amount of data to transfer. The other bitfield is set to indicate it is a multiple of 512B as well.
The idea I have is to try the TPSIU field which is "Transport specific information unit" which is a long way of saying "Check the data length in the USB packet layer instead" which some USB devices support and some do not.
I didn't see output indicating this was working in 512e mode though so I'm not confident this would work.
However, there is another part of the bitfield called "Byte-block" which may affect the interpretation as well and maybe that will help, so I could also see if that matters when using TPSIU (past experience has told me this doesn't but that may be different in newer firmware).

Let me see if I can get this adapter here as I would like to figure out if there is a possible workaround since USB adapters are often used to try quicky moving drives around and recover data from them. Luckily it does at least read and write still, but it would be nice to pull other drive data as well.