Seagate/openSeaChest #159: ST32000444SS refuses firmware update "Firmware Download failed"

gdevenyi, 2024-10-14:
```
/dev/sg1 - ST32000444SS - <Redacted> - 0006 - SCSI
        Vendor ID: SEAGATE
        Model Number: ST32000444SS
        Serial Number: <Redacted>
        PCBA Serial Number: <Redacted>
        Firmware Revision: 0006
        World Wide Name: <Redacted>
        Date Of Manufacture: Week 18, 2011
        Copyright: Copyright (c) 2010 Seagate All rights reserved
        Drive Capacity (TB/TiB): 2.00/1.82
        Temperature Data:
                Current Temperature (C): 33
                Highest Temperature (C): Not Reported
                Lowest Temperature (C): Not Reported
        Power On Time:  8 years 247 days 13 hours 28 minutes
        Power On Hours: 76021.47
        MaxLBA: 3907029167
        Native MaxLBA: Not Reported
        Logical Sector Size (B): 512
        Physical Sector Size (B): 512
        Sector Alignment: 0
        Rotation Rate (RPM): 7200
        Form Factor: 3.5"
        Last DST information:
                Time since last DST (hours): 10486.47
                DST Status/Result: 0x2
                DST Test run: 0x2
        Long Drive Self Test Time:  5 hours 9 minutes
        Interface speed:
                Port 0
                        Max Speed (GB/s): 6.0
                        Negotiated Speed (Gb/s): Not Reported
                Port 1 (Current Port)
                        Max Speed (GB/s): 6.0
                        Negotiated Speed (Gb/s): 6.0
        Annualized Workload Rate (TB/yr): 52.08
        Total Bytes Read (TB): 384.05
        Total Bytes Written (TB): 67.94
        Encryption Support: Not Supported
        Cache Size (MiB): Not Reported
        Read Look-Ahead: Enabled
        Write Cache: Enabled
        SMART Status: Unknown or Not Supported
        ATA Security Information: Not Supported
        Firmware Download Support: Full, Segmented
        Number of Logical Units: 1
        Specifications Supported:
                SPC-3
        Features Supported:
                Self Test
                Automatic Write Reassignment [Enabled]
                Automatic Read Reassignment [Enabled]
                EPC
                Informational Exceptions [Mode 0]
                Background Media Scan [Enabled]
                Background Pre-Scan [Enabled]
                Translate Address
                Format Unit
        Adapter Information:
                Adapter Type: PCI
                Vendor ID: 1000h
                Product ID: 0072h
                Revision: 0003h
```

Attempted update
```
./openSeaChest_Firmware -d /dev/sg1 --downloadFW firmware/MU-SAS-0008.LOD --verbose 3
==========================================================================================
[... 1098 lines cut ...]
  0x00 3B 07 00 0F 80 00 00 80 00 00

SG IO Issued as Indirect IO
SG Masked Status = 01h - Check Condition
SG Driver Status = 08h - Driver Sense Data Available

  Sense Data Buffer:

        0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
  0x00 70 00 05 00 00 00 00 0A 00 00 00 00 26 99 06 00
  0x10 00 00

Sense Key: 5h = Illegal Request
ASC & ASCQ: 26h - 99h = Vendor specific ascq code
FRU: 6h = Vendor Specific
Information: 00000000h
Command Specific Information: 00000000h
Command Time (us): 820.62

Write Buffer returning: UNKNOWN

Firmware Download failed
```

Disk is attached to a `Broadcom / LSI SAS2008 PCI-Express Fusion-MPT SAS-2`

gdevenyi, 2024-10-14:
I should note I also have some of these drives behind a mp2sas HBA as well which throws the same errors refusing to update the firmware.

vonericsen, 2024-10-15:
Hi @gdevenyi,

Based on the final sense code being reported from the download attempt, the drive is indicating that the firmware is not compatible with the hardware within the drive.

gdevenyi, 2024-10-15:
Steps:
```
root:~/openSeaChest-v24.08.1-linux-x86_64-portable# ./openSeaChest_Firmware --scan
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2024 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 4.2.0-8_0_1 X86_64
 Build Date: Sep 25 2024
 Today: 20241015T111838 User: root
==========================================================================================
Vendor   Handle       Model Number            Serial Number          FwRev
ATA      /dev/sg0     TS32GMTS400             <REDACTED>             N1126I
SEAGATE  /dev/sg1     ST32000444SS            <REDACTED>                0006
SEAGATE  /dev/sg2     ST2000NM0001            <REDACTED>                0002
SEAGATE  /dev/sg3     ST2000NM0001            <REDACTED>                0002
SEAGATE  /dev/sg4     ST32000444SS            <REDACTED>                0006
SEAGATE  /dev/sg5     ST32000444SS            <REDACTED>                0006
SEAGATE  /dev/sg6     ST32000444SS            <REDACTED>                0006
SEAGATE  /dev/sg7     ST32000444SS            <REDACTED>                0006
SEAGATE  /dev/sg9     ST2000NM0001            <REDACTED>                0002
```

Take Serial number from /dev/sg1, put in https://apps1.seagate.com/downloads/request.html?userPreferredLocaleCookie=en_US_

Get this page:
[image]

Download leads to:
https://apps1.seagate.com/downloads/certificate.html?action=performDownload&key=680363741004

File: ConstellationES1-Muskie-SAS-StdOEM-0008.zip

Unzip:
Contains firmware/MU-SAS-0008.LOD

Run command above.

vonericsen, 2024-10-15:
> Am I doing something wrong?

No, you are getting the firmware from the website correctly. The drive is rejecting the firmware because it is detecting an incompatibility with some of the hardware used when building it. I don't have enough information on this drive to determine if the reason is it needs 0007 first or if this firmware was released only for certain drives with certain hardware internally. Sometimes Seagate will release firmware that patches drives and it only applies to a specific piece of hardware within the drive and drives without that hardware will not accept the firmware as they are not affected by the patch.
The website is not able to determine from the SN this low-level detail which is why it offers up this firmware in this case.

gdevenyi, 2024-10-16:
Ok, unless you can help me get my hands on 0007, I guess there's nothing to be done.