Seagate/openSeaChest #163: Failed set sector size Is there any way to recover or do I have to buy a new drive?

KeinNiemand, 2024-10-18:
I tired setting the sector size to 4096 on my brand new ST16000NM000J, attached via a USB-SATA adapter (specifically a FANTEC QB-35U31), I stupefy ignored all the warnings and probably destroyed my brand new drive.

```
This is the output of running the command:
Set Sector Size to 4096
Setting the drive sector size quickly.
It should complete in under 5 minutes, but interrupting it may make
the drive unusable or require performing this command again!!
The command is issued with a timeout of 1 hour, much longer than necessary,
however there are rare cases where a drive is processing something in the
background when this command is received that may make this take longer than
a few minutes. Please wait at least this hour to ensure that this completes
successfully or with the error recovery built into this software.
ERROR: The device was reset during sector size change. Device may not be usable!
Attempting Seagate quick format to recover the device.
WARNING: Seagate quick format did not complete successfully!
ERROR: Quick format did not recover the device. The device may not be usable!
Failed to set sector size!
For SCSI Drives, try a format unit operation to recover the device
Trying to do it again and even trying something as simple as reading smart status now gives me errors.
```

Trying the command again results in an error that setting sector size is not supported
```
/dev/sg1 - M000J-2TW103 - 0000000000000001 - SN04 - ATA
Set Sector Size to 4096
Setting sector size not supported on this device

```
Trying to do anything else on the drive like reading smart data also fails

```
sudo openSeaChest_SMART -d /dev/sdb --smartAttributes hybrid
==========================================================================================
 openSeaChest_SMART - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_SMART Version: 2.3.2-6_2_0 X86_64
 Build Date: Dec  4 2023
 Today: Fri Oct 18 19:27:23 2024        User: root
==========================================================================================

/dev/sg1 - M000J-2TW103 - 0000000000000001 - SN04 - ATA
Error retreiving the logs.
A failure occured while trying to get SMART attributes
```
Is there anything I can try to recover the drive or is it just straight up dead?

onebitboy, 2024-10-23:
I had the same error after trying to set the sector size with an external USB case. It worked fine again after connecting it to an internal SATA port and setting the sector size once more.

vonericsen, 2024-10-24:
@KeinNiemand,

Sorry you had this happen. The only thing you can do is to try the command again. I did add a `--force` option to try and force it to run the command again, but I cannot guarantee it will fix it.
I have been testing other possible code changes to mitigate this further for a while now, but I still have not found any improvements to this process yet.

vonericsen, 2025-05-30:
I have pushed a major change that should help prevent issues when changing sector size.
I'm marking this fixed for now with this change that is part of [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

Please reopen this issue or open a new issue if this is still a problem.

closed 2025-05-30 by vonericsen

maxpain, 2025-08-14:
I have the same problem.

First try with `v24.08.1`: 
```
# openSeaChest_Format -d all --setSectorSize=4096 --onlySeagate --confirm this-will-erase-data-and-may-render-the-drive-inoperable
==========================================================================================
 openSeaChest_Format - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2024 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Format Version: 3.2.0-8_0_1 X86_64
 Build Date: Sep 25 2024
 Today: 20250814T114513	User: root
==========================================================================================
nvme_ioctl_id: Not a tty
WARN: Not all devices enumerated correctly
One or more of the options provided will perform a low-level format that cannot
be interrupted once started. All background software should be stopped, any filesystems
that are currently mounted should first be unmounted in order to reduce the risk of
interruption. Do not attempt these operations on multiple devices at the same time
to ensure the best possible outcome. Many controllers/drivers/HBAs cannot handle these
operations running in parallel without issuing a device reset.
Not all background activities can be stopped. Some are managed by the OS and are not
configurable. It is recommended that a format change is done from a live/bootable
environment to reduce the risk of these interuptions. If the OS is unable to complete
certain commands for it's background polling of the device, it may trigger a device
reset and interrupt the format, leaving the drive inoperable if it cannot be recovered.
		There is a risk when performing a low-level format/fast format that may
		make the drive inoperable if it is reset at any time while it is formatting.
		WARNING: Any interruption to the device while it is formatting may render the
		         drive inoperable! Use this at your own risk!
		WARNING: Set sector size may affect all LUNs/namespaces for devices
		         with multiple logical units or namespaces.
		WARNING (SATA): Do not interrupt this operation once it has started or
		         it may cause the drive to become unusable. Stop all possible background
		         activity that would attempt to communicate with the device while this
		         operation is in progress
		WARNING: It is not recommended to do this on USB as not
		         all USB adapters can handle a 4k sector size.
		WARNING: Disable any out-of-band management systems/services/daemons
		         before using this option. Interruptions can be caused by these
		         and may prevent completion of a sector size change.
		WARNING: It is recommended that this operation is done from a bootable environment
		         (Live USB) to reduce the risk of OS background activities running and
		         triggering a device reset while reformating the drive.

If you wish to cancel this operation, press CTRL-C now to exit the software.
 0

/dev/sg10 - ST24000NM002H-3KS133 - WYD01BSA - SE05 - ATA
Set Sector Size to 4096
Setting the drive sector size quickly.
This command may appear to hang the utiliy. Do NOT interrupt this
command for at least 1 hour if it appears hung. The drive is busy
performing the sector size change and is not able to indicate its
progress during this time.
This often only takes a few minutes to complete and return completion
of this command to the utility but in some cases it may take this full
hour before the drive is ready to use.
After this command has completed successfully the drive is ready to
read and write data.
Be aware that the drive may begin performing vendor unique background
activity which may prevent idle and standby timers from taking affect
until the background activity has completed.
If this command takes an hour or the command reports a failure due to
interruption by the system with a reset, recovery will be attempted
automatically. You may attempt to run this command again if recovery
does not appear successfull.
ERROR: The device was reset during sector size change. Device may not be usable!
Attempting Seagate quick format to recover the device.
WARNING: Seagate quick format did not complete successfully!
ERROR: Quick format did not recover the device. The device may not be usable!
[... 49 lines cut ...]
Setting the drive sector size quickly.
This command may appear to hang the utility. Do NOT interrupt this
command for at least 1 hour if it appears hung. The drive is busy
performing the sector size change and is not able to indicate its
progress during this time.
This often only takes a few minutes to complete and return completion
of this command to the utility but in some cases it may take this full
hour before the drive is ready to use.
After this command has completed successfully the drive is ready to
read and write data.
Be aware that the drive may begin performing vendor unique background
activity which may prevent idle and standby timers from taking affect
until the background activity has completed.
If this command takes an hour or the command reports a failure due to
interruption by the system with a reset, recovery will be attempted
automatically. You may attempt to run this command again if recovery
does not appear successfull.
WARNING: Unable to erase MBR. If unable to write a partition after this operation, erase the first sector of the device
         and the last sector (max LBA) then try creating new partitions again.
ERROR: The device was reset during sector size change. Device may not be usable!
Attempting Seagate quick format to recover the device.
WARNING: Seagate quick format did not complete successfully!
ERROR: Quick format did not recover the device. The device may not be usable!
Setting sector size not supported on this device
```

maxpain, 2025-08-14:
I also tried it on another server. The same problem.

```
root@h3-1:/home/maxpain# openSeaChest_Format -d /dev/sda --setSectorSize=4096 --onlySeagate --confirm this-will-erase-data-and-may-render-the-drive-inoperable
==========================================================================================
 openSeaChest_Format - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2025 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Format Version: 3.4.0 X86_64
 Build Date: Jul 30 2025
 Today: 20250814T132911	User: root
==========================================================================================
One or more of the options provided will perform a low-level format that cannot
be interrupted once started. All background software should be stopped, any filesystems
that are currently mounted should first be unmounted in order to reduce the risk of
interruption. Do not attempt these operations on multiple devices at the same time
to ensure the best possible outcome. Many controllers/drivers/HBAs cannot handle these
operations running in parallel without issuing a device reset.
Not all background activities can be stopped. Some are managed by the OS and are not
configurable. It is recommended that a format change is done from a live/bootable
environment to reduce the risk of these interruptions. If the OS is unable to complete
certain commands for it's background polling of the device, it may trigger a device
reset and interrupt the format, leaving the drive inoperable if it cannot be recovered.
		There is a risk when performing a low-level format/fast format that may
		make the drive inoperable if it is reset at any time while it is formatting.
		WARNING: Any interruption to the device while it is formatting may render the
		         drive inoperable! Use this at your own risk!
		WARNING: Set sector size may affect all LUNs/namespaces for devices
		         with multiple logical units or namespaces.
		WARNING (SATA): Do not interrupt this operation once it has started or
		         it may cause the drive to become unusable. Stop all possible background
		         activity that would attempt to communicate with the device while this
		         operation is in progress
		WARNING: It is not recommended to do this on USB as not
		         all USB adapters can handle a 4k sector size.
		WARNING: Disable any out-of-band management systems/services/daemons
		         before using this option. Interruptions can be caused by these
		         and may prevent completion of a sector size change.
		WARNING: It is recommended that this operation is done from a bootable environment
		         (Live USB) to reduce the risk of OS background activities running and
		         triggering a device reset while reformating the drive.

If you wish to cancel this operation, press CTRL-C now to exit the software.
 0

/dev/sg1 - ST24000NM002H-3KS133 - WYD01BM2 - SE05 - ATA
Set Sector Size to 4096
Setting the drive sector size quickly.
This command may appear to hang the utility. Do NOT interrupt this
command for at least 1 hour if it appears hung. The drive is busy
performing the sector size change and is not able to indicate its
progress during this time.
This often only takes a few minutes to complete and return completion
of this command to the utility but in some cases it may take this full
hour before the drive is ready to use.
After this command has completed successfully the drive is ready to
read and write data.
Be aware that the drive may begin performing vendor unique background
activity which may prevent idle and standby timers from taking affect
until the background activity has completed.
If this command takes an hour or the command reports a failure due to
interruption by the system with a reset, recovery will be attempted
automatically. You may attempt to run this command again if recovery
does not appear successfull.
ERROR: The device was reset during sector size change. Device may not be usable!
Attempting Seagate quick format to recover the device.
WARNING: Seagate quick format did not complete successfully!
ERROR: Quick format did not recover the device. The device may not be usable!
open: No such device or address
open failure
Error: 6 - No such device or address
Failed to set sector size!
```

vonericsen, 2025-08-20:
@maxpain,

Sorry I did not see your posts earlier!

I'm going to take this back to the firmware team to research what else can be done.
I've exhausted my ideas on preventing this from happening as well as how to recover.
I have no idea how long that will take so my advice is to reach out to Seagate support for what they can do since these drives are now in this weird state.

If you are able to provide some info, are the `smartd` or `udisks` services running?
I was chatting with someone in Seagate doing IO testing and these were skewing their results when they were running. I do not know if they are directly related to this issue or not though. It could be something else entirely....just trying to find anything else to look at.
smartd was easy to disable; udisks would automatically re-enable itself, so it had to be masked off instead.

These are the commands they used (if you have more drives to try and want to try this again....no worries if you don't. We can recreate the issue and test more, but it will likely take some time to get a solution).

> sudo systemctl status smartd
sudo systemctl status udisks2
 
>sudo systemctl mask udisks2
sudo systemctl unmask udisks2
sudo systemctl enable udisks2
sudo systemctl stop smartd
sudo systemctl disable smartd