Seagate/openSeaChest #164: Issues over ESATA

Deepcuts, 2024-10-28:
Hi,

I am not sure if this is a problem with openSeaChest or the equipment I am using.
Because I need to update and test various drives, I purchased an ESATA enclosure and an ESATA controller from StarTech.
Both openSeaChest and SeaTools see the drive just fine. (PD5 is ESATA)
[image]

SeaTools cannot run any self tests on the drive connected via ESATA and as a bonus, WWN, form factor and firmware version are not reported:
[image]

openSeaChest cannot execute most commands on the drive.
For example these commands do not work:

`openSeaChest_PowerControl.exe -d PD5 --powerBalanceFeature disable`
[image]
`openSeaChest_PowerControl.exe -d PD5 --EPCfeature disable`
[image]
`openSeaChest_PowerControl.exe -d PD5 --showEPCSettings`
[image]

I am also unable to upgrade the firmware with either SeaTools (the program just crashes) or openSeaChest.

The same drive works just fine when connected via SATA directly.

I have searched but was unable to find any issues with ESATA and openSeaChest.

Any clue what migth be the reason for these issues?
Or am I wrong in assuming self tests, upgrading the firmware and setting various option like EPC via ESATA is a no go?

Thank you

vonericsen, 2024-10-28:
Hi @Deepcuts,

This is a weird thing to see. e-SATA is just an external SATA connection without any other thing going on (like USB where it translates from SCSI to ATA). So it *should* work without any fuss, but maybe it's related to a driver or controller bug specific to the adapter you are using.

One of the first odd things is that the model number and serial number can be read but the firmware revision is missing....these all report from the same command: identify device. So, it's odd that some of that got through but not all of it.

Can you share the verbose output from `openSeaChest_PowerControl -d <handle> -i -v 4 > verboseidentify.txt`?
That may give me more information to see what is happening in the drive's responses and the drive's command completion information.
Also, can you share the output of `openSeaChest_PowerControl -d <handle> --llInfo` (two lowerccase L's and capital I). This reads some information about what openSeaChest detected for how to communicate with the drive, the driver in use, etc which can help me figure out what code path was taken to talk to the drive.

I have a couple other things that might also help to debug this:

- verify the e-SATA cables are connected all the way...some I've used in the past had a habbit of coming loose resulting in some odd behavior every once in a while.
- Can you let me know if you hot plugged the drive? (Like put it in the enclosure then powered it on while your system was running)
- Can you try putting the drive in the enclose and powered on, then booting up the computer or rebooting the computer and letting me know if that changes anything in either openSeaChest or SeaTools? I just want to rule out some weird hot-plug issue with this test.

Deepcuts, 2024-11-01:
Hi @vonericsen,

Sorry for my delayed response, as I had a few emails back-and-forth with StarTech support, the maker(?) of both the ESATA controller and HDD enclosure.
Their conclusion: 
_Yes I understand. My statement still stands that the enclosure does not support passing drive information which can include values like firmware version, form factor, WWN, etc. 
In summary, although some information is passed through the enclosure, we do not expect any information to be reliably passed thus it is not supported._ 

What I discovered is that this HDD enclosure can work for firmware updates over ESATA, but the trick is to either have both HDD bays filled or just the top one.
With both or just the top bay having a HDD inserted, the firmware version is not being displayed just fine and I can issue commands via openSeaChest.
With a HDD in the bottom bay only, nothing works.

Conclusion: who needs to follow standards when one can just slap some marketing words on the box and call it a day.
If you need something that works, stay away from StarTech products. Or at least this one:
https://www.startech.com/en-us/hdd/s352bu33rer
[image]

Thank you for your time

closed 2024-11-01 by Deepcuts

vonericsen, 2024-11-01:
@Deepcuts,

Thank you for reporting back with this information! 
This is good to know in case this comes up again in the future.

I'm guessing they ran into issues with eSATA cards supporting SATA port multipliers properly and came up with this as the workaround for "compatibility". SATA port multipliers are not commonly used in consumer systems I've seen. I have run into some SATA controllers not supporting them before which is why I'm guessing this was the reason. (Intel chipsets do not support them in my experience).
Only startech will know the real reason though.