Seagate/openSeaChest #165: Solaris and descendants define offset_t in sys/types.h

drboone, 2024-10-29:
The opensea-common section of openSeaChest defines offset_t.  Solaris and at least some of its descendants define offset_t in sys/types.h for use with the llseek system call:
```
LLSEEK(2)                        System Calls                        LLSEEK(2)

NAME
       llseek - move extended read/write file pointer

SYNOPSIS
       #include <sys/types.h>
       #include <unistd.h>

       offset_t llseek(int fildes, offset_t offset, int whence);
```
Thus, building openSeaChest currently fails on these platforms.  It looks like the problem is restricted to just the -common subtree.  Perhaps openSeaChest could use a slightly less collidey typedef name?  A naive replacement of offset_t with OSCoffset_t seems to eliminate this build problem.  From a SmartOS (Illumos distro) build attempt:
```
lesmiz 3 $ ninja -C builddir
ninja: Entering directory `builddir'
[1/152] Compiling C object subprojects...pensea-common.a.p/src_safe_bsearch.c.o
FAILED: subprojects/opensea-common/libopensea-common.a.p/src_safe_bsearch.c.o 
cc -Isubprojects/opensea-common/libopensea-common.a.p -Isubprojects/opensea-common -I../subprojects/opensea-common -I../subprojects/opensea-common/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=gnu99 -O3 -Wshadow=compatible-local -Wvla -Wfloat-equal -Wnull-dereference -Wunused-const-variable -Wunused-parameter -Wunused-value -Wduplicated-cond -Wjump-misses-init -Wstringop-overflow -Wlogical-op -Wshift-overflow -Wshift-overflow=1 -Wshift-overflow=2 -Wdouble-promotion -Wformat-security -Wold-style-definition -Wstrict-prototypes -Wmissing-declarations -Wmissing-prototypes -Wchar-subscripts -Wundef -Wformat -Wformat=2 -Wint-conversion -Wenum-conversion -Wfloat-conversion -Wint-to-pointer-cast -Wimplicit-fallthrough -D_GLIBCXX_ASSERTIONS -fstack-protector-strong -fno-delete-null-pointer-checks -fno-strict-overflow -fno-strict-aliasing -Wtrampolines -Werror=implicit -Werror=incompatible-pointer-types -Werror=int-conversion -Werror=implicit-int -Woverlength-strings -Wparentheses -Wcast-qual -Wuninitialized -Wvarargs -Wwrite-strings -Wrestrict -Wstringop-truncation -Werror=trigraphs -Wunreachable-code -Wcomment -Wsequence-point -Wreturn-type -fvisibility=hidden -Wsign-conversion -fstack-clash-protection -fcf-protection=full -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -ffunction-sections -fdata-sections -fPIC -DHAVE_MEMSET_S -DHAVE_POSIX_STRERR_R -MD -MQ subprojects/opensea-common/libopensea-common.a.p/src_safe_bsearch.c.o -MF subprojects/opensea-common/libopensea-common.a.p/src_safe_bsearch.c.o.d -o subprojects/opensea-common/libopensea-common.a.p/src_safe_bsearch.c.o -c ../subprojects/opensea-common/src/safe_bsearch.c
In file included from ../subprojects/opensea-common/include/sort_and_search.h:2,
                 from ../subprojects/opensea-common/src/safe_bsearch.c:24:
../subprojects/opensea-common/include/common_types.h:79:23: error: conflicting types for 'offset_t'
   79 |         typedef off_t offset_t;//to deal with windows differences in off_t definitions in stat
      |                       ^~~~~~~~
In file included from /usr/include/unistd.h:41,
                 from ../subprojects/opensea-common/include/predef_env_detect.h:29,
                 from ../subprojects/opensea-common/include/common_types.h:32,
                 from ../subprojects/opensea-common/include/sort_and_search.h:2,
                 from ../subprojects/opensea-common/src/safe_bsearch.c:24:
/usr/include/sys/types.h:274:20: note: previous declaration of 'offset_t' was here
  274 | typedef longlong_t offset_t;
      |                    ^~~~~~~~
```
Compiler:
```
lesmiz 14 $ cc -v
Using built-in specs.
COLLECT_GCC=/opt/local/gcc10/bin/cc
COLLECT_LTO_WRAPPER=/opt/local/gcc10/libexec/gcc/x86_64-sun-solaris2.11/10.4.0/lto-wrapper
Target: x86_64-sun-solaris2.11
Configured with: ../gcc-1f8c6b1d90dd69d36e9de7c1962b42e349561085/configure --with-local-prefix=/opt/local --enable-languages=c,c++,fortran,go,objc,lto --enable-__cxa_atexit --enable-initfini-array --disable-nls --disable-libitm --with-gnu-as --with-as=/opt/local/bin/gas --without-gnu-ld --with-ld=/bin/ld --with-zstd=no --prefix=/opt/local/gcc10 --build=x86_64-sun-solaris2.11 --host=x86_64-sun-solaris2.11 --infodir=/opt/local/gcc10/info --mandir=/opt/local/gcc10/man
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.4.0 (GCC) 
```

vonericsen, 2024-10-30:
Hi @drboone,

Thank you for reporting this!

I did not know offset_t was a typedef already in Solaris/Illumos! 
I unfortunately have not had a Illumos test system running for quite some time so I missed this.

Changing the typedef like you mentioned will be the solution. I will work on that.

vonericsen, 2024-10-30:
I have pushed a fix for this issue as well as the one the CI identified in opensea-transport when I pushed the fix.
It's on our develop branch currently, but will get pulled into our next release.

vonericsen, 2025-05-30:
Fixed in [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

closed 2025-05-30 by vonericsen