Seagate/openSeaChest #166: io_utils.h uses va_list without previously having included stdarg.h (Solaris derived systems)

drboone, 2024-10-29:
A number of C files in the common and transport subtrees include `subprojects/opensea-common/include/io_utils.h`.  On Solaris derived systems (in this case, SmartOS, an Illumos distro), since this hasn't included stdargs.h, va_list is undefined.  There's a second idential issue in secure_file.h  Build log:

[va_list.txt](https://github.com/user-attachments/files/17563318/va_list.txt)

Not sure whether you would prefer to include this in the .h files, or from each of the C modules.

Compiler:
```
lesmiz 49 $ cc -v
Using built-in specs.
COLLECT_GCC=/opt/local/gcc10/bin/cc
COLLECT_LTO_WRAPPER=/opt/local/gcc10/libexec/gcc/x86_64-sun-solaris2.11/10.4.0/lto-wrapper
Target: x86_64-sun-solaris2.11
Configured with: ../gcc-1f8c6b1d90dd69d36e9de7c1962b42e349561085/configure --with-local-prefix=/opt/local --enable-languages=c,c++,fortran,go,objc,lto --enable-__cxa_atexit --enable-initfini-array --disable-nls --disable-libitm --with-gnu-as --with-as=/opt/local/bin/gas --without-gnu-ld --with-ld=/bin/ld --with-zstd=no --prefix=/opt/local/gcc10 --build=x86_64-sun-solaris2.11 --host=x86_64-sun-solaris2.11 --infodir=/opt/local/gcc10/info --mandir=/opt/local/gcc10/man
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.4.0 (GCC) 
```

vonericsen, 2024-10-30:
I have pushed a fix for this issue as well as the one the CI identified in opensea-transport when I pushed the fix.
It's on our develop branch currently, but will get pulled into our next release.

drboone, 2024-10-30:
Much appreciated!

vonericsen, 2025-05-30:
Fixed in [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

closed 2025-05-30 by vonericsen