Seagate/openSeaChest #168: undefined use of `num_devs` in uscsi_helper.c

drboone, 2024-10-30:
In fucnction get_Device_List at line 610 in subprojects/opensea-transport/src/uscsi_helper.c, the variable num_devs is referenced.  In one other function, num_devs is defined, but not in this one.  Adding a definition for num_devs obviously resolves the compile error, but this particular function defines `uint32_t numberOfDevices = 0;` which makes me suspect that 
just defining num_devs is not enough for proper function.

Compiler error (-Wmissing-declarations):
```
lesmiz 74 $ ninja -C builddir
ninja: Entering directory `builddir'
[1/21] Compiling C object subprojects/opensea-transport/libopensea-transport.a.p/src_uscsi_helper.c.o
FAILED: subprojects/opensea-transport/libopensea-transport.a.p/src_uscsi_helper.c.o 
cc -Isubprojects/opensea-transport/libopensea-transport.a.p -Isubprojects/opensea-transport -I../subprojects/opensea-transport -I../subprojects/opensea-transport/include -I../subprojects/opensea-transport/include/vendor -I../subprojects/opensea-common/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=gnu99 -O3 -Wshadow=compatible-local -Wvla -Wfloat-equal -Wnull-dereference -Wunused-const-variable -Wunused-parameter -Wunused-value -Wduplicated-cond -Wjump-misses-init -Wstringop-overflow -Wlogical-op -Wshift-overflow -Wshift-overflow=1 -Wshift-overflow=2 -Wdouble-promotion -Wformat-security -Wold-style-definition -Wstrict-prototypes -Wmissing-declarations -Wmissing-prototypes -Wchar-subscripts -Wundef -Wformat -Wformat=2 -Wint-conversion -Wenum-conversion -Wfloat-conversion -Wint-to-pointer-cast -Wimplicit-fallthrough -D_GLIBCXX_ASSERTIONS -fstack-protector-strong -fno-delete-null-pointer-checks -fno-strict-overflow -fno-strict-aliasing -Wtrampolines -Werror=implicit -Werror=incompatible-pointer-types -Werror=int-conversion -Werror=implicit-int -Woverlength-strings -Wparentheses -Wcast-qual -Wuninitialized -Wvarargs -Wwrite-strings -Wrestrict -Wstringop-truncation -Werror=trigraphs -Wunreachable-code -Wcomment -Wsequence-point -Wreturn-type -fvisibility=hidden -Wsign-conversion -fstack-clash-protection -fcf-protection=full -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -DENABLE_CSMI -DDISABLE_NVME_PASSTHROUGH -ffunction-sections -fdata-sections -fPIC -DHAVE_MEMSET_S -DHAVE_POSIX_STRERR_R -MD -MQ subprojects/opensea-transport/libopensea-transport.a.p/src_uscsi_helper.c.o -MF subprojects/opensea-transport/libopensea-transport.a.p/src_uscsi_helper.c.o.d -o subprojects/opensea-transport/libopensea-transport.a.p/src_uscsi_helper.c.o -c ../subprojects/opensea-transport/src/uscsi_helper.c
../subprojects/opensea-transport/src/uscsi_helper.c: In function 'set_Device_Name':
../subprojects/opensea-transport/src/uscsi_helper.c:61:20: warning: conversion to 'size_t' {aka 'long unsigned int'} from 'int' may change the sign of the result [-Wsign-conversion]
   61 |     snprintf(name, sizeOfName, "%s", s);
      |                    ^~~~~~~~~~
../subprojects/opensea-transport/src/uscsi_helper.c: In function 'get_Device_List':
../subprojects/opensea-transport/src/uscsi_helper.c:567:45: warning: comparison of unsigned expression in '>= 0' is always true [-Wtype-limits]
  567 |         for (driveNumber = 0; ((driveNumber >= 0 && driveNumber < MAX_DEVICES_TO_SCAN && driveNumber < num_rdsk) && (found < numberOfDevices)); ++driveNumber)
      |                                             ^~
../subprojects/opensea-transport/src/uscsi_helper.c:610:43: error: 'num_devs' undeclared (first use in this function)
  610 |         else if(permissionDeniedCount == (num_devs))
      |                                           ^~~~~~~~
../subprojects/opensea-transport/src/uscsi_helper.c:610:43: note: each undeclared identifier is reported only once for each function it appears in
[2/21] Compiling C object subprojects/opensea-operations/libopensea-operations.a.p/src_drive_info.c.o
ninja: build stopped: subcommand failed.
```
FWIW, the same gcc 10.4 and SmartOS environment as the other three bugs I've filed.

vonericsen, 2024-10-30:
@drboone,

Thank you for reporting all of these!
I'm sorry there have been so many issues you've found just trying to compile this right now.

I would love to find a CI solution for Illumos distros to catch these issues since we do not always have an Illumos machine available to test. If you know of one, please let me know!

drboone, 2024-10-30:
The SmartOS folks, being basically a cloud datacenter solution with easy ways to spin up guests for this sort of thing, have a lot of their own tooling.  I've asked about github CI in their IRC channel, and will forward an answers I get.

Glad I could provide something vaguely useful!

vonericsen, 2024-10-30:
@drboone,

Thank you for asking! Let me know if you find out anything we can potentially use and I will look into how to implement it.

Also, thank you for reporting these issues. If you run into more, please feel free to report them and we will do our best to correct them.