Seagate/openSeaChest #169: openSeaChest 24.08.1 cannot compile with ENABLE_ATA_SET_PASSWORD set

achelon5, 2024-11-03:
When **ENABLE_ATA_SET_PASSWORD** flag is set in the Makefile like so:

`PROJECT_DEFINES += -DENABLE_ATA_SET_PASSWORD -DDISABLE_TCG_SUPPORT -DUSING_MUSL_LIBC=$(USING_MUSL)`

Compilation errors like so: 

```
../../utils/C/openSeaChest/openSeaChest_Security.c: In function ‘main’:
../../utils/C/openSeaChest/openSeaChest_Security.c:455:79: error: incompatible type for argument 3 of ‘get_And_Validate_Integer_Input_Uint16’
  455 |                 if (!get_And_Validate_Integer_Input_Uint16(optarg, M_NULLPTR, &ATA_SECURITY_MASTER_PW_ID))
In file included from ../../utils/C/openSeaChest/openSeaChest_Security.c:22:
../../subprojects/opensea-common/include/io_utils.h:71:118: note: expected ‘eAllowedUnitInput’ {aka ‘enum _eAllowedUnitInput’} but argument is of type ‘uint16_t *’ {aka ‘short unsigned int *’}
   71 |     M_NODISCARD bool get_And_Validate_Integer_Input_Uint16(const char * strToConvert, char** unit, eAllowedUnitInput unittype, uint16_t * outputInteger);
      |                                                                                                    ~~~~~~~~~~~~~~~~~~^~~~~~~~
../../utils/C/openSeaChest/openSeaChest_Security.c:455:22: error: too few arguments to function ‘get_And_Validate_Integer_Input_Uint16’
  455 |                 if (!get_And_Validate_Integer_Input_Uint16(optarg, M_NULLPTR, &ATA_SECURITY_MASTER_PW_ID))
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../subprojects/opensea-common/include/io_utils.h:71:22: note: declared here
   71 |     M_NODISCARD bool get_And_Validate_Integer_Input_Uint16(const char * strToConvert, char** unit, eAllowedUnitInput unittype, uint16_t * outputInteger);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gmake: *** [Makefile:391: ../../utils/C/openSeaChest/openSeaChest_Security.o] Error 1
```

vonericsen, 2024-11-04:
Hi @achelon5,

Thank you for reporting this error. I have pushed a fix to the develop branch to correct this build error.

achelon5, 2024-11-04:
I can confirm this is now fixed

closed 2024-11-05 by vonericsen