Seagate/openSeaChest #170: Exos X16 16TB drives on Intel Controller timing out when SATA link power management enabled

lwfitzgerald, 2024-11-16:
I've got a bunch of Exos X16 16TB (`ST16000NM001G`) drives connected to the Intel W680 chipset SATA ports on my [motherboard](https://www.asus.com/motherboards-components/motherboards/workstation/pro-ws-w680-ace-ipmi/). If SATA link power management is enabled there appear to be occasional ATA timeouts communicating with the drives when they're put under load. The errors look like this:

    Nov 01 22:54:51 ubuntu kernel: ata5.00: exception Emask 0x0 SAct 0x0 SErr 0x50000 action 0x6 frozen
    Nov 01 22:54:51 ubuntu kernel: ata5: SError: { PHYRdyChg CommWake }
    Nov 01 22:54:51 ubuntu kernel: ata5.00: failed command: FLUSH CACHE EXT
    Nov 01 22:54:51 ubuntu kernel: ata5.00: cmd ea/00:00:00:00:00/00:00:00:00:00/a0 tag 13
                                            res 40/00:81:81:00:00/00:00:00:00:00/40 Emask 0x4 (timeout)
    Nov 01 22:54:51 ubuntu kernel: ata5.00: status: { DRDY }
    Nov 01 22:54:51 ubuntu kernel: ata5: hard resetting link
    Nov 01 22:54:51 ubuntu kernel: ata5: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
    Nov 01 22:54:51 ubuntu kernel: ata5.00: configured for UDMA/133
    Nov 01 22:54:51 ubuntu kernel: ata5.00: retrying FLUSH 0xea Emask 0x4
    Nov 01 22:54:51 ubuntu kernel: ata5: EH complete

I believe I've ruled out a few of the common causes:

* SATA cables - replaced
* SATA port on [motherboard](https://www.asus.com/motherboards-components/motherboards/workstation/pro-ws-w680-ace-ipmi/) - Issue occurs with Exos drives connected to any of the ports
* Single faulty drive -
   * The issue occurs across the multiple Exos drives I have
   * I've completed `badblocks` write+read passes over the drives with no issues found
   * SMART data appears to be clean on the drives (see below for a full report)
* [PSU](https://www.corsair.com/uk/en/p/psu/cp-9020200-uk/rmx-series-rm850x-850-watt-80-plus-gold-fully-modular-atx-psu-uk-cp-9020200-uk) - Brand new (over-sized PSU)
* I'm unable to reproduce the issue with SATA link power management disabled.

I've started to suspect the issue is the Exos drives as I'm unable to reproduce the issue with other models of drive (still SATA 6Gbps, with SATA link power management support) connected to the same ports.

The only configuration changes I've made to the drives (using bootable SeaChest) are:
* FastFormat to native 4K sectors (disable 512e)
* Disable Idle_b EPC (head parking), Idle_a remains enabled with the default timer

I've checked the serial numbers of the drives on the SeaGate website and it does not report any firmware updates currently available (currently on `SN03`).

I've included the SMART data report and `openSeaChest_Info --showPhyEvents` output for one of the drives at the end of this issue.

I was wondering if it might be possible to reach out to someone internally at Seagate that would know if there is a known incompatibility / firmware bug that might cause this. I've reached out to support via the Seagate website (Case No. `12871167`), but they weren't able to help due to the drives not being connected to either a Windows or Mac machine.

Also of note is the similarity with this other open issue - https://github.com/Seagate/openSeaChest/issues/162.
---

SMART data from one of the drives:

    # smartctl -x /dev/sdb
    smartctl 7.4 2023-08-01 r5530 [x86_64-linux-6.11.0-9-generic] (local build)
    Copyright (C) 2002-23, Bruce Allen, Christian Franke, [www.smartmontools.org](http://www.smartmontools.org/)

    === START OF INFORMATION SECTION ===
    Model Family:     Seagate Exos X16
    Device Model:     ST16000NM001G-2KK103
    Serial Number:    <redacted>
    LU WWN Device Id: <redacted>
    Firmware Version: SN03
    User Capacity:    16,000,900,661,248 bytes [16.0 TB]
    Sector Size:      4096 bytes logical/physical
    Rotation Rate:    7200 rpm
    Form Factor:      3.5 inches
    Device is:        In smartctl database 7.3/5528
    ATA Version is:   ACS-4 (minor revision not indicated)
    SATA Version is:  SATA 3.3, 6.0 Gb/s (current: 6.0 Gb/s)
    Local Time is:    Wed Nov  6 17:44:46 2024 GMT
    SMART support is: Available - device has SMART capability.
    SMART support is: Enabled
    AAM feature is:   Unavailable
    APM feature is:   Unavailable
    Rd look-ahead is: Enabled
    Write cache is:   Enabled
    DSN feature is:   Disabled
    ATA Security is:  Disabled, NOT FROZEN [SEC1]
    Write SCT (Get) Feature Control Command failed: Connection timed out
    Wt Cache Reorder: Unknown (SCT Feature Control command failed)
[... 295 lines cut ...]
`openSeaChest_Info --showPhyEvents` for the same drive as the above SMART data:

    # openSeaChest_Info --showPhyEvents -d /dev/sg1
    ==========================================================================================
     openSeaChest_Info - openSeaChest drive utilities - NVMe Enabled
     Copyright (c) 2014-2024 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
     openSeaChest_Info Version: 2.7.0-8_0_1 X86_64
     Build Date: Sep 25 2024
     Today: 20241106T174542 User: root
    ==========================================================================================

    /dev/sg1 - ST16000NM001G-2KK103 - <redacted> - SN03 - ATA

    ====SATA Phy Event Counters====
    V = Vendor Unique event tracker
    M = Counter maximum value reached
    D2H = Device to Host
    H2D = Host to Device
        ID                Value Description
        10                    1 H2D FISes sent due to COMRESET
         1                    0 Command failed with iCRC error
         3                    0 R_ERR response for D2H data FIS
         4                    0 R_ERR response for H2D data FIS
         6                    0 R_ERR response for D2H non-data FIS
         7                    0 R_ERR response for H2D non-data FIS

vonericsen, 2024-11-26:
@lwfitzgerald,

Thanks for reporting this here!

Can you share the output of `openSeaChest_Info -d <handle> -i -v4 | tee verboseInfo.txt`
If you can get it for both the motherboard and the asmedia controller for comparison that could be helpful to see if there are differences in how the drive is setup by the controller on boot.
The -v4 runs in very very verbose mode with the raw data buffers returned so I can review the bit fields manually. Not everything it humanized out of the low-level drive info since some of it doesn't impact most of what these tools provide and what users are after.

I'm also working on figuring out who inside Seagate may be the right person to help debug what this issue is, but that may take some time to figure out who that should be.

Temtaime, 2025-08-26:
Same issue here.
Disabled Idle_b and formatted to 4K. Now i/o errors in dmesg :(

lwfitzgerald, 2025-09-14:
@vonericsen Apologies for the very delayed response here! Finally found a moment to capture the data today.

Here are the outputs for the command with the drive attached to the two controllers:
* [ASMedia ASM1166](https://github.com/user-attachments/files/22321060/asm1166-verbose-info.txt)
* [Intel W680](https://github.com/user-attachments/files/22321062/intel-verbose-info.txt)

@Temtaime I'm curious, did the issue only show up for you once you formatted to 4K? If you go back to 512B, do you still see the issue?