Seagate/openSeaChest #173: Add errorLimit unit requirement to docs

GrigoryEvko, 2024-12-28:
All tools across SeaChest (from Seagate website) and openSeaChest use strange parsing method for --errorLimit.

```
sudo ./SeaChest_GenericTests -d /dev/sg1 --longGeneric --genericMode verify --repairAtEnd --errorLimit 65000

Error in option --errorLimit. Invalid argument given '65000'.
Use -h option to view command line help
```

looking at the source code I figured out that I need to add 'l' or 'p' to the end of the value, i.e. `65000l`

```
sudo ./SeaChest_GenericTests -d /dev/sg1 --longGeneric --genericMode verify --repairAtEnd --errorLimit 65000l
```

Please add the according clarification to the docs. Thanks!

vonericsen, 2024-12-30:
Hi @GrigoryEvko,

Thank you for reporting this!

The reason for this is to specify limits as either logical sectors or physical sectors.
There was confusion many years ago about the difference and what Seagate and customers were using for the limits so this was added.

Most likely with updates to command line parsing as part of security enhancements this went from a default mode of "logical" to requiring the unit. I will double check if that can be fixed to allow for no units (`l` or `p`) to continue functioning as a logical sector count still.
If I cannot get the previous behavior back I will make sure the docs are updated to describe the required units.

vonericsen, 2025-05-30:
Fixed in [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)