Seagate/openSeaChest #181: PowerControl Standby invalid argument

Master1X, 2025-04-25:
Hey I read a bit and successfully compiled the program. Also in the --help section this command is written down like this and others posted same arguments. 

However if I do that I always get that " Invalid argument given '5000' ". Don't know if it's a bug or I am missing something. 

For example --showEPCSettings works fine

`sudo ./openSeaChest_PowerControl -d /dev/sg0 --idle_a 5000 --standby_z 30000 --idle_b enable --idle_c disable

Error in option --idle_a. Invalid argument given '5000'.
Use -h option to view command line help
`

vonericsen, 2025-04-25:
Hi @Master1X,

Thanks for reporting this!

I found and fixed the issue on the develop branch of our library opensea-common.
I've also pulled that into openSeaChest with the second commit to update the submodule pointer.

The issue was due to refactoring the reading and validating command line inputs and the case for not providing a time unit was not handled properly.
A quick workaround without these commits is to add `ms` to the end of your inputs like this:
`openSeaChest_PowerControl -d /dev/sg0 --idle_a 5000ms --standby_z 30000ms --idle_b enable --idle_c disable`

You can also use other units like `5s` for 5 seconds or `30s` for 30 seconds as well.

If you pull these other two commits in it will restore the previous behavior of allowing no units for the input.

This issue also comes in at a great time as I am working on polishing up things before we do a new release, so thank you for pointing this out!
I'll go test other command line options that require units to see if they are also having similar problems.

vonericsen, 2025-05-30:
Fixed in [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

closed 2025-05-30 by vonericsen