Seagate/openSeaChest #196: [BUG] - Incompatibility with mpt3sas

Forza-tng, 2025-06-09:
**Describe the bug**
I cannot update firmware on SATA Exos X18 drives on mpt3sas - Broadcom HBA 9500-8i (SAS3808)

**To Reproduce**
Steps to reproduce the behavior:
1. Attach Exos SATA drives
2. use openSeaChest to update firmware

3. update aborts after 30 seconds
4. See error in dmesg

**Expected behavior**
Firmware should have been updated, but the download fails.

**Verbose Output**
```
[2277096.910173] [ T1514] sd 0:0:2:0: task abort: SUCCESS scmd(0x000000004b102cf7)
[2277166.049222] [ T1514] sd 0:0:2:0: attempting task abort!scmd(0x0000000043239087), outstanding for 30132 ms & timeout 30000 ms
[2277166.049227] [ T1514] sd 0:0:2:0: [sdc] tag#3761 CDB: ATA command pass through(12)/Blank a1 16 03 03 00 02 00 00 00 92 00 00
[2277166.049229] [ T1514] scsi target0:0:2: handle(0x0028), sas_address(0x5003048020db4542), phy(2)
[2277166.049231] [ T1514] scsi target0:0:2: enclosure logical id(0x5003048020db457f), slot(2)
[2277166.049233] [ T1514] scsi target0:0:2: enclosure level(0x0000), connector name( C0  )
[2277166.409679] [ T1514] sd 0:0:2:0: task abort: SUCCESS scmd(0x0000000043239087)
```

**Desktop (please complete the following information):**
 - OS: Alpine Linux 3.22
 - Kernel 6.12.25
 - openSeaChest-v25.05.1-linux-i686-portable

[verboseinfo.zip](https://github.com/user-attachments/files/20657415/verboseinfo.zip)
```
Version Info for openSeaChest_Firmware:
        Utility Version: 4.3.0
        opensea-common Version: 6.0.2
        opensea-transport Version: 10.0.0
        opensea-operations Version: 9.1.0
        Build Date: Jun  5 2025
        Compiled Architecture: X86
        Detected Endianness: Little Endian
        Compiler Used: GCC
        Compiler Version: 11.2.1
        Operating System Type: Linux
        Operating System Version: 6.12.25-0
        Operating System Name: Alpine Linux v3.21
```
**Additional context**
I have /sys/block/sdc/device/timeout and eh_timeout at 300 seconds, but it seems that `mpt3sas` driver uses a hard coded 30 second value, so it aborts the command before the firmware is fully updated. 

In `mpt3sas_scsih.c`function `scsih_abort()`:
https://github.com/torvalds/linux/blob/19272b37aa4f83ca52bdf9c16d5d81bdd1354494/drivers/scsi/mpt3sas/mpt3sas_scsih.c#L3292
```c
	u8 timeout = 30;
	struct _pcie_device *pcie_device = NULL;
	sdev_printk(KERN_INFO, scmd->device, "attempting task abort!"
	    "scmd(0x%p), outstanding for %u ms & timeout %u ms\n",
	    scmd, jiffies_to_msecs(jiffies - scmd->jiffies_at_alloc),
	    (scsi_cmd_to_rq(scmd)->timeout / HZ) * 1000);
	_scsih_tm_display_info(ioc, scmd);
```

vonericsen, 2025-06-09:
Hi @Forza-tng,

Thanks for reporting this.
What is odd is that this is happening on the first download segment of many segments that will be transmitted....that is bizarre since these generally complete quickly (milliseconds at most) until the activation event at the end of the download.

Can you try these to see if either of them work properly?
`openSeaChest_Firmware -d <handle> --downloadFW file.lod --downloadMode segmented`
`openSeaChest_Firmware -d <handle> --downloadFW file.lod --downloadMode segmented --forceSCSI`

Forza-tng, 2025-06-09:
Thanks for the quick reply. I did indeed try segmented download but forgot to mention it. I will try `forceSCSI` when I get back to office.

Forza-tng, 2025-06-11:
I appreciate the help. Let me know if you want additional testing on my side. I am glad the workaround with `--forceSCSI` worked, so from my side this enough to close this ticket.

vonericsen, 2025-06-11:
If I think of anything else I will let you know, but for now I want to leave this open.
I know we have some of these HBA's around inside Seagate so I just need to get some more testing to see if I can repeat the issue.

Would you be able to share the output of `openSeaChest_Firmware -d <driveHandle> --llInfo`?
This gets the low-level info that is read from parts of `/sys` to understand the HBA ID, driver version, etc which could be helpful.
Another couple pieces of info that would be helpful are `openSeaChest_Firmware -d <driveHandle> --fwdlInfo` and `openSeaChest_Firmware -d <driveHandle> --fwdlInfo --forceSCSI -v4`.
This collects capabilities for updating firmware that the drive supports. The second with `--forceSCSI` forces it to use the SCSI translated data from the HBA. What I'm looking for is if the HBA supports one of the newer methods available from the SAT specification because that would make it easy to identify when to use the translated method and that would be easy to auto-detect. The `-v4` is so I can look at all the raw data reported in case the parsed output misses something.

The only other info that could help for debugging would be the firmware version of the HBA. I think this would have to be read by Broadcom's storecli (or similar...I cannot remember the name off the top of my head this morning 😆 ...need more coffee). That could also help since sometimes HBA firmware versions affect this kind of behavior too.

Forza-tng, 2025-06-11:
> [@Forza-tng](https://github.com/Forza-tng), Thank you for sharing all that info! That will be very helpful for debugging and repeating this issue!
> 
> I realized one other thing that will be helpful to have: `openSeaChest_Firmware -d <handle> -i --SATInfo -v4` This reports the data we can get from both the translator (HBA) and the SATA drive so I can compare for other differences and capabilities. In this case I want to see the very verbose output to check all the raw bytes to make sure there isn't anything else we can use in deciding when to attempt the translated download versus the passthrough download.

I will check this on Friday as I am not in office tomorrow.

vonericsen, 2025-07-03:
I now know the root cause of this and it was fixed in Seagate/opensea-transport#23

The issue that I missed in my first viewing of the verbose log is that the download command issues was the PIO mode version, however it was submitted with the protocol set to UDMA which caused the hang.

closed 2025-08-01 by vonericsen