Seagate/openSeaChest #211: [Feature] - Title

nickhayhurst-interact, 2025-08-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

No - my feature request is not related to a problem.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Something I have always figured would be a benefit to openSeaChest, is a global "JSON" output format.

openSeaChest provides great data, and in many cases, data that other tools does not provide.

Please consider implementing an output format for openSeaChest in the JSON format.

`--outputFormat=json`

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Tools like smartmontools provide a JSON output format for their toolkit - hdsentinel for Linux provides an XML output - which could be another alternative output format.

**Additional context**
Add any other context or screenshots about the feature request here.
You may also include examples of other software that implement this feature

No further additional context. :)

nickhayhurstdev, 2025-10-24:
Hey @vonericsen 

Hope you are well.

Is JSON this available for any testing? Happy to support on testing here! 👍 

Thanks.

vonericsen, 2025-10-24:
@nickhayhurstdev,

I'm doing well and I hope you are too!

JSON support is in progress. #232 is part of integrating that support. #231 is another part of pulling in JSON output.
We've been working through some issues with the json-c library and assessing the CI jobs.
I ended up forking it and submitting some pull requests back to the main project which resolve some of the issues.

One issue we are hitting and trying to decide what to do with is in regards to the makefiles we still preserve. Those builds fail in the CI and we do not know why (they work locally, just not in the CI). We do have some need to maintain them since we occasionally need to support something more rare like AIX or ESXI which don't have meson targets or the ability to build with meson.
Most likely we'll implement some kind of flag to turn support on or off for now with it defaulting to `on` for meson based builds.

I want to get JSON support completed for our next release....no exact timeline yet, but before the end of the year is the target that I have in mind.

vonericsen, 2025-12-19:
@nickhayhurst-interact,

My earlier comment was trying to get this out before end of 2025, but we've just been swamped with too much to get it all done in that timeframe, but after talking with Nidhi about how to resolve this Windows-Clang issue, we are in agreement that we can get a release in the new year, Jan 2026.

Do give this a test and let us know what you like/don't like/want to see next, etc.

fthobe, 2026-01-09:
@vonericsen in Europe the DPP (Device Passport) will become legally mandatory next year documenting the lifecycle actions of a device. Can you provide me with a sample output? On [schema.org](https://github.com/schemaorg/schemaorg) some work is ongoing to align this output with the fields required by the GS1 DPP standard applicable in Europe? @kevinbouti is extensively working on this over at schema. 

I saw Seagate is a major sponsor of [Circular Drives](https://circulardrives.org/members/), I think it would actually be nice to have some discussion happening over there regarding DPP.

vonericsen, 2026-01-12:
@fthobe,

I was not aware of that law. I will read into it and see what it mentions and how it can fit into this.
For now, would you mind opening a separate issue for the device passport? I'm thinking a new feature for it as this seems important to be on its own unique issue to address.

While I can see how it fits in with this issue, I think we will be rolling out changes to JSON output in a few phases and I think this could be a good update after this first round of JSON support to make sure we have it implemented correctly.
Phase 1 (device info, device statistics, smart, scan) is nearly ready so I don't want to hold it up right now.