Seagate/openSeaChest #218: "Device is in the PM0: Active state or PM1: Idle State" is confusing

lispwright, 2025-08-20:
That --checkPowerMode report _sounds_ like it will always be correct, a drive will be in either of those states.  If that's what the drive says, that's what the drive says and we can search on the net to infer that means the drive is not idle thanks to a couple of other openSeaChest issues here on GitHub, but something more clear would be nice.

I've got 3 Exos X18 ST16000NM000J-2TW103 with firmware SN04, and 1 X24 ST16000NM002H-3KW133 with firmware RE05.  All are fine with Idle A and Idle B enabled, the X24 is new and currently stuck in this reported state after changing to 4K sector size yesterday.  One new x18 probably did that starting a couple of weeks ago, the others have been in service for a year and a half.  Today all the x18s flip between idle and that report that if I read from them.

A note on the factory settings, none of the times match the manuals except for Idle A.  Idle C and Standby Z are not enabled which is good for a default, the X18s Idle B are set to 1200, 12 seconds, the X24 is set to 6000, 1 minute.  The manuals say it should be 2 minutes.

If you care about drive longevity that might be diminished by many head load cycles, and minor power savings don't matter to you, is there a recommended setting for Idle B?  Any reason not to disable Idle B and keep the heads flying at all times, or will that wear out the media faster and effectively cut into the 550TB/year maximum for these drives?

vonericsen, 2025-08-20:
Hi @lispwright,

Thanks for reporting this.

The reason it reports the text `Device is in the PM0: Active state or PM1: Idle State` is due to the way the spec defines the output of the check power mode command in the ATA specifications.

[image]

Unlike the standby state, there are no specific notes to interpret this differently for EPC enabled drives in the standard.
I will need to test what happens if an EPC drive is issued the idle immediate command, then check the reported power mode. I would expect it to translate to idle_a, but will need to verify that. It may report this same state.

Drives without EPC have no way to indicate differences between idle and active, so the most we could do is a change for drives with EPC enabled.

> A note on the factory settings, none of the times match the manuals except for Idle A. Idle C and Standby Z are not enabled which is good for a default, the X18s Idle B are set to 1200, 12 seconds, the X24 is set to 6000, 1 minute. The manuals say it should be 2 minutes.

If I can figure out who wrote that manual, I will pass this along.

> If you care about drive longevity that might be diminished by many head load cycles, and minor power savings don't matter to you, is there a recommended setting for Idle B? Any reason not to disable [...]

I do not know the answer to this. I will see if I can find anyone who is able to answer this question.

vonericsen, 2025-09-02:
Hi @lispwright,

This took a bit of time to find the right person to ask.

I am not sure why the manual mismatches the drive's defaults but I think the reason is due to changes made to products over time not getting communicated back to the writers.
Some earlier products probably specified 2 minutes, but a later generation changed the defaults, and a manual update was missed, causing the mismatch.

The two minutes before entering idle b and parking the heads was likely acceptable for that product during design and testing, but it was likely changed on the X24 to a longer time (10 minutes) due to changes in that product's design.
I have heard that other products change the defaults as well and some may even leave idle_b disabled when shipped too.

The default values from firmware are what will allow the best balance of power saving and reliability through the design and testing of that product.
Changing the defaults to longer times should not cause any adverse effects other than higher power consumption.

> If you care about drive longevity that might be diminished by many head load cycles, and minor power savings don't matter to you, is there a recommended setting for Idle B? Any reason not to disable [...]

Based on my understanding and a conversation I had earlier today, disabling idle_b will not cause any adverse effects.
The manual is trying to communicate that aggressively parking the heads (either by telling it to manually or with a short timer value) may degrade the drive over time.

The drive also needs to be in idle_a with enough time to periodically run some background tasks to continue operating. So leaving idle_b disabled will give the drive more than enough time to do this since it will not need to pause and drop to this lower power state as often due to a timer expiring.