Seagate/openSeaChest #219: [BUG] - openSeaChest_Format causes system crash

Temtaime, 2025-08-22:
**Describe the bug**
I was able to reproduce twice with two of those HDDs.
System hangs completely. No messages in logs, no kernel crash message, no output on the monitor. Just freezes.

**To Reproduce**
Steps to reproduce the behavior:
1. Buy ST24000DM001 HDD
2. Install a debian system, for example, i can reproduce on Proxmox 9.0 with AMD Ryzen 9950X3D
3. openSeaChest_Format -d /dev/sda --setSectorSize 4096 --confirm this-will-erase-data-and-may-render-the-drive-inoperable

**Expected behavior**
Command completes successfully.

**Verbose Output**
```
==========================================================================================
 openSeaChest_Format - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2025 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Format Version: 3.4.0 X86_64
 Build Date: Jul 30 2025
 Today: 20250823T011326 User: root
==========================================================================================
One or more of the options provided will perform a low-level format that cannot
be interrupted once started. All background software should be stopped, any filesystems
that are currently mounted should first be unmounted in order to reduce the risk of
interruption. Do not attempt these operations on multiple devices at the same time
to ensure the best possible outcome. Many controllers/drivers/HBAs cannot handle these
operations running in parallel without issuing a device reset.
Not all background activities can be stopped. Some are managed by the OS and are not
configurable. It is recommended that a format change is done from a live/bootable
environment to reduce the risk of these interruptions. If the OS is unable to complete
certain commands for it's background polling of the device, it may trigger a device
reset and interrupt the format, leaving the drive inoperable if it cannot be recovered.
                There is a risk when performing a low-level format/fast format that may
                make the drive inoperable if it is reset at any time while it is formatting.
                WARNING: Any interruption to the device while it is formatting may render the
                         drive inoperable! Use this at your own risk!
                WARNING: Set sector size may affect all LUNs/namespaces for devices
                         with multiple logical units or namespaces.
                WARNING (SATA): Do not interrupt this operation once it has started or
                         it may cause the drive to become unusable. Stop all possible background
                         activity that would attempt to communicate with the device while this
                         operation is in progress
                WARNING: It is not recommended to do this on USB as not
                         all USB adapters can handle a 4k sector size.
                WARNING: Disable any out-of-band management systems/services/daemons
                         before using this option. Interruptions can be caused by these
                         and may prevent completion of a sector size change.
                WARNING: It is recommended that this operation is done from a bootable environment
                         (Live USB) to reduce the risk of OS background activities running and
                         triggering a device reset while reformating the drive.

If you wish to cancel this operation, press CTRL-C now to exit the software.
 0

/dev/sg0 - ST24000DM001-3Y7103 - ZXA0WHWX - DN01 - ATA
Set Sector Size to 4096
Setting the drive sector size quickly.
This command may appear to hang the utility. Do NOT interrupt this
command for at least 1 hour if it appears hung. The drive is busy
performing the sector size change and is not able to indicate its
progress during this time.
This often only takes a few minutes to complete and return completion
of this command to the utility but in some cases it may take this full
hour before the drive is ready to use.
After this command has completed successfully the drive is ready to
read and write data.
Be aware that the drive may begin performing vendor unique background
activity which may prevent idle and standby timers from taking affect
until the background activity has completed.
If this command takes an hour or the command reports a failure due to
interruption by the system with a reset, recovery will be attempted
automatically. You may attempt to run this command again if recovery
does not appear successfull.
```

And then it freezes

**Desktop (please complete the following information):**
 - OS: Proxmox(debian based)
 - Version 9.0

Hint: All `openSeaChest` tools support the `--version` option which can report this for you.

**Additional context**
Add any other context about the problem here that would be helpful to understand.
This can be a big picture overview of what you are trying to do or what background information led you to running these openSeaChest tools.
Does another tool run this same operation successfully? (e.g.smartctl, hdparm, sdparm, camcontrol, etc)

vonericsen, 2025-09-02:
> But those drives AFAIK 4Kn, so reformatting should only change visible to OS sectors count, isn't it ?
It uses 512e, not 512n

It's not an instant switch for how it reports to the OS. The drive is reformatting itself internally so the servo can address the sectors correctly.
The reformat is to instruct the servo on how to read/write and handle ECC data associated with the 4k or 512 boundaries depending on which mode the drive is operating in.

> I tried to run the command one more time, and it completes almost instantly.

If you run the command again for the same size the drive is already configured in, it will be much faster since it's only going to update/format itself if there is an error somewhere from the last time it was formatted.

> Those disks already work in the production, but not completely full. Should i format them to 512e and try to format to 4Kn one more time, leaving the system ON for an hour or so ?

I would just run a read test like I mentioned. If they pass that, then it should be ok.
Rerunning the command to format to the same size (4k -> 4k or 512e->512e) will also make sure it's formatted correctly and should be enough.

> Also i want to note that the drives pass a long smart test now

The drives should be ok if they can pass that test without an error. 

I have seen a couple different issues when the format is interrupted:

1. The drive will not respond and may not even show up in the system at all
2. The drive does show up, but will not allow the set sector configuration command to run anymore. It may report 512e or 4KN, but a read test will fail at some LBA on the drive. Short DST and Long DST should also report a read failure.

In the second case, our software does have a vendor unique command to try and recover. It is issued automatically when something is detected as wrong during the initial format, and it may recover the drive back to fully functional. If you add `--force` then it bypasses some checks to force it to run this and ensure the formatting is completed successfully regardless of whether the drive had an issue or not in the process. It's really only intended to be used in case the automatic recovery fails so that there is a way to try again.

avirus, 2026-01-14:
same error. after several hours of waiting linux lost rootfs (on another drive). 
# /root/openSeaChest-v25.05.3-linux-x86_64-portable/openSeaChest_Format -d /dev/sg4  --llInfo
==========================================================================================
 openSeaChest_Format - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2025 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Format Version: 3.5.1 X86_64
 Build Date: Oct  9 2025
 Today: 20260114T072246 User: root
==========================================================================================

/dev/sg4 - ST24000NM002H-3KS133 - ZYD082DZ - SN02 - ATA

---Low Level tDevice information---
        ---Drive Info---
                media type: HDD
                drive type: ATA
                interface type: SCSI/SAS/FC/etc
                zoned type: not zoned
                ---adapter info---
                        PCI/PCIe:
                        VendorID: 9005h
                        ProductID: 028Ch
                        Revision: 0001h
                ---driver info---
                        driver name: aacraid
                        driver version string: 1.2.1[50983]-custom
                                major ver: 1
                                minor ver: 2
                                revision: 1
                                build number: 50983
                ---ata flags---
                SCSI Version: 6
                ---Passthrough Hacks---
                        Hacks were setup from the reported adapter information
                        Passthrough type: SAT/system/none
                                ---SCSI Hacks---
                                        NMSP (no Mode page subpages supported)
                                        NRSUPOP (no report supported operation codes command)
                                        MXFER = 65024B
                                ---NVMe Hacks---
                                ---ATA Hacks---
                                        CHK (SAT check condition bit is always supported)
                                        FDMA (SAT dma commands MUST use protocol set to DMA instead of UDMA-in/out)
        ---OS Info---
                handle name: /dev/sg4
                friendly name: sg4
                minimum memory alignment: 8
                ---Linux Unique info---
                        FD is valid
                        SG Driver Version:
                                Major: 3
                                Minor: 5
                                Revision: 36
                OS read-write recommended: false
                last recorded error: 0
                File system Info:
                        No active file systems detected

hamzaseriat, 2026-09-11:
On 512e drives, automatic 2-sector allocation is typically possible based on the size of the data. After updating the format on your disk, it is useful to enter the command -fsutil fsinfo ntfsinfo [disk name]: so you will see if Sector - Strip has been updated after the format.Note that RAID is not enabled in BIOS modes while formatting.