Seagate/openSeaChest #229: [Question] SAS3008 - Sanitize Support - Weird RTFRs

nickhayhurstdev, 2025-09-10:
**Describe what you need help with**

Sanitize Erase / Status Commands via openSeaChest (and others) to... well any SSD I have tried on this HBA so far - Security Erase commands work fine and the RTFRs are .

When executing Sanitize commands via SAS3008 12Gbps HBAs I am seeing the command execute as expected.

However, when polling for progress with --progress sanitize - we are seeing no RTFRs - and the Sanitize output is incorrect. When running this command on the same drive - we see success and we are able to poll for Progress.

Similarly - HDParm outputs a similar response to openSeaChest when executing the Sanitize/Status commands - with there being very little information to go by.

**SANITIZE BLOCK ERASE OUTPUT**

```
sudo /usr/bin/openSeaChest_Erase --confirm this-will-erase-data --sanitize blockerase -d /dev/sg5 -v4
==========================================================================================
 openSeaChest_Erase - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2025 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Erase Version: 4.7.3 X86_64
 Build Date: Jul 30 2025
 Today: 20250910T184441	User: root
==========================================================================================
Sending SCSI Test Unit Ready

  CDB:

        0  1  2  3  4  5
  0x00 00 00 00 00 00 00                                                    

Sending command with send_IO
SG IO Issued as Indirect IO
SG Masked Status = 01h - Check Condition
SG Driver Status = 08h - Driver Sense Data Available

  Sense Data Buffer:

        0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
  0x00 70 00 00 00 00 00 00 0A 00 00 00 00 00 00 00 00                      
  0x10 00 00                                                                

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
Information: 00000000h
Command Specific Information: 00000000h
Command Time (ms): 2.66

Test Unit Ready returning: SUCCESS

Sending SCSI Inquiry

  CDB:

        0  1  2  3  4  5
  0x00 12 00 00 00 60 00                                                    

Sending command with send_IO
SG IO Issued as Indirect IO

  Sense Data Buffer:

        0  1  2  3  4  5  6  7
  0x00 00 00 00 00 00 00 00 00                                              

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
Information: 0000000000000000h
Command Specific Information: 0000000000000000h
Command Time (ms): 8.20

[... 1603 lines cut ...]
Sanitize - Status returning: SUCCESS

	This device has never been sanitized.
```

So this is a tricky one ... - does anyone know what could cause this to comeback with incomplete RTFRs? Therefore meaning we are not able to see the Progress Indication? Even though I know the command is running - and a few seconds longer than this to run - on a SAS2008 HBA - we are seeing the output correctly... strange behaviour.

**Any additional details**

Body of message is too long so will send another comment below this one :)

**Environment**
Any details you can share about where you need help or what your environment is if this applies.
 - OS: Linux
 - Version: Debian 12 - Kernel - 6.1.0-39-amd64 - although - I wonder if giving this a go on Debian 13 with a newer kernel/driver would help.

```
sudo modinfo mpt3sas  
filename:       /lib/modules/6.1.0-39-amd64/kernel/drivers/scsi/mpt3sas/mpt3sas.ko
alias:          mpt2sas
version:        43.100.00.00
license:        GPL
description:    LSI MPT Fusion SAS 3.0 Device Driver
author:         Avago Technologies <MPT-FusionLinux.pdl@avagotech.com>
```

vonericsen, 2025-09-16:
I've narrowed the issue I've been tracking slightly.

There is some kind of timing problem where if there is no gap between running a pre-write pattern followed by sanitize overwrite, it triggers a soft error from the driver when polling for progress.

If I ran these steps manually from the command line, I would leave some idle time between each process I ran. When I put them into a script is when I ran into the problem.

It happens on Ubuntu 24.04 with the 6.10 kernel on a 9300. It does NOT happen in the same configuration with a 9600...but that has a different driver so I'm not surprised.

One difference is that I am getting the RTFR's back consistently when this bug does not happen. They are never blank like you were seeing in my tests.
I would not be shocked if these two issues are related though since all the other symptoms look identical.

Back to more debugging to find this timing problem...

vonericsen, 2025-10-06:
Testing on a system with a 9400 has a different issue than I have seen all together.

On a 9400-8i when the system is sanitize overwriting multiple drives at once, for some unknown reason, all the commands eventually start timing out.
It does not matter if it's commands within openSeaChest or any other software. Even kernel issued commands start timing out. All commands that timeout are allowed to complete while sanitize is running according to T10 and T13 standards.

I'm testing a 9500 as well but it will take a while to get some results.

The common thing between 9300, 9400, and 9500 HBAs is that they all use the same `mpt3sas` driver. The 9200 uses mpt2sas (which is a symlink to mpt3sas now, however I think this is handled slightly differently internally in that module).

I was not able to repeat any issues on the 9600 which uses the `mpi3mr` driver, which is the same driver as the megaraid cards. Earlier IT/IR and Megaraid cards used different drivers, but they are the same going forward.